morrownr/88x2bu-20210702 #176: I get disconnection if I use require_vht=1 and require_ht=1

sateuwdie, 2023-10-01:
I have tried this driver on two different pc, with at least 4 different kernel version
With hostapd if I use this configuration all works fine
```

interface=wlan3
bridge=br0
driver=nl80211
ssid=myssid
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
country_code=IT
hw_mode=a
channel=36
beacon_int=100
dtim_period=1
max_num_sta=16
rts_threshold=2347
fragm_threshold=2346
ignore_broadcast_ssid=0
wmm_enabled=1
ieee80211n=1
ht_capab=[RXLDPC][HT40+][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC1][SU-BEAMFORMEE][MU-BEAMFORMEE][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
```
```

auth_algs=1
wpa=2
wpa_passphrase=whoops
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP
```

but I go slow (10mb of speed for transfer between stations)

If I use require_vht=1 and require_ht=1 in the same configuration, works fine, very high speed but it disconnect after some seconds (sometimes 5 minutes, sometimes 1 minute or less). Those are the debug messages. https://pastebin.com/TJqAXmKg
On client dmesg said

```
[ 3259.532422] RTW: rtw_set_802_11_connect(wlan2)  fw_state=0x00000008
[ 3259.764602] RTW: start auth
[ 3259.765211] RTW: auth success, start assoc
[ 3259.766166] RTW: assoc success
[ 3259.767718] RTW: mac_id : 0
[ 3259.767719] RTW: wireless_mode : 0x44
[ 3259.767720] RTW: mimo_type : 2
[ 3259.767720] RTW: static smps : N
[ 3259.767721] RTW: bw_mode : 80MHz, ra_bw_mode : 80MHz
[ 3259.767721] RTW: rate_id : 9
[ 3259.767722] RTW: rssi : -1 (%), rssi_level : 0
[ 3259.767723] RTW: is_support_sgi : Y, is_vht_enable : Y
[ 3259.767723] RTW: disable_ra : N, disable_pt : N
[ 3259.767724] RTW: is_noisy : N
[ 3259.767724] RTW: txrx_state : 0
[ 3259.767725] RTW: curr_tx_rate : CCK_1M (L)
[ 3259.767725] RTW: curr_tx_bw : 20MHz
[ 3259.767726] RTW: curr_retry_ratio : 0
[ 3259.767726] RTW: ra_mask : 0x00000000fffffff0
[ 3259.768262] RTW: recv eapol packet 1/4
[ 3259.768581] RTW: send eapol packet 2/4
[ 3259.770667] RTW: recv eapol packet 3/4
[ 3259.770760] RTW: send eapol packet 4/4
[ 3259.770812] RTW: set pairwise key camid:0, addr:***, kid:0, type:AES
[ 3259.772554] RTW: set group key camid:1, addr::******, kid:1, type:AES
[ 3270.339838] wlan0: Connection to AP **** lost
[ 3288.300015] wlan0: authenticate with ****
[ 3288.317523] wlan0: send auth to **** (try 1/3)
[ 3288.876101] wlan0: authenticate with ****
[ 3288.878580] wlan0: send auth to **** (try 1/3)
[ 3289.016231] wlan0: authenticated
[ 3289.017542] wlan0: associate with **** (try 1/3)
[ 3289.122533] wlan0: associate with **** (try 2/3)
[ 3289.126722] wlan0: RX AssocResp from **** (capab=0x1511 status=30 aid=0)
[ 3289.126728] wlan0: **** rejected association temporarily; comeback duration 1000 TU (1024 ms) 
[ 3290.210525] wlan0: associate with **** (try 3/3)
[ 3290.338516] wlan0: association with **** timed out
```

On hostapd server dmesg said

```
[dom ott  1 22:30:20 2023] RTW: assoc success
[dom ott  1 22:30:21 2023] IPv6: ADDRCONF(NETDEV_CHANGE): wlan3: link becomes ready
[dom ott  1 22:30:21 2023] br0: port 1(wlan3) entered blocking state
[dom ott  1 22:30:21 2023] br0: port 1(wlan3) entered forwarding state
[dom ott  1 22:30:24 2023] RTW: ============ STA **** ===================
[dom ott  1 22:30:24 2023] RTW: mac_id : 2
[dom ott  1 22:30:24 2023] RTW: wireless_mode : 0x44
[dom ott  1 22:30:24 2023] RTW: mimo_type : 2
[dom ott  1 22:30:24 2023] RTW: static smps : N
[dom ott  1 22:30:24 2023] RTW: bw_mode : 80MHz, ra_bw_mode : 80MHz
[dom ott  1 22:30:24 2023] RTW: rate_id : 9
[dom ott  1 22:30:24 2023] RTW: rssi : -1 (%), rssi_level : 0
[dom ott  1 22:30:24 2023] RTW: is_support_sgi : Y, is_vht_enable : Y
[dom ott  1 22:30:24 2023] RTW: disable_ra : N, disable_pt : N
[dom ott  1 22:30:24 2023] RTW: is_noisy : N
[dom ott  1 22:30:24 2023] RTW: txrx_state : 0
[dom ott  1 22:30:24 2023] RTW: curr_tx_rate : CCK_1M (L)
[dom ott  1 22:30:24 2023] RTW: curr_tx_bw : 20MHz
[dom ott  1 22:30:24 2023] RTW: curr_retry_ratio : 0
[dom ott  1 22:30:24 2023] RTW: ra_mask : 0x00000000fffffff0
[dom ott  1 22:30:24 2023] RTW: send eapol packet 1/4
[dom ott  1 22:30:24 2023] RTW: recv eapol packet 2/4
[dom ott  1 22:30:25 2023] RTW: send eapol packet 1/4
[dom ott  1 22:30:25 2023] RTW: recv eapol packet 2/4
[dom ott  1 22:30:26 2023] RTW: send eapol packet 1/4
[dom ott  1 22:30:26 2023] RTW: recv eapol packet 2/4
[dom ott  1 22:30:27 2023] RTW: send eapol packet 1/4
[dom ott  1 22:30:27 2023] RTW: recv eapol packet 2/4
```

morrownr, 2023-10-02:
Hi @sateuwdie 

Since you did not post the 'options' line in /etc/modprobe.d/88x2bu.conf , I'll assume you have not set it up properly. The easiest way to make sure you get the right file is:

$ sudo edit-options.sh

The default options line is for managed mode but a few lines later in the file is a commented out suggestion for AP mode but you have to read the documentation.

Take care of that and let me know how it goes... oh, and get rid of require_vht=1 and require_ht=1 as I have never found a good use for them.

sateuwdie, 2023-10-02:
Hi, sorry, I provide now the options

```
cat /etc/modprobe.d/rt88x2bu.conf
options 88x2bu rtw_vht_enable=2
```

sateuwdie, 2023-10-02:
Other info, this is the product: https://www.brostrend.com/products/ac3l

sateuwdie, 2023-10-02:
For require_vht=1 and require_ht=1
I use it to reach high speed, I have a vdr box connected via wi-fi
And without those options I go under 10mb/s, when I switch to
hd channel the network congest.
Without those options it not disconnect, but it goes slow

sateuwdie, 2023-10-02:
Actually I try this configuration

88x2bu.conf

`options 88x2bu rtw_drv_log_level=0 rtw_vht_enable=1 rtw_switch_usb_mode=0`

If I use rtw_switch_usb_mode=1 give me a lot of error about bad cable, cannot assign request address..

hostapd.conf

```
interface=wlan3
bridge=br0
driver=nl80211
ssid=Routervdr
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
country_code=IT
hw_mode=a
channel=36
ieee80211n=1
ieee80211ac=1
require_vht=1
require_ht=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
auth_algs=1
wpa=2
sae_password=mypass
wpa_key_mgmt=SAE
rsn_pairwise=CCMP
ieee80211w=2

ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
```
~                                                                                              

Kernel is 5.15.117 on Slackware 15

Hostapd run, the client (Linux Slackware using rtl88x2bu as client) authenticate but..after 3 minutes the network fall, and the client is de-authenticated.

sateuwdie, 2023-10-02:
I try now without use of require_vht=1
and require_ht=1

sateuwdie, 2023-10-02:
I try to disable also the use of 80mhz channel and using 40mhz channel, after 5 minutes it disconnect.

morrownr, 2023-10-02:
> cat /etc/modprobe.d/rt88x2bu.conf
> options 88x2bu rtw_vht_enable=2

This tells me we probably have a mess to clean up.

How did you end up with this file  /etc/modprobe.d/rt88x2bu.conf ?
install-driver.sh installs  /etc/modprobe.d/88x2bu.conf and it is based on the 88x2bu.conf file that is in your clone directory.

You really need to delete  /etc/modprobe.d/rt88x2bu.conf and work with the file that was installed for you. It is likely both files are competing with each other and who knows what options are really being used.

sateuwdie, 2023-10-02:
I have tried also with the options of install-driver.sh

`options 88x2bu rtw_drv_log_level=0 rtw_vht_enable=1 rtw_switch_usb_mode=0`

Same thing.

sateuwdie, 2023-10-02:
I'm trying now with those options using /etc/modprobe.d/88x2bu.conf
`
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_switch_usb_mode=1`

They are identical to install script.

closed 2023-10-03 by sateuwdie

morrownr, 2023-10-03:
Good to see the progress. 

I have a very informative AP guide on the Main Menu:

https://github.com/morrownr/USB-WiFi

Item 9.

You mentioned a Pi4B. Your chipset and this driver and the Pi4B will show drops. We never have been able to figure it out. The fix is to not turn on USB3 mode.

sateuwdie, 2023-10-03:
Yes I know that bug, thanks.
If I not use the usb3 I will loose the speed?
With last configuration I can reach 33MB/s with rsync
between two stations.

morrownr, 2023-10-03:
> If I not use the usb3 I will loose the speed?

Yes. Speed will likely be cut in half. We suspected the USB3 hub chipset as the problem.

sateuwdie, 2023-10-03:
> > If I not use the usb3 I will loose the speed?
> 
> Yes. Speed will likely be cut in half. We suspected the USB3 hub chipset as the problem.

Thanks for answer.

morrownr, 2023-10-03:
I have really enjoyed making my own AP with a RasPi. It has come in handy for several things. Learning hostapd can take a little time but it works well once you have experience using it. You have some options:

- I am going to try to get an updated version of this driver soon so maybe this dropping problem will be fixed. Still not sure what the problem is.
- Kernels 6.2 and later now have an in-kernel driver for this chipset. I've been testing managed mode and it works well. I haven't tested AP mode yet.
- Get an adapter that uses the mt7921au chipset. It has a good in-kernel driver and is WiFi 6 and WiFi 6e (6 Ghz) capable. I use it as my main adapter for AP mode these days.

The Plug and Play list on the Main Menu has several adapters that use the mt7921au chipset. The Main Menu is:

https://github.com/morrownr/USB-WiFi

The Plug and Play list is menu item 2. Menu item 9 has info about AP mode and I even have a hostapd.conf for WiFi 6.

sateuwdie, 2023-10-03:
I see another strange thing, if I rename the wlan using udev rules the problem of disconnection incredibly return!

SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ATTR{address}=="********", ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="wlan*", NAME="wlan0"`

ping 10.3.1.1

....

```
64 bytes from 10.3.1.1: icmp_seq=137 ttl=64 time=1.38 ms
64 bytes from 10.3.1.1: icmp_seq=138 ttl=64 time=1.68 ms
64 bytes from 10.3.1.1: icmp_seq=139 ttl=64 time=1.41 ms
64 bytes from 10.3.1.1: icmp_seq=140 ttl=64 time=1.45 ms
64 bytes from 10.3.1.1: icmp_seq=141 ttl=64 time=1.51 ms
64 bytes from 10.3.1.1: icmp_seq=142 ttl=64 time=1.34 ms
64 bytes from 10.3.1.1: icmp_seq=143 ttl=64 time=1.35 ms
64 bytes from 10.3.1.1: icmp_seq=144 ttl=64 time=1.31 ms
64 bytes from 10.3.1.1: icmp_seq=145 ttl=64 time=1.29 ms
64 bytes from 10.3.1.1: icmp_seq=146 ttl=64 time=7.21 ms
64 bytes from 10.3.1.1: icmp_seq=147 ttl=64 time=13.5 ms
64 bytes from 10.3.1.1: icmp_seq=148 ttl=64 time=1.98 ms
64 bytes from 10.3.1.1: icmp_seq=149 ttl=64 time=2.05 ms
64 bytes from 10.3.1.1: icmp_seq=150 ttl=64 time=1.62 ms
64 bytes from 10.3.1.1: icmp_seq=151 ttl=64 time=1.22 ms
64 bytes from 10.3.1.1: icmp_seq=152 ttl=64 time=1.28 ms
64 bytes from 10.3.1.1: icmp_seq=153 ttl=64 time=10.5 ms
64 bytes from 10.3.1.1: icmp_seq=154 ttl=64 time=12.7 ms
64 bytes from 10.3.1.1: icmp_seq=155 ttl=64 time=1.22 ms
64 bytes from 10.3.1.1: icmp_seq=156 ttl=64 time=2.26 ms
From 10.3.1.20 icmp_seq=183 Destination Host Unreachable
From 10.3.1.20 icmp_seq=184 Destination Host Unreachable
From 10.3.1.20 icmp_seq=185 Destination Host Unreachable
From 10.3.1.20 icmp_seq=186 Destination Host Unreachable
```

sateuwdie, 2023-10-03:
> I have really enjoyed making my own AP with a RasPi. It has come in handy for several things. Learning hostapd can take a little time but it works well once you have experience using it. You have [...]

Thanks,[ this one](https://www.amazon.it/Lechnical-Dual-Band-Adattatore-Trasmissione-Interfaccia/dp/B0BNV9MFD6) can be a good choice? Raspberry support it without recompile kernel?

morrownr, 2023-10-03:
The Comfast CF-953AX was the first mt7921au based adapter that went on the market in July of 2022. I do not have one but we discuss things in Issues on the same page as the Main Menu so you can search the discussions there and you can read the reviews at the site where many people have orders their 953:

https://www.aliexpress.us/item/3256804283254522.html?gatewayAdapt=glo2usa

Even though the 953 is a low cost adapter, I have seen very few complaints. Is the quality such that you would get from an Alfa AXM? No, not at all but the AXM is not a low cost adapter.

The 953 is advertised as Dual Band WiFi 6. That is not correct for Linux users. The in-kernel Linux driver, mt7921u, supports all of the capabilities of the chip so what you get is a Tri-Band, WiFi 6e adapter.

> Raspberry support it without recompile kernel?

The 953 and all other mt7921au chip based adapters are Plug and Play on RasPiOS 2023-05-03 and later. The only thing you might need to do is compile a new hostapd from source if you want to activate WiFi 6. I have a guide on the Main Menu. Expect a new released of the RasPiOS within the next 2 weeks. It is rebased on Debian 12 and should include a new version of hostapd... it should anyway. You can even activate WiFi 6e (6 Ghz) depending on the regdb for your country. I have an example hostapd.conf for WiFi 6 and it is on the Main Meni... testers requested, I don't have a good example for WiFi 6e but will get one up eventually but I can probably help you if you want to do it.

The mt7921au chip is fast and uses very little power (that helps with the RasPi).

FYI: In-kernel drivers are very different than the Realtek out-of-kernel drivers:

- Almost no options. They are like an automatic transmission. If a USB3 adapter is plugged into a USB3 port, USB3 is detected and set. It just works.
- There is no compiling as the driver comes with the kernel and is always up to date. The firmware is installed by the maintainer of your distro.
- These Mediatek in-kernel drivers are made for regular desktop and server users. The Realtek drivers are make for skilled programmers working on embedded projects.

sateuwdie, 2024-01-15:
Actually I have a stable connection using this configuration for 88x2bu

```
server hostapd
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=0

client (using 8852bu card)
options 8852bu rtw_switch_usb_mode=1 rtw_he_enable=1 rtw_vht_enable=1 rtw_dfs_region_domain=3
```