morrownr/88x2bu-20210702 #177: Wpa2 ok, wpa3..sometimes.

sateuwdie, 2023-10-03:
On hostapd server I have this configuration file
```

hostapd.conf

interface=wlan0
bridge=br0
driver=nl80211
ssid=mysid
hw_mode=a
channel=44
ieee80211d=1
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
auth_algs=1
ignore_broadcast_ssid=0
country_code=IT

macaddr_acl=1
accept_mac_file=/etc/hostapd/hostapd.accept

# WPA
wpa=2
wpa_passphrase=****
wpa_key_mgmt=WPA-PSK SAE
wpa_pairwise=CCMP
rsn_pairwise=CCMP
sae_password=****
wps_cred_add_sae=1
ieee80211w=2
own_ip_addr=127.0.0.1

#HT
#user options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_switch_usb_mode=1
require_ht=1
require_vht=1
ht_capab=[MAX-AMSDU-7935][HT40+][SHORT-GI-20][SHORT-GI-40]
```

88x2bu.conf

`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_switch_usb_mode=1`

On client I use the Network-manager

Server is Debian 12, kernel 6.10 from Debian
Client is Slackware 15, kernel 6.5.

The problem: using wpa2 OK, using wpa3 stall for some minutes..and no authentication.
This happen only on Linux clients, Ipad connect with wpa3 without problem

morrownr, 2023-10-04:
Have you tried that Slackware 15 client connected to another AP that supports WPA3?

sateuwdie, 2023-10-04:
No, I don't have any AP which support wpa3.

morrownr, 2023-10-04:
WPA3 requires a lot of parts to be up to speed with WPA3 support. It has been a challenge to get the many distros that support WPA3 up to speed over the last few years.

When I looked at your post, the one thing I do not know if it is working well with WPA3 is Slackware. I've tested this driver with numerous distros and I am not seeing problems so it is likely something besides this driver. I also see Debian 12 and I know it is good to go. If Slackware has a forum, you might make a post.

For hostapd: You can use WPA2/WPA3 mixed and if WPA3 is supported, it will be used.

Here is a paste from my example WiFi 6 hostapd.conf:

```
# Activate only one of the following 3 sections
#
# 1. WPA2 Personal
#wpa_key_mgmt=WPA-PSK
#
# 2. WPA2/WPA3 Transitional (mixed)
wpa_key_mgmt=SAE WPA-PSK
ieee80211w=1
sae_require_mfp=1
#
# 3. WPA3 Personal (required for WiFi 6e - 6 GHz)
#wpa_key_mgmt=SAE
#ieee80211w=2
```

sateuwdie, 2023-10-04:
Ok, probably is a Slackware related problem, because ipad works fine.

closed 2023-10-04 by sateuwdie

sateuwdie, 2023-10-25:
I have solved, the problem was very stupid.
Slackware add this line to the configuration

auth_alg=open

and this cause the "non-connection"

On Debian this line is not added, and the connection
is immediate.

So the solution was to remove the line auth_alg=open from .nm file, this is an example which works

```
/etc/NetworkManager/system-connections/mysid.nmconnection

[connection]
id=mysid
uuid=**********
type=wifi
interface-name=wlan1

[wifi]
mode=infrastructure
ssid=mysid

[wifi-security]
key-mgmt=sae
psk=****

[ipv4]
dns-search=
method=auto

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=auto

```