morrownr/88x2bu-20210702 #178: tplink archer t4uv3

Appleuy, 2023-10-16:
tplink archer t4uv3 use chipset rtl88x2bu in kali linux kernel 6.3 have not driver in-kernel

morrownr, 2023-10-16:
Hi @Appleuy 

I see a statement but no question or request. Where are you headed with this?

How do you know the in-kernel driver is not there?

Appleuy, 2023-10-20:
Ok

morrownr, 2023-10-20:
If you need me to go over the details for checking if the in-kernel dirver is there, I can. Keep in my that if you install this driver, it will blacklist the in-kernel driver so you will have to remove this driver to be able to use the in-kernel driver.

Nick

Appleuy, 2023-10-21:
Thank you for your enthusiasm. My USB wifi has received the in-kernel
driver. Once again, thank you for your enthusiasm.

morrownr, 2023-10-21:
Glad you are going.

Nick