morrownr/88x2bu-20210702 #179: Unable to connect to access point

Ekiben542, 2023-10-25:
I'm using tp-link's Archer T3U.
I was trying to create an access point using hostapd and dnsmasq, and when I tried to connect to the Archer T3U's radio waves, I was unable to communicate. (Accurately, "No Internet" is displayed. No IP is assigned, etc.)
What should I do in this case?
The environment is Bookworm Debian 12 with Raspberry pi 4 Model B Rev 1.5.
Depending on your reply, I will also show you the hostapd logs (sorry, I don't have time right now).

morrownr, 2023-10-26:
Hi @Ekiben542 

How about we try this:

The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu item 9 is about AP mode. It has example hostapd.conf files in the main document. Info for your adapter is there.

Debian 12 uses NetworkManager. For AP mode to work, you either have to tell NM to not message with the interface you are using or you need to purge/inactivate NM. There are examples of how to do as much in the document.

Let me know how it goes.

Ekiben542, 2023-10-26:
How to keep Network Manager from causing problems. section? (I wonder if there is a way to purge/deactivate NM here.)

morrownr, 2023-10-26:
From the guide:

This is the least intrusive way to keep NM from messing with hostapd:

How to keep Network Manager from causing problems.

Option 1:

Tell Network Manager to ignore specific devices.

```
$ sudo nano /etc/NetworkManager/NetworkManager.conf
```

add

```
[keyfile]
unmanaged-devices=interface-name:wlan0
```

Note: Remember to replace wlan0 with your interface name.

----

Here is the direct link to the guide: (Main Menu item 9)

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

If you go to the bottom of the document and work your way up, you will see other options.

Ekiben542, 2023-10-26:
It's working! Thank you.

morrownr, 2023-10-26:
Glad you got it going.

Ekiben542, 2023-11-15:
I'm experiencing the same thing again.

morrownr, 2023-11-17:
NM can be a pain in the backside. I've been working on my AP guide more given the new version of RasPiOS, which change a lot of things.

I made the decision to fully disable NM:

```
sudo systemctl disable NetworkManager
```

I went with the following:

The networking:

systemd-networkd
systemd-resolved

The AP:

hostapd

That is for a bridged setup but it looks like the capability to do a routed setup are available but it would take some work. I went with systemd-networkd starting a new years ago as it is 24/7/365 dependable and other ways of setting up an AP are not quite that good.

You can see what I did in the current AP Mode guide.

Ekiben542, 2023-11-17:
I purge NetworkManager (sudo apt purge NetworkManager), is it okay? (Can the above solution work?)

morrownr, 2023-11-18:
@Ekiben542 

Let's back up a moment and go over some things.

> I was trying to create an access point using hostapd and dnsmasq...

dnsmasq tells me you want a routed AP. Does it need to be routed or can it be bridged? Bridged is easier.

If it can be bridged, I'll recommend you start with a clean sd with Debian 12 or RasPiOS 2023-10-10 and follow my AP guide exactly. It should work if you are able to understand the guide. I realize it is a long detailed guide but it helps you learn. Taking it slow and asking questions is a good idea. Setting up the driver for this adapter is important as well as you may need to stat in USB2 mode. I can go over that with you as well.

Keep in mind that what you are trying to do is somewhat complicated and one thing wrong will cause not to work. Relax, start clean and don't get in a hurry.

@Ekiben542

Ekiben542, 2023-11-19:
Understood.
And I .ave a question: I'm currently running the AP with RaspAP, but should I install RaspAP when I do a clean install? Or should I follow your guide?

morrownr, 2023-11-19:
> The environment is Bookworm Debian 12 ...

You said the above so I assumed you are running Debian 12. I'm not sure what RaspAP is.

Please do a clean installation of the Raspberry Pi OS 2023-10-10 64 bit if you see that as an option. Once complete, please feel free to follow my guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

My guide does not cover installation of the Raspberry Pi OS or the installation of the driver in this repo. It does cover some details of driver configuration. You will need to pay attention to the driver options that you set and please send a copy to me so that I can confirm you used the correct options for AP mode.

Ekiben542, 2023-11-21:
I'ts Working! thanks(If it doesn't work again, I'll open it (issue).)

closed 2023-11-21 by Ekiben542