morrownr/88x2bu-20210702 #180: Driver install not working on TrueNAS Cobia 22.22

Weasy666, 2023-10-29:
I have successfully installed and run this driver on TrueNAS Angelfish and Bluefin, but after updating to TrueNAS Cobia the installation will fail.

This is the output of the install command:

```
root@truenas[~/88x2bu-20210702]# sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20230830
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 24/24 (in-use/total processing units)
: 65499072 (total system memory)
: 6.1.55-production+truenas (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
EFI variables are not supported on this system
EFI variables are not supported on this system
EFI variables are not supported on this system
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.
: ---------------------------

Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.55-production+truenas (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

and here is the content of the log file:

<details>
<summary>log content (click to reveal)</summary>

```
DKMS make.log for rtl88x2bu-5.13.1 for kernel 6.1.55-production+truenas (x86_64)
Sun Oct 29 20:11:07 CET 2023
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.55-production+truenas/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.55-production+truenas'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_wlan_util.o
[... 102 lines cut ...]
      |                   ^~~~~~~~~~~~~
      |                   ieee802154_ptr
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: In function 'cfg80211_rtw_add_virtual_intf':
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:6679:38: error: 'struct net_device' has no member named 'ieee80211_ptr'; did you mean 'ieee802154_ptr'?
 6679 |                         wdev = ndev->ieee80211_ptr;
      |                                      ^~~~~~~~~~~~~
      |                                      ieee802154_ptr
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:6718:30: error: 'struct net_device' has no member named 'ieee80211_ptr'; did you mean 'ieee802154_ptr'?
 6718 |                 wdev = ndev->ieee80211_ptr;
      |                              ^~~~~~~~~~~~~
      |                              ieee802154_ptr
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: In function 'rtw_wdev_alloc':
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10664:18: error: 'struct net_device' has no member named 'ieee80211_ptr'; did you mean 'ieee802154_ptr'?
10664 |         pnetdev->ieee80211_ptr = wdev;
      |                  ^~~~~~~~~~~~~
      |                  ieee802154_ptr
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:2015: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.55-production+truenas'
make: *** [Makefile:2523: modules] Error 2
```

</details>

Is there anything that can be done, to make the installationg work again?

morrownr, 2023-10-29:
Hi @Weasy666 

> Is there anything that can be done, to make the installationg work again?

I'm sure it can be done but we need to figure out the problem so as to see if it is feasible.

I'm not familiar with TrueNAS so help me understand. What distro is it based on? Red Hat, Debian, Ubuntu or ?

Weasy666, 2023-10-29:
As far as i know, it's based on Debian, but they left out a lot of drivers:
[image]

If you click `log content (click to reveal)` and expand that, you can see, that some member named `ieee80211_ptr` seems to be missing in the kernel headers of TrueNAS.

morrownr, 2023-10-30:
> you can see, that some member named ieee80211_ptr seems to be missing in the kernel headers of TrueNAS.

Yes, I saw that. That very well could be a problem in the kernel headers. I think this is something that should be reported to TrueNAS. I just checked and this repo was cloned 254 times yesterday. My point being that this is very heavily used repo and I am not seeing this from anyone else.

Weasy666, 2023-11-01:
The new TrueNAS version was just released last week and maybe i am a bit of an odd ball to Wi-Fi with my server. Reporting to TrueNAS is probably useless, they don't want to support Wi-Fi on their professional/enterprise server software 🫤

morrownr, 2023-11-02:
What could be easier than discovering how to work around the TrueNAS broken header files is to look inside to see if they compiled any usb wifi drivers in their distro.

$ cd /usr/lib/modules/$(uname -r)/kernel/drivers/net/wireless/mediatek/mt76
$ ls -l

Paste what you get and we will know what is available.