morrownr/88x2bu-20210702 #181: Hostapd: works fine but disappear fast.

sateuwdie, 2023-10-30:
My hostapd AP disappear fast.

This is the configuration for AP

```
interface=wlan0
bridge=br0
driver=nl80211
ssid=Router1
hw_mode=a
channel=36	
ieee80211d=1
ieee80211h=1
ieee80211ac=1
ieee80211n=1
wmm_enabled=1
macaddr_acl=1
accept_mac_file=/etc/hostapd/hostapd.accept

# Control
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

# Log
logger_syslog=-1
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=4

#WPA
wpa_passphrase=***
ignore_broadcast_ssid=0
country_code=IT
auth_algs=1
wpa=2
wpa_key_mgmt=SAE
wpa_pairwise=CCMP
rsn_pairwise=CCMP
wpa_group_rekey=1800
ieee80211w=2

#HT
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
# hw ht capab: 0x963
ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# hw vht capab: 0x03c001a2
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
```

this is 88x2bu.conf on AP

`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=1 rtw_switch_usb_mode=1 rtw_dfs_region_domain=1`

this is 88x2bu.conf on client

`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=1 rtw_switch_usb_mode=1`

The network is fast..but after some minutes the AP disappear and all STA are disconnected

`nmcli device wifi list |grep -i router1`

To make it visible again I had to connect via cable to AP and restart hostapd. After 20 seconds the AP return visible

```
nmcli device wifi list |grep -i router1
************  Router1             Infra  36    270 Mbit/s  77      ▂▄▆_  WPA3  
```

I have copied the settings from latext 88x2bu.conf from git.
Any advice? thanks

sateuwdie, 2023-10-30:
The systems are

AP: Slackware 15 kernel 6.5.3
Clients: Slackware 15 and Debian with default kernel.

morrownr, 2023-10-30:
I see a couple of things to check:

> rtw_dfs_region_domain=1

Given your country code of IT, I think the above is wrong. See 88x2bu.conf for documentation. I think you need 3 but please research it.

You are to be using NetworkManager (NM). NM can do what you are describing to hostapd if it is not told to leave it alone.

How to keep Network Manager from causing problems with hostapd

Tell Network Manager to ignore specific devices.

sudo nano /etc/NetworkManager/NetworkManager.conf
add

[keyfile]
unmanaged-devices=interface-name:wlan0
Note: Remember to replace wlan0 with your interface name.

sudo reboot

-----

Tell me if these help.

Cheers

sateuwdie, 2023-10-30:
I have changed rtw_dfs_region_domain=1 to rtw_dfs_region_domain=3 (etsi, Europe)
Now I try in the next days I see if disconnections happens

morrownr, 2023-10-31:
> Now I try in the next days I see if disconnections happens

Did you understand what I recommended with NM? NM is likely what is causing the disconnect.

sateuwdie, 2023-10-31:
Yes.
Another disconnection happen just 2 minutes ago.
Now I'm trying without NM

sateuwdie, 2023-10-31:
I try now without nm, this is wpa_supplicant.conf

```
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=root
update_config=1

network={
ssid="Router1"
proto=RSN
key_mgmt=SAE
ieee80211w=2
sae_password="mypass"
}

But don't authenticate, seems messages are not received on AP side (i cannot see any message even using -dd with hostapd)
```

morrownr, 2023-10-31:
> I try now without nm, this is wpa_supplicant.conf

I realize there is more than one way to set up an AP but...

Take a look at the AP guide I have on the Main Menu:

https://github.com/morrownr/USB-WiFi

See item 9.

Also, I may not have the big picture. How is the internet going into your AP computer?

sateuwdie, 2023-10-31:
> I realize there is more than one way to set up an AP but...
>
> Take a look at the AP guide I have on the Main Menu:
>
> https://github.com/morrownr/USB-WiFi
>
> See item 9.
>
> Also, I may not have the big [...]

Internet via eth0 connected to router.

morrownr, 2023-10-31:
> Internet via eth0 connected to router.

Okay. Internet is coming into your AP system via cabled internet connection, eth0. 

Is your intent to set your AP up as bridged or routed?

My guide is for bridged setups. 

It is not clear to me why you are doing what you are doing with wpa_supplicant. Can you point me to the guide you are using?

sateuwdie, 2023-10-31:
> Okay. Internet is coming into your AP system via cabled internet connection, eth0.
>
> Is your intent to set your AP up as bridged or routed?
>
> My guide is for bridged setups.
>
> It is not clear to me why [...]

Bridged, btw I have had a bssid=MAC:ADDRESS to hostapd
and now seems not disconnecting anymore, but I have to try
with multiple STA connected (actually I have only one STA con
nnected)

sateuwdie, 2023-11-01:
The uptime is over 9 hours with no disconnection, I think probably the problem is fixed using the option BSSID=MAC:AD:RESS in hostapd.conf

closed 2023-11-01 by sateuwdie

sateuwdie, 2023-11-05:
If is helpful for someone actually I don't get disconnection since 2 days (a record).
This are my actually configuration

client /etc/modprobe.d/88x2bu.conf
`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=1 rtw_switch_usb_mode=1`

server /etc/modprobe.d/88x2bu.conf
`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_switch_usb_mode=1`

server /etc/hostapd/hostapd.conf
```

interface=wlan0
bridge=br0
driver=nl80211
ssid=Router1
bssid=MACADRR
hw_mode=a
channel=36	
ieee80211d=1
ieee80211h=1
ieee80211ac=1
ieee80211n=1
wmm_enabled=1
macaddr_acl=1
accept_mac_file=/etc/hostapd/hostapd.accept

# Control
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

# Log
logger_syslog=-1
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=4

#WPA
wpa_passphrase=mypass
ignore_broadcast_ssid=0
country_code=IT
auth_algs=1
wpa=2
wpa_key_mgmt=SAE
wpa_pairwise=CCMP
rsn_pairwise=CCMP
wpa_group_rekey=1800
ieee80211w=2
#sae_groups=19 20 21 25 26
sae_require_mfp=1
#sae_anti_clogging_threshold=10

#HT
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
# hw ht capab: 0x963
ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# hw vht capab: 0x03c001a2
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
```

morrownr, 2023-11-06:
@sateuwdie 

I'm updating my Bridged Wireless AP guide and am wondering if you want to help. The issue is that Raspberry Pi OS 2023-10-10 made Network Manager the default and they removed dhcpcd. I could use some additional eyes looking at the project so if you have any interest, let me know.

sateuwdie, 2023-11-06:
> @sateuwdie
> 
> I'm updating my Bridged Wireless AP guide and am wondering if you want to help. The issue is that Raspberry Pi OS 2023-10-10 made Network Manager the default and they removed dhcpcd. I could use some additional eyes looking at the project so if you have any interest, let me know.

I don't use RaspberryPi atm, I'm using a standard amd pc (ryzen3)

morrownr, 2023-11-06:
> I don't use RaspberryPi atm, I'm using a standard amd pc (ryzen3)

Which I see as an advantage. I know very little about NM at the technical level because I have never needed to learn and now I either need to use it or get rid of it and restore dhcpcd. Pain in the ass so far but I just started.

morrownr, 2024-01-15:
Sounds good. What kind of iperf3 reading are you seeing?

sateuwdie, 2024-01-15:
Iperf3? I don't understand