morrownr/88x2bu-20210702 #184: binary package for rtl88x2bu: 5.13.1 not found (Segmentation fault)

knpshkn, 2023-11-13:
Hello,

I have problem with installing driver and maybe you have any thoughts on what the problem might be
install-driver.sh
```
$ sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20230830
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 2/2 (in-use/total processing units)
: 3599340 (total system memory)
: 5.4.0-166-generic (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
: dkms:2.8
This system doesn't support Secure Boot
This system doesn't support Secure Boot
This system doesn't support Secure Boot
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1

Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source ->
                 /usr/src/rtl88x2bu-5.13.1

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh...................(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.13.1 not found
Error! Bad return status for module build on kernel: 5.4.0-166-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 2.23 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

/var/lib/dkms/rtl88x2bu/5.13.1/build/make.log
```
$ cat /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log
DKMS make.log for rtl88x2bu-5.13.1 for kernel 5.4.0-166-generic (x86_64)
Mon Nov 13 08:28:18 UTC 2023
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-166-generic/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-166-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o
gcc: internal compiler error: Segmentation fault signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
make[2]: *** [scripts/Makefile.build:270: /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o] Error 4
make[2]: *** Waiting for unfinished jobs....
gcc: internal compiler error: Segmentation fault signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
make[2]: *** [scripts/Makefile.build:270: /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o] Error 4
make[1]: *** [Makefile:1778: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-166-generic'
make: *** [Makefile:2523: modules] Error 2
```

make
```
$ make clean
#make -C /lib/modules/5.4.0-166-generic/build M=/home/asd/88x2bu-20210702 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
rm -fr MOK.der MOK.priv

$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-166-generic/build M=/home/asd/88x2bu-20210702  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-166-generic'
  CC [M]  /home/asd/88x2bu-20210702/core/rtw_cmd.o
gcc: internal compiler error: Segmentation fault signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
make[2]: *** [scripts/Makefile.build:270: /home/asd/88x2bu-20210702/core/rtw_cmd.o] Error 4
make[1]: *** [Makefile:1778: /home/asd/88x2bu-20210702] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-166-generic'
make: *** [Makefile:2523: modules] Error 2
```

other
```
$ uname -r
5.4.0-166-generic

$ uname -m
x86_64

$ cat /proc/version
Linux version 5.4.0-166-generic (buildd@lcy02-amd64-011) (gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)) #183-Ubuntu SMP Mon Oct 2 11:28:33 UTC 2023

$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

$ lsusb | grep TP
Bus 002 Device 003: ID 2357:0138 TP-Link 802.11ac NIC
```

Ekiben542, 2023-11-14:
Hmm, why don't you try reinstalling gcc?

Ekiben542, 2023-11-14:
Also, Ubuntu 20.04 should not be recommended. (We recommend using 22.04.)

knpshkn, 2023-11-14:
@Ekiben542 same errors with gcc-10.
we have bunch of mini PCs (dell wyse) with the same OS version (same environment, like gcc and stuff), and for some reason only one of them has this problem

Ekiben542, 2023-11-14:
Do you have kernel headers installed?

knpshkn, 2023-11-15:
@Ekiben542 yes
```
$ sudo apt list linux-headers-$(uname -r)
Listing... Done
linux-headers-5.4.0-166-generic/focal-updates,focal-security,now 5.4.0-166.183 amd64 [installed,automatic]
```

morrownr, 2023-11-17:
> I'm just wondering if there could be such an error due to a faulty USB Wi-Fi dongle

Try compiling without an adapter in a port. I regularly compile the drivers at this site without a supported adapter in a port. Successful compilation does not depend on the adapter being installed.

My first guess would be a problem somewhere in the dev environment. It might be best if you can move on to 22.04 as it might be hard to figure this out.

knpshkn, 2023-11-21:
got it, will try. also will try not to forget to update this issue

nama1arpit, 2023-11-24:
@knpshkn Did you have any progress with this? I also had the exact same error.

knpshkn, 2023-11-24:
@nama1arpit Unfortunately, not yet. I think it will be no earlier than February 2024

jtraut, 2025-03-29:
Just for reference for any others encountering this error, I was receiving the same message while working under constraints of an outdated environment: 
```
user@host:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.4 LTS
Release:	14.04
Codename:	trusty
user@host:~$ uname -r
4.2.0-27-generic
user@host:~$ gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
```
I was finally able to get a working build only after uninstalling `dkms` package, such that the `install-driver.sh` script is forced to use an alternative install method that circumvents the error. 

It is also worth mentioning the new WiFi card (`EWM-W172H01E`) requires BOTH antenna cables to be attached for it to hold consistent stable connections; our manufacturing team had only attached a single antenna before having myself work on it because that is all the previous chipset had required.