morrownr/88x2bu-20210702 #187: Not working under secure boot with dkms installed

GabrielP77, 2023-11-19:
Hello,

I just recently installed Debian 12 with secure boot and after successfully installing the driver through the script installation with dkms my adapter is only recognized by Debian when I disable SecureBoot.

install-driver.sh v20230830
x86_64 (kernel architecture)
x86_64 (architecture to send to gcc)
12/12 (in-use/total processing units)
6.1.0-13-1md64 (kernel version)
gcc-12 (version of gcc used to compile kernel)
dkms-3.0.10
SecureBoot enabled

morrownr, 2023-11-19:
Hi @GabrielP77 

The first section in the file FAQ.md is about Secure Boot. I recommend that you read it and maybe the link to Debian that is in it. I have put a lot of effort into trying to make Secure Boot work for those that want to use it but this is a difficult task because distros have various ways of doing things.

If Secure Boot is important to you and you want a solution that is easier to deal with, I recomment you look at adapters with in-kernel drivers:

https://github.com/morrownr/USB-WiFi

The above is the Main Menu for this site. Recommend you read menu items 1 and 2.

@morrownr

GabrielP77, 2023-11-20:
I tried the manual installation and now it works

Thanks a lot!

morrownr, 2023-11-22:
Glad you got it working.

@morrownr