morrownr/88x2bu-20210702 #189: Why the upload speed is so slow?

sateuwdie, 2023-11-25:
I have a rtl88x2bu card. [This one](https://www.amazon.it/gp/product/B07FCN6WGX)
On Download speed I have no problem, I can reach without disconnections 28MB/s
on upload is a serious problem.
Those are the speed

from client (wifi) to hostapd server using wifi 27MB/s = OK
from client (wifi)  to another client using wifi 5-8MB/s BAD
from client (wifi) to  another client using ethernet 1MB/s VERY BAD

Those are the configuration files on client

`/etc/modprobe.d/88x2bu.conf`
`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 `

on server

`/etc/modprobe.d/88x2bu.conf`
`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_switch_usb_mode=1
`

/etc/hostapd/hostapd.conf

```
interface=wlan0
bridge=br0
driver=nl80211
ssid=myssid
bssid=***
hw_mode=a
channel=36	
ieee80211d=1
ieee80211h=1
ieee80211ac=1
ieee80211n=1
wmm_enabled=1
macaddr_acl=1
accept_mac_file=/etc/hostapd/hostapd.accept

# Control
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

# Log
logger_syslog=-1
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=4

#WPA
wpa_passphrase=********
ignore_broadcast_ssid=0
country_code=IT
auth_algs=1
wpa=2
wpa_key_mgmt=SAE
wpa_pairwise=CCMP
rsn_pairwise=CCMP
wpa_group_rekey=1800
ieee80211w=2
#sae_groups=19 20 21 25 26
sae_require_mfp=1
#sae_anti_clogging_threshold=10

#HT
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
# hw ht capab: 0x963
ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# hw vht capab: 0x03c001a2
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
```

sateuwdie, 2023-11-25:
I forgot: the ethernet client has powerline.

sateuwdie, 2023-11-25:
A "live" example

```
rsync video/films/1.mkv  ethernetclient:
sending incremental file list
1.mkv
         36,08M   9%    2,87MB/s    0:01:53 
```
```
rsync video/films/1.mkv hostapdserverwifi:
sending incremental file list
1.mkv
         87,75M  23%   27,82MB/s    0:00:09
        ....
        370,93M 100%   33,38MB/s    0:00:10 (xfr#1, to-chk=0/1)

```

morrownr, 2023-11-25:
Hi @sateuwdie 

I need clarification on this situation:

You appear to have two systems that are running this driver. A client and one you call a server but I'm thinking the server is an AP.

What I need are the details of the client and AP computers. What hardware is in use and what distro?

sateuwdie, 2023-11-26:
I add now the description.

morrownr, 2023-11-26:
Okay. That helps some. I'd also like to know the cpu type of each platform.

> sae_require_mfp=1

It appears that you are trying to support only WPA3 and not transition mode. If that is the case then the above line is not necessary.

I can't be of specific help with Slackware but you can handle that.

I have a guide that may be of some use with your project, It is menu item 9 on the Main Menu:

https://github.com/morrownr/USB-WiFi

Take a look and let me know how you are progressing.