morrownr/88x2bu-20210702 #190: 88x2bu and RHEL 9.3

sjtmbg, 2023-11-27:
@misha4gps:

@morrownr suggested I contact you about creating a test pull request for a mod for RHEL 9.3.

I added "defined(RHEL92)" in a couple places in ioctl_cfg80211.c that gets it to compile and seems to work.  Let me know if you see this and we can discuss details either here or off-line...

morrownr, 2023-11-28:
@aki-k @sjtmbg 

I might be able to help:

```
cd /usr/lib/modules/6.5.0-0.deb12.1-amd64/kernel/drivers/net/wireless/mediatek/mt76
```

```
ls -l
```

What do you see?

I don't know if RHEL or Rocky include the in-kernel drivers but, if they do, that might be the direction you want to go.

@morrownr

sjtmbg, 2023-11-28:
> @sjtmbg > I added "defined(RHEL92)" in a couple places in ioctl_cfg80211.c
> 
> Could you be a bit more specific? I'm hitting this same compiling problem

Sorry, I opened the issue in order to work on getting a patch in place in github.  As a quick fix this should work for os_dep/linux/ioctl_cfg80211.c:

468c468
< #if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0))
---
> #if ((LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0)) || defined(RHEL92))
487c487
< #if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0))
---
> #if ((LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0)) || defined(RHEL92))

But this might break other RHEL versions so more work needs to be done

misha4gps, 2023-11-29:
I added [PR#192](https://github.com/morrownr/88x2bu-20210702/pull/192) with support rhel9.3
be so kind to test the PR

aki-k, 2023-11-29:
@morrownr Thank you all. Now the TP-Link Archer T3U is working in Rocky Linux 9.3.

Kernel 5.14.0-362.8.1.el9_3.x86_64

closed 2023-12-01 by sjtmbg