morrownr/88x2bu-20210702 #191: 88x2bu and Rocky 9.3

wimein, 2023-11-28:
Compilation on Rocky 9.3 fails (maybe related to issues on RHEL 9.3)

[root@lukas 88x2bu-20210702]# ./install-driver.sh 
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 4/4 (in-use/total processing units)
: 7763268 (total system memory)
: 5.14.0-362.8.1.el9_3.x86_64 (kernel version)
: gcc (GCC) 11.4.1 20230605 (Red Hat 11.4.1-2)
: dkms-3.0.12
EFI variables are not supported on this system
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.
: ---------------------------
Sign command: /lib/modules/5.14.0-362.8.1.el9_3.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh...............(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.0-362.8.1.el9_3.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 98.26 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
[root@lukas 88x2bu-20210702]# more /etc/rocky-release
Rocky Linux release 9.3 (Blue Onyx)
[root@lukas 88x2bu-20210702]# rfkill list all; dkms status; iw reg get
0: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no
rtl88x2bu/5.13.1: added
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

[make.log](https://github.com/morrownr/88x2bu-20210702/files/13491723/make.log)

morrownr, 2023-11-28:
See #190 

I don't so the support for RHEL or anything based on it.

misha4gps, 2023-11-29:
I added [PR#192](https://github.com/morrownr/88x2bu-20210702/pull/192) with support rhel9.3
be so kind to test the PR

wimein, 2023-11-29:
Works with 5.14.0-362.8.1. Thank you very much!

sjtmbg, 2023-11-29:
Successful build and adapter test on:

-   5.14.0-284.30.1.el9_2
-   5.14.0-362.8.1.el9_3
-   4.18.0-477.27.1.el8_8

However, got the same errors as before (ioctl_cfg80211.c) when building for 4.18.0-513.9.1.el8_9.

Added "-DRHEL93" to line 62 of Makefile and built and tested successfully on RHEL8.9, however RHEL93 is misleading in this case so better solution may be to create RHEL89 def?

morrownr, 2023-11-30:
The PR for RHEL89 has been merged. Please test and report.

The README has been updated to reflect the following RHEL support:

RHEL 8.4, 8.6, 8.8, 8.9, 9.1, 9.2 and 9.3 

Thanks go to @misha4gps

sjtmbg, 2023-11-30:
Successfully built and tested 88x2bu driver on Redhat/Rocky 8.8, 8.9, 9.2, and 9.3.

Thanks @misha4gps