morrownr/88x2bu-20210702 #194: Support for bridging?

sateuwdie, 2023-11-30:
The command brctl addif fail

```
brctl addif br0 wlan1
can't add wlan1 to bridge br0: Operation not supported
```

nmcli  command also fail with timeout 
```
nmcli con add ifname br0 type bridge con-name br0
nmcli con add type bridge-slave ifname wlan1 master br0
```

The iw command to "enable" the wlan1 as slave fail

```
iw dev wlan1 set 4addr on
command failed: Operation not supported (-95)
```

Searching on net I found this option for hostapd

```
# WDS (4-address frame) mode with per-station virtual interfaces
# (only supported with driver=nl80211)
# This mode allows associated stations to use 4-address frames to allow layer 2
# bridging to be used.
#wds_sta=1

I have enabled it, restart hostapd and...nothing change.
Fast question: is possible to add support for bridging?
```

morrownr, 2023-11-30:
Hi @sateuwdie 

> Fast question: is possible to add support for bridging?

I have not seen problems with this driver and bridging but my AP guide is likely doing things differently than you are.

This Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu item 9 is the AP guide. It might give you some ideas. I have tested many ways to set up an AP over the years. The guide uses systemd-networkd. If Slackware uses systemd-networks, that may be an option for you.

IT would help me understand if you could start the story from the beginning and tell me what you are trying to accomplish.

> nmcli command also fail with timeout

If setting up an AP, if the distro has NM active, it is the first thing I deactivate.

> iw dev wlan1 set 4addr on

I'm pretty sure these Realtek out-of-kernel drivers do not support 4addr but I would have to take a look.

> the commands fail on a wifi client, the bridging in server hostapd works.

It is not clear to me why you would execute the commands on the client.

> brctl addif br0 wlan1

I haven't used this tool for years. It is not clear to me why it would be needed but then I'm not sure exactly what you are trying to do.

@morrownr

sateuwdie, 2023-12-01:
Hi, I need a bridge on client because I use virtual machines, I also know is possible to route to wlan0 the vm's network interfaces, but I prefer bridge (I can use the same ip network on vm and host).
The commands nmcli, brctl and iw are all run on client, not server.