morrownr/88x2bu-20210702 #196: Ubuntu 23.10 Secure Boot?

Amondale, 2023-12-04:
Can you recommend any workarounds when sign-file "mandatory" package isn't found on my system?

sudo modprobe 88x2bu
modprobe: ERROR: could not insert '88x2bu': Key was rejected by service

sudo apt install sign-file
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package sign-file

Amondale, 2023-12-05:
Not certain, but probably did have secure boot turned on. I am wondering if there is a way to rip out all trace of this driver, hoping that this usb adapter is supported natively, like Ubuntu 22.04.3 (modprobe finds 88x2bu) after insertion with no driver loaded from this github site.

morrownr, 2023-12-05:
> Not certain, but probably did have secure boot turned on. 

If it is not turned on when installing the os, things won't go well for things like installing this driver. Well, it will install but it won't work. 

> I am wondering if there is a way to rip out all trace of this driver,

$ sudo sh remove-driver.sh

The in-kernel driver and this driver are not feature equal. They both have features the other does not so you just have to see which works better for you. I recommend using at least kernel 6.4 for the in-kernel driver.

morrownr, 2023-12-05:
> If it is not turned on when installing the os, things won't go well for things like installing this driver. Well, it will install but it won't work.

To test, simple go into the bios and turn secure boot off and reboot. If the driver works, then you know if it is a secure boot setup problem.