morrownr/88x2bu-20210702 #199: Does not compile on 6.2.0-37-generic on Ubuntu 22.04 - running OK on 6.2.0-34-generic

elpraga, 2023-12-10:
I've just run into a problem. I have been using this module for some time, and everything had beed going OK until today when I upgraded kernel to 6.2.0-37-generic.

Without any other change, the module does not compile.

```
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/rtw_android.o
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.c: In function ‘rtw_regd_init’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.c:409:36: error: ‘REGULATORY_IGNORE_STALE_KICKOFF’ undeclared (first use in this function)
  409 |         wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.c:409:36: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/rtw_proc.o
make[2]: *** [scripts/Makefile.build:260: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2026: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-37-generic'
make: *** [Makefile:2506: modules] Error 2
```

This seems to be the relevant part of `/var/lib/dkms/rtl88x2bu/5.13.1/build/make.log`

Any ideas how to fix this issue?

morrownr, 2023-12-10:
Hi @elpraga 

I seem to remember this problem from maybe a year ago. I would have to dig in to refresh my memory but try this first:

$ git pull
$ sudo sh install-driver.sh

elpraga, 2023-12-10:
Thank you for your quick reply @morrownr !  I had run the script again before opening this issue. 

Now, I have tried what you have suggested (including the `git pull`), but the result remains the same.

It still does finishes compiling on kernel **6.2.0-34-generic** but fails with the same error on **6.2.0-37-generic**. 

I had been running this driver for quite some time on the same machine, and I had never had problems before. Everything had run smoothly via `dkms`. 

I am absolutely baffled, and I'd appreciate any help you can give me!

Thank you!

morrownr, 2023-12-11:
> I am absolutely baffled

We'll figure this out. I'm getting more reports in. Remember that I have 6 of these Realtek out-of-kernel drivers up here and the additional reports are giving some hints as I think what I am seeing is related somehow.

I'm seeing kernel 6.1.65 was working fine and now 6.1.66 is broken. That may indicate a backport is breaking things and since Ubuntu is using kernel 6.2.x, it could be that the Ubuntu devs picked and included the same bad backport. If I am correct, we are going to see a flood of reports this week as distros upgrade. If I can figure out which backport caused this, I can probably figure out a fix in the driver... maybe. 

One way or another, this may not be a quick fix so running that previously working kernel is likely your best bet for now.

@morrownr

elpraga, 2023-12-11:
It's probably unrelated  @morrownr 

https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.6.6

morrownr, 2023-12-12:
We'll see. You should get a new kernel flowing in soon. Tell me what happens,

elpraga, 2023-12-15:
I've installed the 6.2.0-39 kernel, and I still cannot compile the driver. I'm getting the same error as with 6.2.0.-37

morrownr, 2023-12-15:
I'll try to set aside some time to look at this something this afternoon.

morrownr, 2023-12-16:
@elpraga 

I have now accomplished numerous installations on Ubuntu, Debian 12 and the most recent RasPiOS. I've tested on kernels 6.1, 6.5, 6.6 and 6.7. I am not seeing the problem. My to-do list is long and sometimes I have to offer alternatives due to the time I have available.

In this case, you can go back to the last known working kernel while waiting for future upgrades that may fix the problem or you can see if your distro has the hardware enablement version of kernel 6.5 available and upgrade to it.

I hope you understand.

@morrownr

elpraga, 2023-12-17:
@morrownr Thank you for your time and your help!

Did I understand correctly that you did not test the kennel I've actually reported the issue on?  I was having issue with the latest hwe kennel (6.2.0-39 and also with 37), and I noticed that you did not mention kennel 6.2. 

I just want to double check if I should be checking for some file corruption issue it something similar.

In any case, if you were able to compile the driver successfully on later kernels, I should be able to compile it again on the future hwe kennel from 24.04, so I should not be stuck on the same kernel forever.

Thank you so much for looking into the issue!

morrownr, 2023-12-18:
> Did I understand correctly that you did not test the kennel I've actually reported the issue on? 

That is correct. I currently do not have Ubuntu 22.04 installed on my test systems. I do have Ubuntu 23.10 installed. I tested several kernels on multiple distros and platforms. It is best to work problems with the exact distro in use but there hundreds of distros and most have more than one version that is currently supported so it is impossible for me to support every distro that is out there. I have to make choices based on my experience that allow me to solve as many problems as I can.

This repo averages about 250 clones per day. I will be watching issues. You might do some reading over at the Ubuntu forums to see if anything interesting is being discussed. For now I have a long to-do list for issues that I do know how to fix or enhance so I will be working on those until more info is available. Your best option for now is probably to boot back into the last working setup you had.

Cheers,

@morrownr

elpraga, 2024-01-13:
Hello @morrownr 

I have managed to install a 6.5 kernel on my Ubuntu 22.04 and I have tried to compile the driver on this kernel. 

I know that you have managed to compile the module on this kernel, but I am getting this error. 

```
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8592:6: note: declared here
 8592 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:483:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  483 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8575:6: note: declared here
 8575 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:251: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.c: In function ‘rtw_regd_init’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.c:409:36: error: ‘REGULATORY_IGNORE_STALE_KICKOFF’ undeclared (first use in this function)
  409 |         wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.c:409:36: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [scripts/Makefile.build:251: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.5.0-14-generic/Makefile:2037: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-14-generic'
make: *** [Makefile:2506: modules] Error 2
```

Fortunetely (for me), the integrated Wi-Fi started working, but I did want to let you know about this problem, because I loved the dongle when it was working, and it would be great to have it working for everybody.

morrownr, 2024-01-13:
@elpraga 

I have been looking at this today. The errors you are seeing can be explained by kernel API changes. Those changes have been made to the driver as it is in the repo. The changes are actually fairly old. I have tested the driver on an Ubuntu system with kernal 6.5. The driver compiles and works well.

My conclusion is that the driver code you have on your system is old. My recommendation is for you to do the following:

$ sudo sh remove-driver.sh
delete the driver directory
run the installation steps like it is a fresh instrallation

FYI: you can keep the drover code fresh by running the following:

$ git pull
$ sudo sh install-driver.sh

Let me know.

@morrownr

elpraga, 2024-01-13:
Thank you for your answer, @morrownr 

I'd be surprised if the code in my system was old, as I remember pulling the driver code from scratch when first reporting this issue. 

Nevertheless, it's possible I'm misremembering things, I'll follow the steps you've suggested, and I'll report the results.

Thank you for your help!

elpraga, 2024-01-13:
@morrownr I've just followed the steps you have suggested and this is what I've got:

```
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# ./dkms-make.sh 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.5.0-1011-oem/build M=/home/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702  modules
make[1]: Entering directory '/usr/src/linux-headers-6.5.0-1011-oem'
make[2]: *** No rule to make target 'Archer'.  Stop.
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-1011-oem'
make: *** [Makefile:2522: modules] Error 2
```

elpraga, 2024-01-13:
@morrownr When I rebooted my computer, I have noticed that even though the driver did not compile and `dkms status` does not list the driver as installed, the driver works out of the box with the **6.5.0-1011-oem** kernel. It does appear to be slower than before, but it does work out of the box.

```
root@HeroBox:~# dkms status
rtl8821ce/v5.5.2_34066.20200325, 6.2.0-34-generic, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-1009-oem, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-1011-oem, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-14-generic, x86_64: installed
rtl8821cu/5.12.0.4, 6.2.0-34-generic, x86_64: installed
rtl88x2bu/5.13.1: added
root@HeroBox:~# 
```

morrownr, 2024-01-13:
> root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# ./dkms-make.sh 

The above is not in my installation steps.

The last installation step is as follows:

$ sudo sh install-driver.sh

>  ./dkms-make.sh 

That script is not a script you should run as it is designed to be run by another script. There is nowhere in my installation steps that says to run it.

elpraga, 2024-01-13:
I'm sorry, @morrownr I did run the script first, and I got en error. Let me go through the whole process again (including removing `sh remove-driver.sh`)

elpraga, 2024-01-13:
I'm back. I've run `sudo sh remove-driver.sh`, `reboot` 

```
rm -rf 88x2bu-20210702/
git clone https://github.com/morrownr/88x2bu-20210702.git
cd 88x2bu-20210702/
git pull
sudo sh install-driver.sh
```

And I've got:

```
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# sudo sh install-driver.sh

: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 7863276 (total system memory)
: 6.5.0-1011-oem (kernel version)
: gcc-12 (version of gcc used to compile the kernel)
: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
: dkms-2.8.7
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers...
--> rtl88x2bu 5.13.1: added
Error! The module/version combo: rtl88x2bu-5.13.1: is not located in the DKMS tree.
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
Error! DKMS tree already contains: rtl88x2bu-5.13.1
You cannot add the same module/version combo more than once.
This driver may already be installed.
Run the following and then reattempt installation.
$ sudo ./remove-driver.sh
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# 
```
Note that `dkms status` says:
```root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# dkms status
rtl8821ce/v5.5.2_34066.20200325, 6.2.0-34-generic, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-1009-oem, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-1011-oem, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-14-generic, x86_64: installed
rtl8821cu/5.12.0.4, 6.2.0-34-generic, x86_64: installed
rtl88x2bu/5.13.1: added
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# 
```

I've tried to remove the driver, although it does not seem to be installed via `dkms`

```
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# dkms remove 
Error! Arguments <module> and <module-version> are not specified.
Usage: remove <module>/<module-version> or
       remove -m <module>/<module-version> or
       remove -m <module> -v <module-version>
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# dkms remove rtl88x2bu
Error! Arguments <module> and <module-version> are not specified.
Usage: remove <module>/<module-version> or
       remove -m <module>/<module-version> or
       remove -m <module> -v <module-version>
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# dkms remove rtl88x2bu/5.13.1
Error! There is no instance of rtl88x2bu 5.13.1 for kernel 6.5.0-1011-oem (x86_64) located in the DKMS tree.
root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702#
```

morrownr, 2024-01-13:
> root@HeroBox:~/INSTALL/tp-link Archer T3U Nano AC1300/88x2bu-20210702# dkms remove rtl88x2bu/5.13.1

You made a heck of an effort but try this:

$ dkms remove -m "88x2bu" -v "5.13.1" --all

That it was removed automatically is a bug in the script.

elpraga, 2024-01-13:
I've run `dkms remove rtl88x2bu/5.13.1 --all` and then followed the steps you have mentioned earlier.

I've successfully compieled the module on my `6.5.0-1011-oem` kernel. 

Unforntunately, I cannot confirm if simply running
```
$ git pull
$ sudo sh install-driver.sh
```

in the driver's original directory updates the drives as expected after a breakdown of automatic dmks compilation (I've done too many other steps in between) 

I can confirm though that the driver does compile without an error on the following kernels:

```
root@HeroBox:~# dkms status
rtl8821ce/v5.5.2_34066.20200325, 5.19.0-50-generic, x86_64: installed (WARNING! Diff between built and installed module!)
rtl8821ce/v5.5.2_34066.20200325, 6.2.0-34-generic, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.2.0-39-generic, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-1009-oem, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-1011-oem, x86_64: installed
rtl8821ce/v5.5.2_34066.20200325, 6.5.0-14-generic, x86_64: installed
rtl8821cu/5.12.0.4, 5.19.0-50-generic, x86_64: installed (WARNING! Diff between built and installed module!)
rtl8821cu/5.12.0.4, 6.2.0-34-generic, x86_64: installed
rtl8821cu/5.12.0.4, 6.2.0-39-generic, x86_64: installed
rtl88x2bu/5.13.1, 6.2.0-34-generic, x86_64: installed
rtl88x2bu/5.13.1, 6.2.0-39-generic, x86_64: installed
rtl88x2bu/5.13.1, 6.5.0-1009-oem, x86_64: installed
rtl88x2bu/5.13.1, 6.5.0-1011-oem, x86_64: installed
rtl88x2bu/5.13.1, 6.5.0-14-generic, x86_64: installed
root@HeroBox:~# 
```

Thank you so much for your help!! I honestly do appreciate it! Even thought all my thoubles were probably caused by not knowing what exactly to do after the automatic dkms install fails.

I'm sorry for all the confusion I have caused and thank you again for your helpful support!

closed 2024-01-13 by elpraga

morrownr, 2024-01-13:
You are seeing a driver when this driver is removed because the kernel 6.5 has an in-kernel driver for this chipset. Remember to run remove-driver.sh when removing this driver as it will delete the blacklist of the in-kernel driver. 

This driver is a little faster in my testing but seems to work well. The in-kernel driver needs more users so as to get bug reports.

@morrownr