morrownr/88x2bu-20210702 #200: Not working with Slackware-current and kernel 6.1.66

Jesus-M, 2023-12-10:
No compilation errors, apparently the module is installed:
```
root@liet:[88x2bu-20210702]# dkms status
rtl88x2bu/5.13.1, 6.1.66, x86_64: installed
```
But I have not been able to use it. Any network operation takes forever, until it times out, the interface will not acquire IP address from the DHCP server, and when I try to add a static address, the 'ifconfig' command never completes. I cannot even interrupt it with 'kill' or 'kill -9'.

Installed back kernel 6.1.65 and everything goes well.

morrownr, 2023-12-10:
Hi @Jesus-M 

I also saw a report from an Ubuntu user this morning and it sounded like this. He was using kernel 6.2.x.

This sounds like a backport that was just applied to kernel 6.1.66 and the Ubuntu kernel 6.2.x is the problem. Now to figure out what backports were applied to kernel 6.1.66 and figure out which one is the problem. Wanna help?

Jesus-M, 2023-12-10:
Sure, happy to help.

But you know, I'm not a developer, so I can help if you tell me what to do and what info you need from me.

//JM

morrownr, 2023-12-10:
> But you know, I'm not a developer, 

Sometimes I wonder if I am a developer. I learned programming back in the dark ages with FORTRAN on an IBM Mainframe. I have a couple of C reference books and I just dig in.

I don't have time to look at this today and all indications are that this is going to blow up over the next week as 6.1.66 finds its way into more distros. I am having a senor moment in that I forget the best place to find the backports that went into 6.1.66 so if you could do some research and find what backports went into 6.1.66, that would be a really good start and a big help.

lgair, 2023-12-11:
hi, I am having basically the same issue on an older kernel, is it possible for you to run dmesg on your system after tring to insert the module? Im curious to see if it errors out in the same way

avallach2000, 2023-12-12:
The same problem occurs in Debian 12 (bookworm). A couple days ago they updated the default kernel:
linux-image-6.1.0-13-amd64 (6.1.**55**-1) -> linux-image-6.1.0-15-amd64 (6.1.**66**-1)

Here is what I got after decode_stacktrace.sh:
```
Dec 12 17:02:57 sora kernel: usb 1-4: new high-speed USB device number 5 using xhci_hcd
Dec 12 17:02:57 sora kernel: usb 1-4: New USB device found, idVendor=0b05, idProduct=19aa, bcdDevice= 2.10
Dec 12 17:02:57 sora kernel: usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 12 17:02:57 sora kernel: usb 1-4: Product: ASUS USB-AC58 USB Wireless adapter
Dec 12 17:02:57 sora kernel: usb 1-4: Manufacturer: Asus
Dec 12 17:02:57 sora kernel: usb 1-4: SerialNumber: 123
Dec 12 17:02:57 sora kernel: cfg80211: Loading compiled-in X.509 certificates for regulatory database
Dec 12 17:02:57 sora kernel: cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
Dec 12 17:02:57 sora kernel: cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
Dec 12 17:02:57 sora kernel: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
Dec 12 17:02:57 sora kernel: platform regulatory.0: firmware: direct-loading firmware regulatory.db
Dec 12 17:02:57 sora kernel: platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
Dec 12 17:02:58 sora kernel: usbcore: registered new interface driver rtl88x2bu
Dec 12 17:02:59 sora kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Dec 12 17:03:04 sora kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Dec 12 17:05:34 sora kernel: INFO: task wpa_supplicant:1019 blocked for more than 120 seconds.
Dec 12 17:05:34 sora kernel:       Tainted: P           OE      6.1.0-15-amd64 #1 Debian 6.1.66-1
Dec 12 17:05:34 sora kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Dec 12 17:05:34 sora kernel: task:wpa_supplicant  state:D stack:0     pid:1019  ppid:1      flags:0x00004002
Dec 12 17:05:34 sora kernel: Call Trace:
Dec 12 17:05:34 sora kernel:  <TASK>
Dec 12 17:05:34 sora kernel: __schedule+0x34d/0x9e0 
Dec 12 17:05:34 sora kernel: schedule+0x5a/0xd0 
Dec 12 17:05:34 sora kernel: schedule_preempt_disabled+0x11/0x20 
Dec 12 17:05:34 sora kernel: __mutex_lock.constprop.0+0x399/0x700 
Dec 12 17:05:34 sora kernel: ? __nla_reserve+0x38/0x50 
Dec 12 17:05:34 sora kernel: nl80211_send_iface (debian/build/build_amd64_none_amd64/net/wireless/nl80211.c:3822) cfg80211
Dec 12 17:05:34 sora kernel: ? kmalloc_reserve+0x4b/0xa0 
Dec 12 17:05:34 sora kernel: ? __alloc_skb+0xd9/0x1a0 
Dec 12 17:05:34 sora kernel: nl80211_get_interface (debian/build/build_amd64_none_amd64/net/wireless/nl80211.c:3986) cfg80211
Dec 12 17:05:34 sora kernel: genl_family_rcv_msg_doit+0xfd/0x160 
Dec 12 17:05:34 sora kernel: genl_rcv_msg+0x122/0x250 
Dec 12 17:05:34 sora kernel: ? nl80211_dump_interface (debian/build/build_amd64_none_amd64/net/wireless/nl80211.c:3977) cfg80211
Dec 12 17:05:34 sora kernel: ? nl80211_send_iface (debian/build/build_amd64_none_amd64/net/wireless/nl80211.c:3902) cfg80211
Dec 12 17:05:34 sora kernel: ? genl_start+0x160/0x160 
Dec 12 17:05:34 sora kernel: netlink_rcv_skb+0x4e/0x100 
Dec 12 17:05:34 sora kernel: genl_rcv+0x24/0x40 
Dec 12 17:05:34 sora kernel: netlink_unicast+0x247/0x390 
Dec 12 17:05:34 sora kernel: netlink_sendmsg+0x250/0x4c0 
Dec 12 17:05:34 sora kernel: __sock_sendmsg+0x5c/0x70 
Dec 12 17:05:34 sora kernel: ____sys_sendmsg+0x277/0x2f0 
Dec 12 17:05:34 sora kernel: ? copy_msghdr_from_user+0x7d/0xc0 
Dec 12 17:05:34 sora kernel: ___sys_sendmsg+0x9a/0xe0 
Dec 12 17:05:34 sora kernel: __sys_sendmsg+0x76/0xc0 
Dec 12 17:05:34 sora kernel: do_syscall_64+0x58/0xc0 
Dec 12 17:05:34 sora kernel: ? syscall_exit_to_user_mode+0x27/0x40 
Dec 12 17:05:34 sora kernel: ? do_syscall_64+0x67/0xc0 
Dec 12 17:05:34 sora kernel: ? do_syscall_64+0x67/0xc0 
Dec 12 17:05:34 sora kernel: ? do_syscall_64+0x67/0xc0 
Dec 12 17:05:34 sora kernel: entry_SYSCALL_64_after_hwframe+0x64/0xce 
Dec 12 17:05:34 sora kernel: RIP: 0033:0x7fb830d29930
Dec 12 17:05:34 sora kernel: RSP: 002b:00007ffd84295bc8 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
Dec 12 17:05:34 sora kernel: RAX: ffffffffffffffda RBX: 000055f4af79f960 RCX: 00007fb830d29930
Dec 12 17:05:34 sora kernel: RDX: 0000000000000000 RSI: 00007ffd84295c00 RDI: 0000000000000006
Dec 12 17:05:34 sora kernel: RBP: 000055f4af82d5a0 R08: 0000000000000004 R09: 00007fb830df1d00
Dec 12 17:05:34 sora kernel: R10: 00007ffd84295ce4 R11: 0000000000000202 R12: 000055f4af79fd00
Dec 12 17:05:34 sora kernel: R13: 00007ffd84295c00 R14: 000055f4ae993a40 R15: 00007ffd84295ce4
Dec 12 17:05:34 sora kernel:  </TASK>
```

lgair, 2023-12-12:
Thanks @morrownr!
Im still trying to figure out if this issue is indeed the same as mine, since the kernel Im using should technically be independant, but Im not sure since im still very very green at kernel dev work. but the behaviour that @Jesus-M described is the same behaviour I am seeing.

morrownr, 2023-12-12:
This bug, caused by the kernel wifi stack mess up, can manifest in different ways depending on your distro. If your distro uses IWD or wpa_supplicant, it may show differently. If your distro is using NetworkManager or something else, you may see different problems. This is a wifi stack problem and it will affect many parts of the wifi system depending on what is used in your system.

I simply cannot chase bugs until this problem is cleaned up. If you need a stable system to get work done, boot into the last stable kernel and worry about this when a new kernel has flowed into your system.

Regards,

@morrownr