morrownr/88x2bu-20210702 #202: AP mode on OpenWrt latest snapshot master only worked a few mins then died

fakemanhk, 2024-01-14:
I am trying a China no name AC1200 USB dongle (USB ID 0bda:b812 which I believe it's 88x2bu) on my Raspberry Pi running latest snapshot master  r24806-735efbfb7c (kernel 6.1.71). I created an 5GHz AP on it, my phone tried to connect but failed a few times (but OpenWrt reflected that my phone had tried to connect), after a short moement  I observed a few lines of `timed out to flush queue` in system log, then the AP is completely down, any idea what's happening?

Here is the log:
```
Mon Jan 15 01:55:42 2024 daemon.notice hostapd: Set new config for phy phy2:
Mon Jan 15 01:55:42 2024 daemon.notice hostapd: Remove interface 'phy2'
Mon Jan 15 01:55:42 2024 daemon.notice hostapd: phy2-ap0: interface state ENABLED->DISABLED
Mon Jan 15 01:55:42 2024 daemon.notice hostapd: phy2-ap0: AP-DISABLED
Mon Jan 15 01:55:42 2024 daemon.notice hostapd: phy2-ap0: CTRL-EVENT-TERMINATING
Mon Jan 15 01:55:42 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Mon Jan 15 01:55:42 2024 daemon.notice hostapd: nl80211: deinit ifname=phy2-ap0 disabled_11b_rates=0
Mon Jan 15 01:55:42 2024 kern.info kernel: [ 1019.043072] device phy2-ap0 left promiscuous mode
Mon Jan 15 01:55:42 2024 kern.info kernel: [ 1019.048740] br-lan: port 2(phy2-ap0) entered disabled state
Mon Jan 15 01:55:42 2024 daemon.notice netifd: Network device 'phy2-ap0' link is down
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.332016] rtw_8822bu 1-1.4:1.0: timed out to flush queue 3
Mon Jan 15 01:55:43 2024 daemon.notice wpa_supplicant[1071]: Set new config for phy phy2
Mon Jan 15 01:55:43 2024 daemon.notice netifd: Wireless device 'radio2' is now down
Mon Jan 15 01:55:43 2024 daemon.notice netifd: radio2 (6187): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jan 15 01:55:43 2024 daemon.notice hostapd: Set new config for phy phy2:
Mon Jan 15 01:55:43 2024 daemon.notice wpa_supplicant[1071]: Set new config for phy phy2
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.747407] ------------[ cut here ]------------
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.752980] write RF mode table fail
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.757474] WARNING: CPU: 3 PID: 6207 at backports-6.5/drivers/net/wireless/realtek/rtw88/rtw8822b.c:823 0xffffffc00100808c
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.770430] Modules linked in: brcmfmac_wcc xt_connlimit vport_vxlan rtw88_8822cu rtw88_8822c rtw88_8822bu rtw88_8822b rtw88_8821cu rtw88_8821c rtl_usb rt2800usb rt2800lib pppoe ppp_async openvswitch nft_fib_inet nf_flow_table_inet nf_conncount mt7921u mt7921_common mt76x0u mt76x0_common brcmfmac xt_state xt_helper xt_conntrack xt_connmark xt_connbytes xt_CT rtw88_usb rtw88_core rtlwifi rtl8xxxu rtl8812au rt2x00usb rt2x00lib pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt792x_usb mt792x_lib mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76_connac_lib mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ip_tables cfg80211 xt_time xt_tcpudp xt_recent xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables usbhid ums_usbat
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.770811]  ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc r8152 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon hid_generic crc_ccitt compat cdc_ncm cdc_ether brcmutil snd_bcm2835(C) hid evdev veth snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore mpls_iptunnel mpls_router ip_tunnel mpls_gso autofs4 nls_utf8 vxlan udp_tunnel ip6_udp_tunnel nsh cmac nls_iso8859_1 nls_cp437 realtek vfat fat
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.928330] CPU: 3 PID: 6207 Comm: iw Tainted: G        WC         6.1.71 #0
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.936461] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.943352] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.951376] pc : 0xffffffc00100808c
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.955885] lr : 0xffffffc00100808c
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.960355] sp : ffffffc01035b7e0
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.964615] x29: ffffffc01035b7e0 x28: ffffff8007fcc800 x27: ffffffc01035bdc8
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.972715] x26: 0000000000000003 x25: 0000000000000024 x24: ffffffc00100a268
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.980797] x23: 0000000000000003 x22: ffffffc00100a010 x21: 0000000000000000
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.988858] x20: 00000000000aeaea x19: ffffff8006412080 x18: 000000000000024d
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1019.996889] x17: 0000000000000000 x16: 0000000000000000 x15: ffffffc008b19e50
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.004904] x14: 00000000000006e7 x13: 000000000000024d x12: 00000000ffffffea
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.012895] x11: 00000000ffffefff x10: ffffffc008b71e50 x9 : ffffffc008b19df8
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.020868] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.028828] x5 : ffffff80371d3780 x4 : 0000000000000000 x3 : 0000000000000027
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.036764] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000018
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.044672] Call trace:
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.047843]  0xffffffc00100808c
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.051687]  0xffffffc00100848c
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.055478]  rtw_core_enable_beacon+0x2ac/0x1a70 [rtw88_core]
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.061903]  ieee80211_nan_func_match+0x22ac/0x39b4 [mac80211]
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.068444]  nl80211_parse_chandef+0x1670/0x30f0 [cfg80211]
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.074684]  genl_family_rcv_msg_doit+0xb8/0x11c
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.079937]  genl_rcv_msg+0x108/0x230
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.084199]  netlink_rcv_skb+0x5c/0x130
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.088647]  genl_rcv+0x38/0x50
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.092406]  netlink_unicast+0x1e8/0x2d4
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.096939]  netlink_sendmsg+0x1a0/0x3d0
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.101455]  ____sys_sendmsg+0x238/0x2a4
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.105956]  ___sys_sendmsg+0x7c/0xc0
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.110194]  __sys_sendmsg+0x48/0xb0
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.114332]  __arm64_sys_sendmsg+0x24/0x30
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.118993]  invoke_syscall.constprop.0+0x5c/0x104
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.124347]  do_el0_svc+0x50/0x13c
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.128294]  el0_svc+0x18/0x54
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.131883]  el0t_64_sync_handler+0xf4/0x120
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.136691]  el0t_64_sync+0x174/0x178
Mon Jan 15 01:55:43 2024 kern.warn kernel: [ 1020.140857] ---[ end trace 0000000000000000 ]---
Mon Jan 15 01:55:43 2024 daemon.notice wpa_supplicant[1071]: Set new config for phy phy2
Mon Jan 15 01:55:44 2024 daemon.notice hostapd: Set new config for phy phy2: /var/run/hostapd-phy2.conf
Mon Jan 15 01:55:44 2024 daemon.notice hostapd: Restart interface for phy phy2
Mon Jan 15 01:55:44 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU- [... cut ...]
Mon Jan 15 01:55:44 2024 daemon.err hostapd: nl80211: Could not set interface 'phy2-ap0' UP
Mon Jan 15 01:55:44 2024 daemon.notice hostapd: nl80211: deinit ifname=phy2-ap0 disabled_11b_rates=0
Mon Jan 15 01:55:44 2024 daemon.err hostapd: nl80211 driver initialization failed.
Mon Jan 15 01:55:44 2024 daemon.notice hostapd: phy2-ap0: CTRL-EVENT-TERMINATING
Mon Jan 15 01:55:44 2024 daemon.err hostapd: hostapd_free_hapd_data: Interface phy2-ap0 wasn't started
Mon Jan 15 01:55:44 2024 daemon.notice netifd: Wireless device 'radio2' is now up
```

henkv1, 2024-01-17:
It looks like you are using the in kernel driver rtw88. AP mode does not work well with this driver. See https://github.com/lwfinger/rtw88/issues/151.
The driver in this repository is the out of kernel Realtek driver. AP mode works pretty well with this driver, but unfortunately it is incompatible with OpenWrt.

fakemanhk, 2024-01-17:
> It looks like you are using the in kernel driver rtw88. AP mode does not work well with this driver. See [lwfinger/rtw88#151](https://github.com/lwfinger/rtw88/issues/151). The driver in this repository is the out of kernel Realtek driver. AP mode works pretty well with this driver, but unfortunately it is incompatible with OpenWrt.

Oh I saw that issue before I posted this, so in kernel RTW88 is mainly for client purpose? How about RTW89? Is it the same status?

morrownr, 2024-01-19:
@fakemanhk 

> so in kernel RTW88 is mainly for client purpose?

No, it is intended to be a full feature driver. The USB support is not written by or maintained by Realtek and there are bugs.

> How about RTW89?

It has no USB support.

@morrownr