morrownr/88x2bu-20210702 #203: If more than one arch is specified on the command line.... MX Linux 23.1

linuxgameconsortium, 2024-01-16:
Apparently "more than one arch is specified on the command line". What!?

Description:	Debian GNU/Linux 12 (bookworm)
Linux version 6.1.0-17-amd64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30)

@morrownr I appreciate that the driver, but yeah, any help would be great. 

```
$ sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 8130292 (total system memory)
: 6.1.0-17-amd64 (kernel version)
: gcc-12 (version of gcc used to compile the kernel)
: gcc (Debian 12.2.0-14) 12.2.0
: /usr/sbin/dkms.mx --version
dkms-3.0.10
: This system doesn't support Secure Boot
: ---------------------------

Checking for previously installed drivers...
Deprecated feature: MODULES_CONF (/var/lib/dkms/rtl8821ce/5.5.2.1+git20230504/source/dkms.conf)
Deprecated feature: MODULES_CONF (/var/lib/dkms/rtl8821ce/5.5.2.1+git20230504/source/dkms.conf)
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
/usr/sbin/dkms.mx add -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.
: ---------------------------

/usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-13-amd64 -k 6.1.0-17-amd64 -a x86_64
Error!  If more than one arch is specified on the command line, then there
must be an equal number of kernel versions also specified (1:1 relationship).
An error occurred. dkms build error:  1
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

morrownr, 2024-01-18:
Hi @linuxgameconsortium 

I'm seeing some things that are hard to explain. Let's start with:

$ dkms status

Post the results. Let's see what else is on your system that uses dkms.

@morrownr

linuxgameconsortium, 2024-01-18:
Thanks for the reply. 

I've already run `sudo dkms remove rtl88x2bu/5.13.1 --all` to remove the driver. 

if I remove `-a x86_64` from the command to read, `/usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-13-amd64 -k 6.1.0-17-amd64 -a x86_64`, then I can load the module into dkms. But that's it, the rest of the `install-driver.sh` process does not complete.

```
$ dkms status
/usr/sbin/dkms.mx status
Deprecated feature: MODULES_CONF (/var/lib/dkms/rtl8821ce/5.5.2.1+git20230504/source/dkms.conf)
Deprecated feature: MODULES_CONF (/var/lib/dkms/rtl8821ce/5.5.2.1+git20230504/source/dkms.conf)
8812au/5.13.6, 6.1.0-13-amd64, x86_64: installed
8812au/5.13.6, 6.1.0-17-amd64, x86_64: installed
broadcom-sta/6.30.223.271, 6.1.0-13-amd64, x86_64: installed
broadcom-sta/6.30.223.271, 6.1.0-17-amd64, x86_64: installed
nvidia-current/525.147.05, 6.1.0-13-amd64, x86_64: installed
nvidia-current/525.147.05, 6.1.0-17-amd64, x86_64: installed
rtl8821ce/5.5.2.1+git20230504, 6.1.0-13-amd64, x86_64: installed
rtl8821ce/5.5.2.1+git20230504, 6.1.0-17-amd64, x86_64: installed
rtl8821cu/5.12.0, 6.1.0-13-amd64, x86_64: installed
rtl8821cu/5.12.0, 6.1.0-17-amd64, x86_64: installed

```

mshpp, 2024-01-20:
I'm having the exact same issue, also on MX-23 and with the -17 kernel but with the amdgpu DKMS module. OP, I think the problem could be in this particular distro/kernel constellation.

morrownr, 2024-01-22:
I've been looking this over but am not sure what would cause this. I am aware of bugs that are caused by various distros and even try to correct some in the scripts but this is not something I have seen and I've been at this for a few years. My ability to help will depend on you guys helping.

@morrownr

john9631, 2024-02-20:
I had the same error (I'm running 6.1.0-17 but others are present).

The script failed after running
`/usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-13-amd64 -k 6.1.0-16-amd64 -k 6.1.0-17-amd64 -a x86_64
`

So I removed the references to -13 and -16 to create a line that just addressed 6.1.0-17 which ran fine (directly after the crashed script):
` sudo /usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-17-amd64 -a x86_64`

Going to reboot & confirm that it did the work.

john9631, 2024-02-21:
I had the same problem with mx linux (debian derived).  The second part of the build and install commands picked up 13, 16 and 17 whereas the first part picked up 17.  I only needed 17 so I took the failing command, edited out the extra 2 and patched the install script.  FWIW my version follows - you need to fail & correct for your version:

At line 362

```
	if command -v /usr/bin/time >/dev/null 2>&1; then
		# /usr/bin/time -f "Compile time: %U seconds" dkms build -m ${DRV_NAME} -v ${DRV_VERSION} -k "${KVER}/${KARCH}" -c "/usr/src/${DRV_NAME}-${DRV_VERSION}/dkms.conf" --force
        /usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-17-amd64 -a x86_64
	else
		# dkms build -m ${DRV_NAME} -v ${DRV_VERSION} -k "${KVER}/${KARCH}" -c "/usr/src/${DRV_NAME}-${DRV_VERSION}/dkms.conf" --force
        /usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-17-amd64 -a x86_64
	fi
```

at what is now line 383

```
    # dkms install -m ${DRV_NAME} -v ${DRV_VERSION} -k "${KVER}/${KARCH}" -c "/usr/src/${DRV_NAME}-${DRV_VERSION}/dkms.conf" --force
    /usr/sbin/dkms.mx install -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-17-amd64 -a x86_64
    RESULT=$?
```

morrownr, 2024-02-21:
I think I may have the fix for this. I'll try to merge a large patch very soon. I will need testers.

morrownr, 2024-02-21:
If you are reading this message, I have already merged a big patch that may have the fix to this problem. Please test and report.

@morrownr

john9631, 2024-02-21:
Tried it this morning following the instructions for using the script and the result was:

https://pastebin.com/M2hZaU7e

Using my butchered version worked.  It replaced lines for adding, building and installing with:

```
    /usr/sbin/dkms.mx add -m rtl88x2bu -v 5.13.1     -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf	
    /usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1   -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-17-amd64 -a x86_64
    /usr/sbin/dkms.mx install -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-17-amd64 -a x86_64

    /usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1   -k 6.1.0-16-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-16-amd64 -a x86_64
    /usr/sbin/dkms.mx install -m rtl88x2bu -v 5.13.1 -k 6.1.0-16-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-16-amd64 -a x86_64

    /usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1   -k 6.1.0-13-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-13-amd64 -a x86_64
    /usr/sbin/dkms.mx install -m rtl88x2bu -v 5.13.1 -k 6.1.0-13-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-13-amd64 -a x86_64
```

morrownr, 2024-02-22:
After stepping through install-driver.sh and the dkms docs again, I have not figured this out. We are all using Debian 12 based distros and the same version of dkms. I am not seeing this problem here. So what is different in my systems from the systems you guys are running? One obvious thing is that at least 2 of you are using multiple out-of-kernel drivers being handled by dkms. I have 5 dkms drivers installed on the box I am on right now but it is 5 of the drivers I maintain here so that are all installed by basically the same script.

We do see where the problem is happening:

/usr/sbin/dkms.mx build -m rtl88x2bu -v 5.13.1 -k 6.1.0-17-amd64/x86_64 -c /usr/src/rtl88x2bu-5.13.1/dkms.conf --force -k 6.1.0-16-amd64 -k 6.1.0-17-amd64 -k 6.1.0-13-amd64 -a x86_64

Everything beyond `--force` does not show up when I am running the script here. I've never seen those added switches and cannot explain them by looking in the dkms docs. It is like dkms is automatically adding them.

Any ideas?

morrownr, 2024-02-22:
FYI:

$ dkms status
rtl8812au/5.13.6-23, 6.1.0-18-amd64, x86_64: installed
rtl8812au/5.13.6-23, 6.5.0-0.deb12.4-amd64, x86_64: installed
rtl8814au/5.8.5.1, 6.1.0-18-amd64, x86_64: installed
rtl8814au/5.8.5.1, 6.5.0-0.deb12.4-amd64, x86_64: installed
rtl8821au/5.12.5.2, 6.1.0-18-amd64, x86_64: installed
rtl8821au/5.12.5.2, 6.5.0-0.deb12.4-amd64, x86_64: installed
rtl8852bu/1.19.3, 6.1.0-18-amd64, x86_64: installed
rtl8852bu/1.19.3, 6.5.0-0.deb12.4-amd64, x86_64: installed
rtl88x2bu/5.13.1, 6.5.0-0.deb12.4-amd64, x86_64: installed

john9631, 2024-02-22:
In my case its the dkms provided as part of mx linux so that might be the issue.

 /usr/sbin/dkms.mx

At least that's what I assume based on that line in the code generated by the script file.

morrownr, 2024-02-22:
@john9631 

I appreciate the help.

I just merged another patch. I have no idea if it will help with this problem as I still cannot duplicate the problem here. If you could do a `git pull` to grab the updates and give it another try, I would interested in the results.

> In my case its the dkms provided as part of mx linux so that might be the issue.

My dkms shows as version 3.0.10 also. It could be modified by the distro or it could be something else. This problem is a challenge to diagnose.

john9631, 2024-02-22:
Watching it build correctly.  16, 17, 13.  Install.  Reboot.

morrownr, 2024-02-22:
@linuxgameconsortium @mshpp 

@john9631 is reporting success after the patch I merged today. I like to have test reports for both of you as well if you have time.

FYI: I have been working on the installation and removal scripts for a few years now in an effort to make installing the out-of-kernel drivers easier. I have had to work around many distros specific ways of doing things. One of the goals I have here for the out-of-kernel drivers is to support as many distros as it feasible. That can be a challenge. It would help if distros worked together on establishing standards in as many areas as is feasible because the lack of standards for many of the areas causes a lot of work by people like me. I could use that time to make more progress in other areas to help users gain more satisfaction with usb wifi.

FYI: I was unable to fully understand how my latest patch appears to fix this problem but the changes I made should not cause other problems and should make for more robust scripts anyway so once you guys are able to test more over the next few days, if I don't see any problem reports, I will use the patch on all of the scripts here. 

Thanks for the help

@morrownr