morrownr/88x2bu-20210702 #204: Failure during installation on Debian 12 Kernel 6.1.0-15-amd64 - 88x2bu

kraso, 2024-02-03:
Hello, I received an error during the installation and I could not complete it. Below I attach a copy of my system logs and information.

    ,g$$$$$$$$$$$$$$$P.       --------- 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 12 (bookworm) x86_64 
 ,$$P'              `$$$.     Kernel: 6.1.0-15-amd64 
',$$P       ,ggs.     `$$b:   Uptime: 9 mins 
`d$$'     ,$P"'   .    $$$    Packages: 2068 (dpkg), 16 (snap) 
 $$P      d$'     ,    $$P    Shell: bash 5.2.15 
 $$:      $$.   -    ,d$$'    Resolution: 1280x1024 
 $$;      Y$b._   _,d$P'      DE: Xfce 4.18 
 Y$$.    `.`"Y$$$$P"'         WM: Xfwm4 
 `$$b      "-.__              WM Theme: Default 
  `Y$$                        Theme: Adwaita-dark [GTK2/3] 
   `Y$$.                      Icons: Tango [GTK2], Adwaita [GTK3] 
     `$$b.                    Terminal: xfce4-terminal 
       `Y$$b.                 Terminal Font: Monospace 12 
          `"Y$b._             CPU: Intel i7-4770 (8) @ 3.900GHz 
              `"""            GPU: NVIDIA GeForce GT 610 
                              Memory: 1435MiB / 7878MiB 

~$ sudo lsusb; sudo rfkill list all; sudo dkms status; sudo iw dev; sudo iw reg get
Bus 003 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU]
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

[make.log](https://github.com/morrownr/88x2bu-20210702/files/14150615/make.log)

I have no idea what the problem is. I thank you in advance for your collaboration and I take this opportunity to greet you.

morrownr, 2024-02-04:
Hi @kraso 

> I have no idea what the problem is. 

Me either. My main box is running Debian 12 with kernel 6,1 and I am not seeing this so something is different.

Can I get you to run the following and post the lines that come on the screen?

$ sudo sh install-driver.sh

kraso, 2024-02-04:
Hello @morrownr, here I am again with the information you requested.

Running install-driver.sh version 20220108
Operating System: Debian GNU/Linux 12 (bookworm)
6.1.0-15-amd64
x86_64
Starting installation...
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Copying 88x2bu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 KVER=6.1.0-15-amd64 KSRC=/lib/modules/6.1.0-15-amd64/build.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-15-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

morrownr, 2024-02-04:
> Running install-driver.sh version 20220108

Wow! That is an incredibly old version of the driver.

When dealing with these Realtek out-of-kernel drivers, never try to install old versions. You have to use one that is up to date. Suggest you run:

$ sudo ./remove-driver.sh

and then delete what you have entirely. The repo is located...

 https://github.com/morrownr/88x2bu-20210702

Please read and follow the Installation Steps.

In case you are tired of messing with compiling drivers, which sometimes does not work well, here is the Main Menu for this site:

https://github.com/morrownr/USB-WiFi

Read menu items 1 and 2 for a lot of good info.

FYI: If you are just running your adapter in managed mode, you really don't need this driver. If you are running Debian 12, the Debian repos have kernel 6.5 available and it has an in-kernel driver for the 8822bu. Do you know how to upgrade the kernel and check to see if the firmware is installed?

@morrownr

kraso, 2024-02-05:
Hello again, thank you very much for the information and for the time you have dedicated to me.

I have read points 1 and 2 of [the link](https://github.com/morrownr/USB-WiFi) that you indicated to me and I found it very interesting and almost mandatory reading for all users of a GNU/Linux operating system.

Finally, I have downloaded the [88x2bu-20210702](https://github.com/morrownr/88x2bu-20210702) repository and installed it properly.

The result has been a success, the card works correctly in managed mode and in monitor mode.

Regarding what you were saying about the kernel: No, I don't know how to look to see if there is a new version of the kernel in the repositories that I have configured, let alone install it. I assume that applications like apt-file, apt-cache or even apt with the search option would serve this purpose. To see the modules installed in the kernel I usually use the "lsmod" command but I have no idea how to verify the firmware installed in the kernel. If you have any information about this I would appreciate it if you would share it.

Greetings.

@kraso

morrownr, 2024-02-05:
> No, I don't know how to look to see if there is a new version of the kernel in the repositories that I have configured, let alone install it.

Here is a simple guide:

https://wiki.debian.org/HowToUpgradeKernel

It may cause a lot of questions so if things are not clear, ask.

>  I have no idea how to verify the firmware installed in the kernel.

See: https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

While that document is primarily about Mediatek firmware, a few lines into the document it mentions Realtek RTW88 firmware. You will need to check and possible install the rtw8822b_fw.bin file as instructed if it is not already in /lib/firmware/rtw88 .

For the in-kernel driver to work, you will need to run remove-driver.sh from this driver as this driver blacklist the in-kernel driver. Both drivers cannot be active as the same time.

@morrownr

kraso, 2024-02-06:
Thank you @morrownr. 

This information has been very useful to me.

Since I have gotten my card to work with this driver, I will continue using it. However, there was a problem after connecting to my Wi-Fi router, the system hung and did not let me interact with it again, I had to turn it off from the power button with the risks that this entails. After starting and trying again I have obtained the same result. So for now I only connect via the ethernet connection. I just updated my kernel to a more up-to-date version and am going to test it.

morrownr, 2024-02-06:
I have seen no problems with Debian 12 and the v6.5 kernel but things happen. You might want to remove it and work with the default v6.1 kernel.

One problem with upgrading to a new kernel in Debian is that there are so many versions of the kernel and the right one matters.

closed 2024-02-08 by kraso