morrownr/88x2bu-20210702 #206: How to install it. Step 8 ?

Lerwbd, 2024-02-05:
root@localhost:~/src/88x2bu-20210702# sudo ./install-driver.sh
sudo: unable to resolve host localhost: Name or service not known
Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.                            Once the header files are properly installed, please run "sudo ./install-driver.sh"
root@localhost:~/src/88x2bu-20210702#

Tôi đang bị dừng ở đây. Ai đó có thể giúp toii không ?... Đang dùng ubuntu in termux..
Cám ơn ai đã ghé qua và giúp đỡ

morrownr, 2024-02-05:
Hi @Lerwbd 

> Using ubuntu in termux..

Please clarify the version of Ubuntu that you are using.

@morrownr

Lerwbd, 2024-02-06:
Hello morrownr .

Hãy dịch lá thư này tự động. Cám ơn

Là một người mới nên sẽ có những câu hỏi "ngớ ngẩn". Mong là bạn hiểu, việc
bạn giúp đỡ tôi rất cảm kích

Tôi đang dùng ubuntu-in-termux phiên bản 22.04 ( mới nhất phải không ạ ?)
Hiện tại tôi đã cài lại vài lần. Phát hiện là không thẻ cài đặc
"kernel-headers"

Và bị vướng lại ở bước thứ 8. Nếu bỏ qua các gói cần thiết và kiến nghị như
trong bài viết.

Bạn có thể hướng dẩn, dạy tôi không ?

Lần nữa cảm ơn sự giúp đỡ của bạn

morrownr, 2024-02-06:
> ubuntu-in-termux

I use Ubuntu but I do not understand what the above is? `in termux` is not a term I am familiar with.

> Found out that it cannot be installed properly
"kernel-headers"

That is a problem. How are you trying to install the "kernel-keaders"?