morrownr/88x2bu-20210702 #207: T3U Archer Nano Speed issue

ellisonleao, 2024-02-08:
### A brief description of the problem

Connection for download and uploading in speedtest results are showing half of my current speed. My current driver options are:

```
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_switch_usb_mode=0
```

### Paste the screen output provided by the install-driver.sh script.

```
❯ sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 8/8 (in-use/total processing units)
: 16116964 (total system memory)
: 6.7.4-arch1-1 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (GCC) 13.2.1 20230801
: dkms-3.0.12
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers...
--> rtl88x2bu 5.13.1 6.7.4-arch1-1
Module rtl88x2bu-5.13.1 for kernel 6.7.4-arch1-1 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

88x2bu.ko.zst:
 - Uninstallation
   - Deleting from: /usr/lib/modules/6.7.4-arch1-1/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod...
Deleting module rtl88x2bu-5.13.1 completely from the DKMS tree.
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.
: ---------------------------

Sign command: /usr/lib/modules/6.7.4-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh...................
Signing module /var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko
Cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

88x2bu.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.7.4-arch1-1/updates/dkms/
depmod...
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 2 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!

Do you want to edit the driver options file now? (recommended) [Y/n] n
Do you want to apply the new options by rebooting now? (recommended) [Y/n] n
```

### The name and version of the Linux distro you are using.

```sh
❯ uname -a
Linux pilgrim 6.7.4-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 05 Feb 2024 22:07:49 +0000 x86_64 GNU/Linux
```

### The information obtained with the below command.

```
sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get
```
❯ sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 004: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 005: ID 13d3:56b2 IMC Networks Integrated Camera
Bus 001 Device 006: ID 258a:002a SINO WEALTH Gaming KB
Bus 001 Device 007: ID 046d:0ab7 Logitech, Inc. Blue Microphones
Bus 001 Device 008: ID 1532:000c Razer USA, Ltd Lachesis
Bus 001 Device 009: ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU]
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 003: ID 0bda:0328 Realtek Semiconductor Corp. USB3.0-CRW
Bus 002 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 005: ID 046d:085e Logitech, Inc. BRIO Ultra HD Webcam
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: yes
        Hard blocked: no
rtl88x2bu/5.13.1, 6.7.4-arch1-1, x86_64: installed
phy#0
        Interface wlan0
                ifindex 2
                wdev 0x1
                addr 34:60:f9:f5:18:78
                ssid lions
                type managed
                channel 44 (5220 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 15.00 dBm
global
country US: DFS-FCC # I am at Brazil but can't set the right reg domain here
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

morrownr, 2024-02-08:
Hi @ellisonleao 

> rtw_switch_usb_mode=0

The above is likely the problem. To turn on USB3 mode:

Open a terminal is your driver directory and:

$ sudo ./edit-options.sh

Edit the options like and change the option I posted above to:

rtw_switch_usb_mode=1

...and reboot. See if the problem is fixed.

FYI:  Here is a link to the Main Menu for this site:

https://github.com/morrownr/USB-WiFi

There is a lot of good info for those running usn wifi adapters on Linux.

@morrownr

ellisonleao, 2024-02-09:
hey @morrownr thanks for the reply. I've tried changing the USB settings but the issue remains. I had the normal speed with that config previously so I am not sure if this is something related to recent changes in the driver. 

Anything else I can try?

ellisonleao, 2024-02-10:
reinstalled the driver again and it worked, so I think the issue is still on my side but I couldn't figure out 100%. anyway, closing this one

closed 2024-02-10 by ellisonleao

morrownr, 2024-02-11:
Sometimes WiFi works in mysterious ways.