morrownr/88x2bu-20210702 #208: MT7601U weired Active monitor mode behavour on Rasperbry Pi

LightMoon, 2024-02-10:
Hi! 
I am facing a strange issue with AWUS036ACHM - Chipset MT7610U on Kali Linux raspberry Pi 5.
The card stops working and throwing ioctl(SIOCGIFINDEX) failed: No such device error if you run Aireplay-ng while having Airodump-ng running.

Error on shown on  airodump-ng
```
read failed: Network is down                                                                         
Interface wlan1mon:                                                                                  
ioctl(SIOCGIFINDEX) failed: No such device                                                           
Can't reopen wlan1mon  
```
Error on shown on aireplay-ng

```
NB: this attack is more effective when targeting
a connected wireless client (-c <client's mac>).
12:30:31  Sending DeAuth (code 7) to broadcast -- BSSID: [A0:04:60:xx:xx:xx]
12:30:32  Sending DeAuth (code 7) to broadcast -- BSSID: [A0:04:60:xx:xx:xx]
write failed: Network is down
wi_write(): Network is down
```
Good to note, I did ran `airmon-ng check kill` prior to eliminate processes interference, also replaced the driver to the latest, however, none has stopped the issue.

`uname -a                                                                                         
Linux kali-raspberry-pi5 6.1.64-v8+ #1 SMP PREEMPT Sun Dec  3 11:34:54 UTC 2023 aarch64 GNU/Linux
` It seems the issue is related to arm CPUs as it can be reproduced on latest Raspbian OS; although the card on Kali (Intel) works fine. 
I am not sure but if someone can confirm this issue could be related to bug - https://bugzilla.kernel.org/show_bug.cgi?id=217465 would be great. Any tips in getting the card to usable stage will be appreciated.

morrownr, 2024-02-10:
Hi @LightMoon 

Can I get you to repost this issue in Issues at the Main Menu:

https://github.com/morrownr/USB-WiFi

This repo is specific to adapters with the 88x2bu chipset but the Issues at the Main Menu is general and many users of Mediatek based adapters stop by there.

Also, if you can provide the specific steps you are using so that we can try to reproduce the problem, that would be great. I have an ACHM, RasPi and many others do also.

@morrownr

LightMoon, 2024-02-11:
@morrownr the issue should have been opened in USB-WiF in the first place! I guess I had too many tabs opened and mistakenly created the issue here. feel free to delete it to avoid being duplicated.

closed 2024-02-11 by morrownr