morrownr/88x2bu-20210702 #212: Unable to locate package raspberrypi-kernel-headers, 6.1.77-v8+

UdroiuMarius, 2024-03-08:
Hi, @morrownr.

First of all, thank you for maintaining this repo.

I am currently rapping my head around installing the driver for my TP-Link T3U Plus Wi-Fi adapter on a RPi5 running kali linux. 
Everything is working fine until I reach "raspberrypi-kernel-headers". That's when I get "E: Unable to locate package raspberrypi-kernel-headers". As far as I've been reading about it, it looks like there might be a mismatch between the kernel versions (my version is "6.1.77-v8+"). 
I tried running the next commands thinking the headers are not the problem, but, to no surprise, it didn't work. 
Here is the make.log file that states there is actually a problem with the versions:

DKMS make.log for rtl88x2bu-5.13.1 for kernel 6.1.77-v8+ (aarch64)
Fri Mar  8 10:04:14 PM UTC 2024
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.77-v8+/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/kernel'
awk: fatal: cannot open file `include/generated/asm-offsets.h' for reading: No such file or directory
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110
  You are using:           gcc (Debian 13.2.0-13) 13.2.0
make[2]: *** No rule to make target 'scripts/mod/modpost', needed by '/var/lib/dkms/rtl88x2bu/5.13.1/build/Module.symvers'.  Stop.
make[1]: *** [Makefile:1966: modpost] Error 2
make[1]: Leaving directory '/usr/src/kernel'
make: *** [Makefile:2524: modules] Error 2

Version: Linux kali-raspberry-pi5 6.1.77-v8+ #1 SMP PREEMPT Mon Feb 26 17:41:54 UTC 2024 aarch64 GNU/Linux
Adapter: https://www.amazon.com/TP-Link-usb-wifi-adapter-pc/dp/B08D72GSMS

Thank you for your support!

UdroiuMarius, 2024-03-09:
Update: I've managed to figure it out by searching on the kali forums.
It looks like things have changed with the RPI5, because: 
"most instructions will state that you need to install header packages. This is not the case on the Raspberryi Pi 5 image, they are already included and are not a part of a package."

To sum it up, basically when I was trying to install the headers, there were no headers to install because the kali image comes with the headers already pre-installed. You just have to write a few commands to make sure the Pi knows that.
Here is the link if anyone has the same issues with their Raspberryi Pi 5:
https://www.kali.org/docs/arm/raspberry-pi-5/

closed 2024-03-09 by UdroiuMarius

morrownr, 2024-03-09:
I'm glad you found that as I did not have a specific fix in mind. One thing I will say is that below looks strange:

> The kernel was built by: aarch64-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110
> You are using: gcc (Debian 13.2.0-13) 13.2.0

Having a kernel compiled with gcc 10 and the system using gcc 13 is more of spread than I have ever seen and a spread that big usually won't work.

morrownr, 2024-03-09:
If you are not aware of the Main Menu at this site, here it is:

https://github.com/morrownr/USB-WiFi

Since you are running Kali, you may want to look at the Plug and Play List. One of the best adapters for monitor mode work is the Alfa ACHM. You can ask questions in Issues.