morrownr/88x2bu-20210702 #215: Compilation error on AlmaLinux 9.4

chofchof, 2024-05-09:
I had a compilation error on **AlmaLinux 9.4** with the kernel 5.14.0-427.13.1.el9_4.x86_64.

The error message after running `make` was as follows.
```
$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-427.13.1.el9_4.x86_64/build M=/...skip.../88x2bu-20210702  modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-427.13.1.el9_4.x86_64'
  CC [M]  /...skip.../88x2bu-20210702/core/rtw_cmd.o
... (SKIP) ...
  CC [M]  /...skip.../88x2bu-20210702/os_dep/linux/ioctl_cfg80211.o
/...skip.../88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:10474:26: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, struct cfg80211_ap_update *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, struct cfg80211_beacon_data *)’ [-Werror=incompatible-pointer-types]
10474 |         .change_beacon = cfg80211_rtw_change_beacon,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
/...skip.../88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:10474:26: note: (near initialization for ‘rtw_cfg80211_ops.change_beacon’)
/...skip.../88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:10546:22: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, const u8 *, int,  u8,  u8,  u16,  u32,  bool,  const u8 *, size_t)’ {aka ‘int (*)(struct wiphy *, struct net_device *, const unsigned char *, int,  unsigned char,  unsigned char,  short unsigned int,  unsigned int,  _Bool,  const unsigned char *, long unsigned int)’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, const u8 *, u8,  u8,  u16,  u32,  bool,  const u8 *, size_t)’ {aka ‘int (*)(struct wiphy *, struct net_device *, const unsigned char *, unsigned char,  unsigned char,  short unsigned int,  unsigned int,  _Bool,  const unsigned char *, long unsigned int)’} [-Werror=incompatible-pointer-types]
10546 |         .tdls_mgmt = cfg80211_rtw_tdls_mgmt,
      |                      ^~~~~~~~~~~~~~~~~~~~~~
/...skip.../88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:10546:22: note: (near initialization for ‘rtw_cfg80211_ops.tdls_mgmt’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:299: /...skip.../88x2bu-20210702/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1934: /...skip.../88x2bu-20210702] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-427.13.1.el9_4.x86_64'
make: *** [Makefile:2524: modules] Error 2
```

I could solve this problem with the following patch.
```
diff --git a/Makefile b/Makefile
index ff4a3df..33d54e2 100644
--- a/Makefile
+++ b/Makefile
@@ -75,6 +75,9 @@ endif
 ifeq ($(shell test $(RHEL_SVER) -ge 362; echo $$?),0)
 EXTRA_CFLAGS += -DRHEL89
 endif
+ifeq ($(shell test $(RHEL_SVER) -ge 427; echo $$?),0)
+EXTRA_CFLAGS += -DRHEL94 -DRHEL89
+endif
 endif
 endif

diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index f63fb84..e5be490 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -5405,7 +5405,7 @@ exit:
 }

 static int cfg80211_rtw_change_beacon(struct wiphy *wiphy, struct net_device *ndev,
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 7, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 7, 0)) || defined(RHEL94)
        struct cfg80211_ap_update *info)
 #else
        struct cfg80211_beacon_data *info)
@@ -5416,7 +5416,7 @@ static int cfg80211_rtw_change_beacon(struct wiphy *wiphy, struct net_device *nd

        RTW_INFO(FUNC_NDEV_FMT"\n", FUNC_NDEV_ARG(ndev));

-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 7, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 7, 0)) || defined(RHEL94)
        ret = rtw_add_beacon(adapter, info->beacon.head, info->beacon.head_len, info->beacon.tail, info->beacon.tail_len);
 #else
        ret = rtw_add_beacon(adapter, info->head, info->head_len, info->tail, info->tail_len);
@@ -8455,7 +8455,7 @@ static int cfg80211_rtw_tdls_mgmt(struct wiphy *wiphy,
 #else
        u8 *peer,
 #endif
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 5, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 5, 0)) || defined(RHEL94)
        int link_id,
 #endif
        u8 action_code,
diff --git a/os_dep/linux/wifi_regd.c b/os_dep/linux/wifi_regd.c
index a11e21b..0b5c46a 100644
--- a/os_dep/linux/wifi_regd.c
+++ b/os_dep/linux/wifi_regd.c
@@ -405,7 +405,7 @@ int rtw_regd_init(struct wiphy *wiphy)
        wiphy->regulatory_flags &= ~REGULATORY_DISABLE_BEACON_HINTS;
 #endif

-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 19, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 1, 39))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 19, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 1, 39)) && !defined(RHEL94)
        wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
 #endif
```

morrownr, 2024-05-09:
Hi @chofchof 

@misha4gps is the RHEL maintainer for this driver but he could probably use a partner. Can I get you and him to coordinate a PR to solve this problem?

Thanks,

@morrownr

misha4gps, 2024-05-09:
Hi @chofchof 

Thanks for the patch. I will check soon.
@misha4gps

chofchof, 2024-05-10:
> @misha4gps is the RHEL maintainer for this driver but he could probably use a partner. Can I get you and him to coordinate a PR to solve this problem?

Sure!

misha4gps, 2024-05-11:
Hi @chofchof

PR#216 - is ready
@misha4gps

morrownr, 2024-05-11:
Merged. Thanks.

closed 2024-05-13 by chofchof