morrownr/88x2bu-20210702 #219: Driver build error for Tp-Link Archer T3UPlus AC 1300 on Arch Linux Zen Kernel 6.9.1 (build log included)

Hikmat-E-Ustad, 2024-05-18:
I have an 64 bit Arch Linux System running Linux Zen Kernel 6.8.9 

```
hikmateustad@hikmateustad-dell ~ % archey
                                 User: hikmateustad
               +                 Hostname: hikmateustad-dell
               #                 
              ###                Distro: Arch Linux x86_64
             #####               Kernel: Linux 6.8.9-zen1-2-zen
             ######              Uptime: 21 minutes
            ; #####;             Load Average: 1.54 2.51 2.05
           +##.#####             Processes: 205
          +##########            Window Manager: KWin (X11)
         #############;          Desktop Environment: KDE
        ###############+         Shell: /bin/zsh
       #######   #######         Terminal: Konsole ██ ██ ██ ██ ██ ██ ██
     .######;     ;###;`".       Packages: 2183
    .#######;     ;#####.        Temperature: 55.8 C (Max. 57.0 C)
    #########.   .########`      
   ######'           '######     
  ;####                 ####;    
  ##'                     '##    
 #'                         `#   
                                 

```

I have been using Tp-Link Archer T3UPlus AC 1300 with your driver since more than a year perfectly with no downtime.

While upgrading to linux zen kernel 6.9.1 with dkms the rtl88x2bu driver exited with code 10 as you can see from the output below

```
hikmateustad@hikmateustad-dell ~ % sudo pacman -Syu --noconfirm && yay -Syu --noconfirm            
:: Synchronizing package databases...
 core                                          121.3 KiB   105 KiB/s 00:01 [------------------------------------------] 100%
 extra                                           7.8 MiB  2.39 MiB/s 00:03 [------------------------------------------] 100%
 community is up to date
 multilib is up to date
 chaotic-aur                                   774.2 KiB   656 KiB/s 00:01 [------------------------------------------] 100%
:: Starting full system upgrade...
warning: grub: ignoring package upgrade (2:2.06.r591.g6425c12cd-1 => 2:2.12-2)
resolving dependencies...
looking for conflicting packages...

Packages (4) libdatachannel-0.21.1-1  libjuice-1.4.2-1  linux-zen-6.9.1.zen1-1  linux-zen-headers-6.9.1.zen1-1

Total Download Size:   166.60 MiB
Total Installed Size:  274.80 MiB
Net Upgrade Size:        3.45 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 libjuice-1.4.2-1-x86_64                        71.6 KiB  31.2 KiB/s 00:02 [------------------------------------------] 100%
 libdatachannel-0.21.1-1-x86_64                804.5 KiB  94.1 KiB/s 00:09 [------------------------------------------] 100%
 linux-zen-headers-6.9.1.zen1-1-x86_64          26.6 MiB   909 KiB/s 00:30 [------------------------------------------] 100%
 linux-zen-6.9.1.zen1-1-x86_64                 139.2 MiB  2.38 MiB/s 00:59 [------------------------------------------] 100%
 Total (4/4)                                   166.6 MiB  2.84 MiB/s 00:59 [------------------------------------------] 100%
(4/4) checking keys in keyring                                             [------------------------------------------] 100%
(4/4) checking package integrity                                           [------------------------------------------] 100%
(4/4) loading package files                                                [------------------------------------------] 100%
(4/4) checking for file conflicts                                          [------------------------------------------] 100%
:: Running pre-transaction hooks...
(1/2) Removing linux initcpios...
(2/2) Remove DKMS modules
==> dkms remove --no-depmod acpi_call/1.2.2 -k 6.8.9-zen1-2-zen
==> dkms remove --no-depmod nvidia/390.157 -k 6.8.9-zen1-2-zen
==> dkms remove --no-depmod rtl88x2bu/5.13.1 -k 6.8.9-zen1-2-zen
==> dkms remove --no-depmod v4l2loopback/0.13.1 -k 6.8.9-zen1-2-zen
==> depmod 6.8.9-zen1-2-zen
:: Processing package changes...
[... 205 lines cut ...]
                wdev 0x1
                addr 06:f5:f2:1c:27:47
                type managed
                txpower 0.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

hikmateustad@hikmateustad-dell ~ % 

```

For now I will downgrade back to linux zen kernel 6.8.9 where this driver worked perfectly till we have a fix.

Thank you in advance for your time and hard work in providing this driver.

morrownr, 2024-05-18:
Hi @Hikmat-E-Ustad 

Try this while in the driver directory:

$ git pull
$ sudo sh install-driver.sh

I think the problem is that you are not keeping the driver up to date.

Also, you are aware that the in-kernel rtw88 supports this adapter chip in this day in time?

Hikmat-E-Ustad, 2024-05-18:
Greetings @morrownr 

Thank you so much for your guidance. I followed your steps and my driver works again with the newer linux zen kernel 6.9.1.

I was not aware of the in-kernel support; nevertheless it does not work for me because when I removed the driver using 

```
sudo sh remove-driver.sh
```
and then rebooted I did not have internet. 

Then when I git pulled and installed the driver it worked fine again.

Highly appreciate your help.

Hikmat-E-Ustad, 2024-05-18:
Well my desktop arch installation has to be in trouble somehow because the in-kernel support does not work while it does on my laptop arch installation using the same adapter. ( Upgraded the linux zen kernel on the laptop; build failed, removed driver, rebooted and I have the adapter working fine)

Anyway I can investigate as to what may be the issue on my desktop arch linux installation?

closed 2024-05-18 by Hikmat-E-Ustad

morrownr, 2024-05-18:
> Anyway I can investigate as to what may be the issue on my desktop arch linux installation?

I can help. It would be good to figure out why the in-kernel rtw88 is not working on the one machine. If we find the problem and it is a problem with Arch, you can report it.

The first thing to look at is whether the problem system has the firmware for the adapter installed. I have a guide for checking that on the Main Menu here:

https://github.com/morrownr/USB-WiFi

Look for menu item 7. This menu item gives help regarding firmware for wireless devices. The help for rtw88 is a one liner.

The first thing you can do is check your rtw88 firmware by posting the results of:

$ ls -l /lib/firmware/rtw88

It is the file `rtw8822b_fw.bin` that you need to make sure it is there. Here is an example when I run the command on my main system:

$ ls -l /lib/firmware/rtw88
total 812
-rw-r--r-- 1 root root  59332 May 15 18:20 rtl8822cu_fw.bin
-rw-r--r-- 1 root root  28884 May 15 18:20 rtw8723d_fw.bin
-rw-r--r-- 1 root root  30880 May  5 20:54 rtw8811a_fw.bin
-rw-r--r-- 1 root root  27030 May 15 18:20 rtw8812a_fw.bin
-rw-r--r-- 1 root root  31898 May 15 18:20 rtw8821a_fw.bin
-rw-r--r-- 1 root root 138984 May 15 18:20 rtw8821c_fw.bin
-rw-r--r-- 1 root root 150984 May 15 18:20 rtw8822b_fw.bin
-rw-r--r-- 1 root root 202552 May 15 18:20 rtw8822c_fw.bin
-rw-r--r-- 1 root root 145224 May 15 18:20 rtw8822c_wow_fw.bin

I will show more files that you because I am helping to bring more drivers to rtw88.

If that is not the problem, we can continue looking.

> Would you recommend installing your driver when the in-kernel support works based on performance?

Performance with rtw88 is pretty good. I can discuss this more if we get rtw88 going on both of your systems.

Hikmat-E-Ustad, 2024-05-20:
```
hikmateustad@hikmateustad-dell ~ % ls -l /lib/firmware/rtw88
total 336
-rw-r--r-- 1 root root   584 May 13 17:47 README.zst
-rw-r--r-- 1 root root 13117 May 13 17:47 rtw8703b_fw.bin.zst
-rw-r--r-- 1 root root 14647 May 13 17:47 rtw8703b_wow_fw.bin.zst
-rw-r--r-- 1 root root 17918 May 13 17:47 rtw8723d_fw.bin.zst
-rw-r--r-- 1 root root 53954 May 13 17:47 rtw8821c_fw.bin.zst
-rw-r--r-- 1 root root 77712 May 13 17:47 rtw8822b_fw.bin.zst
-rw-r--r-- 1 root root 85925 May 13 17:47 rtw8822c_fw.bin.zst
-rw-r--r-- 1 root root 64058 May 13 17:47 rtw8822c_wow_fw.bin.zst
hikmateustad@hikmateustad-dell ~ % 
```

Well the file is available. 

I installed the rtw88 driver on my laptop too. I feel comfy enabling USB 3 and led blinking which the in-kernel driver does not.

morrownr, 2024-05-20:
@Hikmat-E-Ustad 

> I installed the rtw88 driver on my laptop too.

Can I get you to explain that comment? Modern Linux kernels contain the rtw88 drivers and most modern distros should be installing needed firmware files. What exactly did you install?

Hikmat-E-Ustad, 2024-05-22:
@morrownr  
> What exactly did you install?

Well I installed the drivers from this github and in the post configuration steps turned on USB 3 since I have usb 3 ports on my laptop. Also the led on my Tp link keeps blinking the ferocity of which indicates data in out probably ;-)

On my Dell desktop I have USB 2 ports and anyways the in-kernel driver did not work whose _**ls -l /lib/firmware/rtw88**_ output I provided on my previous post