morrownr/88x2bu-20210702 #220: Successfully installed, but not showing as wlan1...

EATREVITPOOPCAD, 2024-05-26:
Installation seems to be successful, but when I run **iwconfig** its not showing up as a network adapter... Not sure what else to do to troubleshoot this

[image]

morrownr, 2024-05-26:
Hi @EATREVITPOOPCAD 

I see that `lsusb` shows a Netgear A7000. Is that the usb wifi adapter that you are trying to install a driver for?

EATREVITPOOPCAD, 2024-05-26:
yep, that's the one.

I have it plugged into the 3.0 USB port and have set **rtw_switch_usb_mode=1** in **/etc/modprobe.d/88x2bu.conf**

morrownr, 2024-05-26:
You installed the wrong driver. Run:

$ sudo sh remove-driver.sh
and then delete the folder you have the current driver in.

Then, go to and follow the directions:

https://github.com/morrownr/8814au

FYI: The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

It has a lot of good info for us USB WiFi adapter users.

Let me know how it goes.

EATREVITPOOPCAD, 2024-05-27:
A live example of what happens when a chimp flies a plane -_-

Anyways, It worked... Thank you so much!!

Couple things I want to bring to your attention, as this may help other noobies such as myself... 

Because I am using 5.15.44-Re4son-v7l+ & Debian kali-pi instead of 
`sudo apt install -y raspberrypi-kernel-headers build-essential bc dkms git`
I had to use 
`sudo apt install -y kalipi-kernel-headers build-essential bc dkms git`

Also, in **/etc/modprobe.d/8814au.conf** there is no mention that **rtw_switch_usb_mode=1** is needed for USB 3.0. I was able to figure this out because there was a comment that explained that in  **/etc/modprobe.d/88x2bu.conf**

But other than that, the instructions are great... Speaking on behalf of someone who never used Linux before :)

morrownr, 2024-05-27:
Well, this chimp has been flying Linux since 1994 (v1). It just takes a couple of months to get comfortable and then the fun begins.

Glad it worked. I have already made the changes you recommended. Thanks.

Did I point you to the Main Menu of this site? Anyone that uses Linux and USB WiFi adapters should know where the Main Menu is:

https://github.com/morrownr/USB-WiFi

As you have time, scroll down and read menu items 1 and 2.

Good luck.

EATREVITPOOPCAD, 2024-05-29:
Since v1 is impressive! And it is already fun :) but I am sure I will take saying that back soon, haha

The main page is on my radar and looks like a fountain of knowledge. I will definitely spend some time there.