morrownr/88x2bu-20210702 #222: Dual TP-LINK AC1300 adapter setup not working

ioaur, 2024-06-05:
I am using two TP-LINK AC1300 adapter on a RPI 4b+
- The first adapter emits an AP using hostapd.
- The second adapter is used to connect to a wifi using wpa_supplicant.

- A brief description of the problem
> I can't get both adapters to do their job **at the same time**. Is it possible that when the driver configures itself for AP it cannot be used by the other dongle to connect to a wifi and vice-versa?

- Paste the screen output provided by the install-driver.sh script

> 2024-06-04T15:07:06.8713075Z : ---------------------------
2024-06-04T15:07:06.8714046Z : install-driver.sh v20240222
2024-06-04T15:07:06.8714607Z : aarch64 (kernel architecture)
2024-06-04T15:07:06.8714941Z : arm64 (architecture to send to gcc)
2024-06-04T15:07:07.0037737Z : 2/2 (in-use/total processing units)
2024-06-04T15:07:07.0038325Z : 7087680 (total system memory)
2024-06-04T15:07:07.0038954Z : 6.1.21-v8+ (kernel version)
2024-06-04T15:07:07.0987051Z : gcc-11 (version of gcc used to compile the kernel)
2024-06-04T15:07:07.1638339Z : gcc (Debian 10.2.1-6) 10.2.1 20210110
2024-06-04T15:07:08.5342387Z : dkms:2.8
2024-06-04T15:07:08.5343096Z : mokutil not installed (Secure Boot status unknown)
2024-06-04T15:07:08.5343699Z : ---------------------------
2024-06-04T15:07:08.5343877Z 
2024-06-04T15:07:08.5344113Z Checking for previously installed drivers...
2024-06-04T15:07:10.0076434Z Finished checking for and removing previously installed drivers.
2024-06-04T15:07:10.0077661Z : ---------------------------
2024-06-04T15:07:10.0078484Z 
2024-06-04T15:07:10.0078761Z Starting installation.
2024-06-04T15:07:10.0079060Z Installing 88x2bu.conf to /etc/modprobe.d
2024-06-04T15:07:10.0569173Z The dkms installation routines are in use.
2024-06-04T15:07:10.0570093Z Copying source files to /usr/src/rtl88x2bu-5.13.1
2024-06-04T15:07:12.5755053Z 
2024-06-04T15:07:12.5758509Z Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source ->
2024-06-04T15:07:12.5759067Z                  /usr/src/rtl88x2bu-5.13.1
2024-06-04T15:07:12.6645095Z 
2024-06-04T15:07:12.6645773Z DKMS: add completed.
2024-06-04T15:07:12.6690011Z The driver was added to dkms successfully.
2024-06-04T15:07:12.6690648Z : ---------------------------
2024-06-04T15:07:12.6690797Z 
2024-06-04T15:07:15.6763584Z 
2024-06-04T15:07:15.6765726Z Kernel preparation unnecessary for this kernel.  Skipping...
2024-06-04T15:07:15.8557079Z 
2024-06-04T15:07:15.8557870Z Building module:
2024-06-04T15:07:17.4248815Z cleaning build area...
2024-06-04T15:20:51.8068838Z kernelver=6.1.21-v8+ ./dkms-make.sh...........................................................................................................................................................................................................................................................................
2024-06-04T15:21:07.0004481Z cleaning build area...
2024-06-04T15:21:07.2310015Z 
2024-06-04T15:21:07.2310785Z DKMS: build completed.
2024-06-04T15:21:07.2403056Z The driver was built by dkms successfully.
2024-06-04T15:21:07.2403794Z : ---------------------------
2024-06-04T15:21:10.9035774Z 
2024-06-04T15:21:10.9037889Z 88x2bu.ko.xz:
2024-06-04T15:21:10.9042900Z Running module version sanity check.
2024-06-04T15:21:11.4415786Z  - Original module
2024-06-04T15:21:11.4701639Z    - No original module exists within this kernel
2024-06-04T15:21:11.4702219Z  - Installation
2024-06-04T15:21:11.4704080Z    - Installing to /lib/modules/6.1.21-v8+/updates/dkms/
2024-06-04T15:21:11.6598617Z 
2024-06-04T15:21:22.0849257Z depmod......
2024-06-04T15:21:22.0850008Z 
2024-06-04T15:21:22.0850489Z DKMS: install completed.
2024-06-04T15:21:22.0931590Z The driver was installed by dkms successfully.
2024-06-04T15:21:22.0932196Z : ---------------------------
2024-06-04T15:21:22.0932391Z 
2024-06-04T15:21:22.0934211Z Info: Update this driver with the following commands as needed:
2024-06-04T15:21:22.0934396Z 
2024-06-04T15:21:22.0934592Z $ git pull
2024-06-04T15:21:22.0934861Z $ sudo sh install-driver.sh
2024-06-04T15:21:22.0935013Z 
2024-06-04T15:21:22.0935229Z Note: Updates to this driver SHOULD be performed before distro
2024-06-04T15:21:22.0935490Z       upgrades such as Ubuntu 23.10 to 24.04.
2024-06-04T15:21:22.0935735Z Note: Updates to this driver SHOULD be performed before major
2024-06-04T15:21:22.0937111Z       upgrades such as kernel 6.5 to 6.6.
2024-06-04T15:21:22.0937369Z Note: Updates can be performed as often as you like. It is
2024-06-04T15:21:22.0937627Z       recommended to update at least every 3 months.
2024-06-04T15:21:22.0938236Z Note: Work on this driver, like the Linux kernel, is continuous.
2024-06-04T15:21:22.0938397Z 
2024-06-04T15:21:22.0938587Z Enjoy!

- The name and version of the Linux distro you are using
> Debian 11 "bullseye"

- The information obtained with the below command
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 006: ID 2357:012e TP-Link 802.11ac NIC
Bus 001 Device 003: ID 2357:012e TP-Link 802.11ac NIC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: phy2: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl88x2bu, 5.13.1, 6.1.21-v8+, aarch64: installed
phy#2
	Interface wlan_ap
		ifindex 5
		wdev 0x200000001
		addr a8:42:a1:40:f1:22
		ssid ZipCam-io
		type AP
		channel 40 (5200 MHz), width: 40 MHz, center1: 5190 MHz
		txpower 13.00 dBm
phy#1
	Interface wlan_network
		ifindex 3
		wdev 0x100000001
		addr f0:a7:31:1a:5c:81
		type managed
		txpower -100.00 dBm
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, NO-IR
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
- Also, here are the wpa_supplicant's logs when the second adapter tries to connect to the wifi
> nl80211: Connect request send successfully
wlan_network: Setting authentication timeout: 10 sec 0 usec
Not configuring frame filtering - BSS 00:00:00:00:00:00 is not a Hotspot 2.0 network
EAPOL: External notification - EAP success=0
EAPOL: External notification - EAP fail=0
EAPOL: External notification - portControl=Auto
RTM_NEWLINK: ifi_index=3 ifname=wlan_network wext ifi_family=0 ifi_flags=0x1003 ([UP])
nl80211: Drv Event 110 (NL80211_CMD_UNKNOWN) received for wlan_network
nl80211: Channel switch started event
wlan_network: Event CH_SWITCH_STARTED (39) received
wlan_network: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5805 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5775 cf2=0
nl80211: Drv Event 46 (NL80211_CMD_CONNECT) received for wlan_network
nl80211: Connect event (status=1 ignore_next_local_disconnect=0)
wlan_network: Event ASSOC_REJECT (12) received
wlan_network: CTRL-EVENT-ASSOC-REJECT status_code=1
wlan_network: Radio work 'connect'@0x559f5c6700 done in 1.758649 seconds
wlan_network: radio_work_free('connect'@0x559f5c6700): num_active_works --> 0
Added BSSID 26:74:f7:f1:a3:bc into blacklist
Continuous association failures - consider temporary network disabling
wlan_network: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="GP25004512" auth_failures=6 duration=143 reason=CONN_FAILED

morrownr, 2024-06-05:
> I can't get both adapters to do their job at the same time.

And you won't. You can thank Realtek for this oversight.

I can suggest some alternative solutions if you want.

ioaur, 2024-06-06:
The problem seems to be with the initial connection for the second adapter. If it's first connected to a WiFi network, both adapters do their job perfectly.

An alternative solution would be greatly appreciated. Also, a huge thank you for your work and quick response!

morrownr, 2024-06-06:
I did some retesting after a recent request. Things can change and I had not looked at this issue for a while.

What I found was that adapters based on the following chips using in-kernel drivers do support more than one adapter with the same chipset:

mt7610u (AC600)
mt7612u (AC1200)
mt7921au (AXE3000)- the driver is mt7921u
rtw88 - I tested the 8812bu chipset. Instead of the in-kernel rtw88, you might try:

https://github.com/lwfinger/rtw88

It is a downstream of the in-kernel driver and usually stays ahead of the in-kernel version. It does require manual installation so follow the README.

ioaur, 2024-06-06:
Again, thank you for your answers and time. Since you answered all my questions, I'll go ahead and close this issue.

closed 2024-06-06 by ioaur

morrownr, 2024-06-06:
Good luck.

One thing I forgot to mention: rtw88, both the in-kernel version and the one I provided the link for, do not support USB3 yet. Hopefully that can be corrected later this year and it may not be an issue for you anyway. 

To put things in perspective, while Realtek provided and supports rtw88, they only support it for cards, not the slow buses like USB and SDIO so these have to be supported by the Linux community. That contrasts greatly with the full support we get from Mediatek. Mediatek does not discriminate against us USB users,