morrownr/88x2bu-20210702 #224: Project: Improve the 8822bu rtw88 in-kernel driver. Need testers...

morrownr, 2024-06-25:
Greetings to anyone that reads this message.

This Issue is where we coordinate and take bug reports for the effort to improve the 8822bu in-kernel driver. The 8822bu in-kernel driver supports the same adapters as the 88x2by driver in this repo so if you are here for the driver in this repo, you can help us test.

An effort is underway to improve the support for the rtl88x2bu chips in the rtw88 in-kernel driver series. The driver that we are using for testing is available at the following repo:

https://github.com/lwfinger/rtw88

Remember to first remove the out-of-kernel driver in this repo or whatever repo you may have installed. You can run the following to remove it if using this repo:

$ sudo sh remove-driver.sh

It is important to follow the instructions in the README at the repo with the new test driver. You may not be familiar with rtw88 in the kernel but even if you are, there are some necessary mods that you need to know about. The rtw88 that has this new driver is more advanced than the rtw88 in stable kernels as it follows wireless-next and is used to work on and develop new drivers. You do need to Blacklist the existing 8822bu driver in the kernel if you are using kernel 6.2 or later so look for how to do that in the README.

We welcome you to test and report on this rtw88 driver. Your testing will help us improve the driver in the Linux kernel. If you are aware of anyone who is familiar with mac80211 drivers, please invite them as more eyes on the code is a good thing. Your ideas are most welcome. We can do this.

@morrownr

What we need tested:

- USB 3 support. Please test with USB 3 ports and USB 2 ports. Even USB 1.x if you have them. Check to see if the usb port is set to usb3 mode in a usb3 port and the same with USB 2 ports.

- We need testing and report on AP mode, monitor mode, IBSS and PSP.
- 
Reports go in this issue.

ADV-LUT, 2024-06-26:
Hello @morrownr ,

I wanted to provide an update on my ongoing efforts to set up a stable wifi-driver on my device. 

Steps Taken and Findings:

- Driver Implementation:
        I successfully implemented the driver from this repository: https://github.com/lwfinger/rtw88

- Usage as a Normal WiFi Connection:
      - Single Connection Works: Initially connecting to a network works
      - Network Switching Issue: Switching to another network fails.
      - Reboot Fix: After a reboot, the new connection works.
      - DHCP Issue: When the connection fails, the device gets an IP from the DHCP server, but even a ping to the router's address fails with "Destination Host Unreachable".
      - Intermittent Boot Issue: Sometimes this error also appears on boot, but often it works fine after a reboot.

- Usage as a Hotspot:
      - Data Connection Drops: Both with nmcli and hostapd, I can establish a connection, but it drops after a few seconds.

Setup Details:

Hardware: I use a SOM board with interchangeable CPUs, and the 8822cu is permanently connected via USB 2.0.

Hardware Specification (from lshw):
>     description: Bluetooth wireless interface
>     product: 802.11ac NIC
>     vendor: Realtek
>     physical id: 4
>     bus info: usb@1:2.4
>     version: 0.00
>     serial: 123456
>     capabilities: usb-2.00 bluetooth
>     configuration: driver=rtw_8822cu maxpower=500mA speed=480Mbit/s

Steps to Ensure Driver Functionality:

- Different CPU and OS Setup: With another CPU and Ubuntu 18.04 - Linux 4.9, the 8822cu works fine, showing two wireless interfaces, allowing simultaneous hotspot and data connections.

- Alternative Driver: Using this driver, I achieved successful results on the previous setup: https://www.lm-technologies.com/product/wifi-802-11ac-bluetooth-5-0-2t2r-combi-usb-module-lm843/?template=driver
     - This driver does not work on my Linux 5.15.136 (Ubuntu 22.04)

- Issues and Questions:

    - Current Driver Setup: Currently, with the implemented driver, only wlan0 is available, and the above issues persist.
    Previous Setup Comparison: On the successful setup, two wlan interfaces were available, supporting both hotspot and data connections simultaneously.

- Request for Assistance:
    - Further Steps: What steps can I take to ensure the driver works correctly on my current setup?
    - Helpful Information: What additional information would be helpful to diagnose and resolve these issues?

Thank you in advance for your support!

Best regards,

morrownr, 2024-06-26:
@ADV-LUT 

Thanks for the report. It appears that the rtw 8822CU driver is not in good shape at all. The out-of-kernel driver you pointed me to is a very old driver that is not being maintained so no wonder it does not work on anything near a modern kernel. If your needs include something that works well in the short term, you may want to take a look at the Plug and Play List on the Main Menu:

https://github.com/morrownr/USB-WiFi

@dubhater

This message is in the Project Issue in the 882xBU repo. I am also going to channel users with 8822cu based adapters into here as I do not have a repo for the 8822cu... and I don't have an adapter. That the rtw 8822CU driver is not in good shape should come as no surprise. What are your thoughts on the 8822cu?

dubhater, 2024-06-26:
I think 8822cu, 8822bu, and 8821cu have a common problem. They use a common code base (the "halmac" stuff in the official driver) so it makes sense.

@ADV-LUT Unless you want to dive into the code and find what is missing from rtw88, there is nothing you can do to make it work better.

ADV-LUT, 2024-06-27:
@morrownr 

Thanks for the quick response. 

I was just looking for suitable adapters for the 8822cu and found this one: https://www.top-electronicsusa.com/lm842-8426-us-wifi-80211ac-bluetooth-50-2t2r-usb-combi-adapter-with-sma-antenna-connector-p-17843.html

If that's an option, we would order the adapter and send it to you.

Khnaz35, 2024-07-05:
@morrownr 
will this support for 
```$ uname -r
6.8.0-36-generic
```
The original driver which is no more supported can be found here.
[https://www.tp-link.com/us/support/download/archer-t2u/](url)

---

## Issue Report: TP-Link Archer T2U USB Wi-Fi Adapter Driver Installation on Kernel 6.8.0-36-generic

### Kernel Version
```bash
uname -r
# Output: 6.8.0-36-generic
```

### Summary
We attempted to install the driver for the TP-Link Archer T2U USB Wi-Fi adapter on a system running kernel version `6.8.0-36-generic`. Despite multiple efforts, the driver could not be loaded due to unresolved symbol errors related to `cfg80211` and `mac80211`. Below are the detailed steps taken, the commands used, and the errors encountered.

### Steps Taken

#### 1. Initial Setup
We updated the system and installed the required packages:
```bash
sudo apt update
sudo apt install -y dkms build-essential libelf-dev linux-headers-$(uname -r) git
```

#### 2. Clean Up Previous Installations
Removed any existing DKMS modules and cleaned up the build directory:
```bash
sudo dkms remove realtek-rtl88xxau/5.6.4.2~20240509 --all
cd ~/rtl8812au
sudo make clean
```

#### 3. Clone a Reliable Community Driver Repository
Cloned a known reliable driver repository:
```bash
cd ~
git clone https://github.com/morrownr/88x2bu.git
cd 88x2bu
```

#### 4. Build and Install the Driver Using DKMS
Added, built, and installed the DKMS module:
```bash
sudo dkms add .
sudo dkms build 88x2bu/5.8.7
sudo dkms install 88x2bu/5.8.7
```

#### 5. Load the Driver
Attempted to load the driver:
```bash
sudo modprobe 88x2bu
```

#### 6. Check for Errors
Checked `dmesg` for any errors:
```bash
sudo dmesg | tail
# Encountered the following errors:
# [ 3698.455187] 88XXau: Unknown symbol cfg80211_michael_mic_failure (err -2)
# [ 3698.455218] 88XXau: Unknown symbol cfg80211_external_auth_request (err -2)
# [ 3698.455235] 88XXau: Unknown symbol cfg80211_connect_done (err -2)
# [ 3698.455306] 88XXau: Unknown symbol wiphy_register (err -2)
# [ 3698.455321] 88XXau: Unknown symbol wiphy_free (err -2)
# [ 3698.455354] 88XXau: Unknown symbol ieee80211_channel_to_freq_khz (err -2)
# [ 3698.455388] 88XXau: Unknown symbol cfg80211_ch_switch_started_notify (err -2)
```

#### 7. Verify the Modules are Loaded
Ensured the necessary wireless stack modules (`cfg80211` and `mac80211`) were loaded:
```bash
sudo modprobe cfg80211
sudo modprobe mac80211
lsmod | grep cfg80211
lsmod | grep mac80211
# Output:
# cfg80211             1339392  1 mac80211
# mac80211             1728512  0
# libarc4                12288  1 mac80211
# cfg80211             1339392  1 mac80211
```

#### 8. Attempt to Load the Driver Again
Tried loading the driver again:
```bash
sudo modprobe 88x2bu
```

#### 9. Check for Errors Again
Checked `dmesg` for any new errors:
```bash
sudo dmesg | tail
# Same errors persisted:
# [ 3698.455187] 88XXau: Unknown symbol cfg80211_michael_mic_failure (err -2)
# [ 3698.455218] 88XXau: Unknown symbol cfg80211_external_auth_request (err -2)
# [ 3698.455235] 88XXau: Unknown symbol cfg80211_connect_done (err -2)
# [ 3698.455306] 88XXau: Unknown symbol wiphy_register (err -2)
# [ 3698.455321] 88XXau: Unknown symbol wiphy_free (err -2)
# [ 3698.455354] 88XXau: Unknown symbol ieee80211_channel_to_freq_khz (err -2)
# [ 3698.455388] 88XXau: Unknown symbol cfg80211_ch_switch_started_notify (err -2)
```

### Conclusion
Despite following all the steps to install and load the driver, the module failed to load due to missing symbols related to `cfg80211` and `mac80211`. This indicates a potential compatibility issue with the current kernel version `6.8.0-36-generic`. We recommend seeking further assistance from the community or driver maintainers to resolve these symbol issues.

### Logs and Additional Information
- **Kernel Version:** 6.8.0-36-generic
- **Driver Repository Used:** [morrownr/88x2bu](https://github.com/morrownr/88x2bu)
- **Error Messages:**
  ```plaintext
  88XXau: Unknown symbol cfg80211_sinfo_alloc_tid_stats (err -2)
  88XXau: Unknown symbol cfg80211_disconnected (err -2)
  88XXau: Unknown symbol cfg80211_michael_mic_failure (err -2)
  88XXau: Unknown symbol cfg80211_external_auth_request (err -2)
  88XXau: Unknown symbol cfg80211_connect_done (err -2)
  88XXau: Unknown symbol wiphy_register (err -2)
  88XXau: Unknown symbol wiphy_free (err -2)
  88XXau: Unknown symbol ieee80211_channel_to_freq_khz (err -2)
  88XXau: Unknown symbol cfg80211_ch_switch_started_notify (err -2)
  ```

We hope this detailed report helps in diagnosing and resolving the issue with the TP-Link Archer T2U USB Wi-Fi adapter driver on newer kernel versions. If you need any additional information, please feel free to reach out.

morrownr, 2024-07-05:
Hi @Khnaz35 

I do see some problems but this issue is for reports on the work to improve the rtw88 support for rtw88_8822bu/cu. So please keep that in mind and if more than one message is required, I will probably ask you to establish the bug report in another location.

> TP-Link Archer T2U

I think that adapter uses a rtl8811au chip but lets check:

$ lsusb

Post the result.

Tell us what distro you are using. I could not put enough puzzle pieces together to figure it out.

> sudo dkms remove realtek-rtl88xxau/5.6.4.2~20240509 --all

Well, that is not a 88x2bu driver you are removing so the mystery of what chip is in your adapter continues.

> git clone https://github.com/morrownr/88x2bu.git

That is the driver in this repo. It supports the 88x2bu chips.

> sudo dkms build 88x2bu/5.8.7

I can appreciate that you know how to manually use dkms but you did not read the README did you? The installations steps in the README are such that it makes it hard to mess up installing this or any out-of-kernel driver at this site. The driver version in this repo is not 5.8.7 and that is where things went south.

Let's start with the requested info so we can figure out what chip you have and what distro you are using. My best guess is that you adapter uses a rtl8811au chip. Let's figure out the chip and go from there.

@morrownr

Khnaz35, 2024-07-05:
@morrownr Thanks for getting back much appreciated.

Here are results

```$ sudo lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 003: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 005: ID 2357:011f TP-Link 802.11ac WLAN Adapter 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 222a:0001 ILI Technology Corp. Multi-Touch Screen
Bus 003 Device 003: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 003 Device 004: ID 0000:0538   USB OPTICAL MOUSE
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

```
$ dkms status
realtek-rtl88xxau/5.6.4.2~20240509, 6.8.0-31-generic, x86_64: installed (WARNING! Diff between built and installed module!)
```
```$ dkms --version
dkms-3.0.11
```
Tell us what distro you are using.

```
$ cat /etc/os-release
PRETTY_NAME="Ubuntu 24.04 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```

morrownr, 2024-07-05:
@Khnaz35 

> Bus 001 Device 005: ID 2357:011f TP-Link 802.11ac WLAN Adapter 

That ID means your adapter is based on the rtl8821/11au chip so this repo is the wrong driver.

> realtek-rtl88xxau/5.6.4.2~20240509, 6.8.0-31-generic, x86_64: installed

You need to uninstall that driver.

The correct driver repo is:

https://github.com/morrownr/8821au-20210708

Just follow the installation steps in the README.

There is an alternate if you are interested in testing a driver for this adapter that is in testing and should go into the Linux kernel soon:

https://github.com/morrownr/8821au-20210708/issues/133

Read and follow the first message in that thread. 

I'll see you over in the 8821au repo.

morrownr, 2024-07-14:
@ADV-LUT 

Sorry for the delay. Health issues have slowed me greatly lately.

> If that's an option, we would order the adapter and send it to you.

While I appreciate the offer, I don't have a good source for new Realtek vendor drivers for that chip. I have relationships with some adapter makers/retails that will pull and send new code to me at times but none of them make or sell adapters with the rtl8822cu chip. That chip came along very late cycle for WiFi 5 class chips and I could not see any real advantage over the rtl8822bu chip. The rtl8822cu could lose support in the long term due to having a small userbase so it is not a chip that I recommend to users. Not mention that Mediatek has blown pasted Realtek in driver quality and support. My recommendation is to not plan future usb wifi adapter and module purchased unless you check your options for Mediatek chip based devices.

What I do recommend can be found in menu items 1, 2 and 3 on the site Main Menu:

https://github.com/morrownr/USB-WiFi

ADV-LUT, 2024-07-15:
@morrownr 
Thanks also for the late reply.

Unfortunately I can't just use a different adapter because the chip is soldered onto the motherboard. This means I'm currently forced to get this chip working.

Thanks for the very detailed list, in the future we will take this into account when making our selection.

What other options do you see if I absolutely have to get the 8822cu working?

morrownr, 2024-07-15:
> What other options do you see if I absolutely have to get the 8822cu working?

There are options but first I need to know what chip you have. It would be very unusual for a board maker to solder a usb chip onto the mainboard. Let's investigate. Run the following commands and post the results:

$ lsusb
$ lspci

> Unfortunately I can't just use a different adapter because the chip is soldered onto the motherboard. This means I'm currently forced to get this chip working.

We can talk about this but first let's figure out exactly what you have.

ADV-LUT, 2024-07-16:
> $ lsusb

Bus 002 Device 002: ID 0424:7206 Microchip Technology, Inc. (formerly SMSC) USB7206 Smart Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 001 Device 010: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 009: ID 0424:724a Microchip Technology, Inc. (formerly SMSC) USB2 Controller Hub
### Bus 001 Device 007: ID 0bda:c82c Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 005: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 001 Device 004: ID 0424:9e00 Microchip Technology, Inc. (formerly SMSC) LAN9500A/LAN9500Ai
Bus 001 Device 002: ID 0424:4206 Microchip Technology, Inc. (formerly SMSC) USB4206 Smart Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

> $ lspci

0004:00:00.0 PCI bridge: NVIDIA Corporation Device 229c (rev a1)
0004:01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller 980
0008:00:00.0 PCI bridge: NVIDIA Corporation Device 229c (rev a1)
0008:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)

morrownr, 2024-07-16:
@ADV-LUT 

That is a usb chip ID using the rtl8822cu chip.

Okay, are you still using kernel 5.15?

There is a situation with more modern kernels that you need to be aware of. The ID of your device was inadvertently put in the modern Linux kernel (6.1+) for two different drivers which causes a conflict. I recently sent a patch to resolve this issue. That should not be an issue if you are using kernel 5.15.

Could you post the results of:

$ lsmod

What modes are you running?

AP, managed, other?

ADV-LUT, 2024-07-17:
Yes, I'm still using kernel version 5.15

I just re-flashed my board to test the latest version of the driver and now I can no longer install rtw88. But I was able to install the driver successfully before. Now I get the following error messages:

> adves@adves-orinNX:~/rtw88$ make
make -C /lib/modules/5.15.136-tegra/build M=/home/adves/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.136-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source'
  CC [M]  /home/adves/rtw88/rtw8723x.o
In file included from /home/adves/rtw88/rtw8723x.c:13:
/home/adves/rtw88/rtw8723x.h: In function ‘rtw8723x_iqk_path_adda_on’:
/home/adves/rtw88/rtw8723x.h:514:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  514 |         for (int i = 0; i < RTW8723X_IQK_ADDA_REG_NUM; i++)
      |         ^~~
/home/adves/rtw88/rtw8723x.h:514:9: note: use option ‘-std=c99’, ‘-std=gnu99’, ‘-std=c11’ or ‘-std=gnu11’ to compile your code
/home/adves/rtw88/rtw8723x.c: In function ‘__rtw8723x_debug_txpwr_limit’:
/home/adves/rtw88/rtw8723x.c:110:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  110 |         for (int i = 0; i < tx_path_count; i++)
      |         ^~~
/home/adves/rtw88/rtw8723x.c:123:18: error: redefinition of ‘i’
  123 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:110:18: note: previous definition of ‘i’ with type ‘int’
  110 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:123:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  123 |         for (int i = 0; i < tx_path_count; i++)
      |         ^~~
/home/adves/rtw88/rtw8723x.c:133:18: error: redefinition of ‘i’
  133 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:123:18: note: previous definition of ‘i’ with type ‘int’
  123 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:133:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  133 |         for (int i = 0; i < tx_path_count; i++)
      |         ^~~
/home/adves/rtw88/rtw8723x.c:145:18: error: redefinition of ‘i’
  145 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:133:18: note: previous definition of ‘i’ with type ‘int’
  133 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:145:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  145 |         for (int i = 0; i < tx_path_count; i++)
      |         ^~~
/home/adves/rtw88/rtw8723x.c:156:18: error: redefinition of ‘i’
  156 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:145:18: note: previous definition of ‘i’ with type ‘int’
  145 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:156:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  156 |         for (int i = 0; i < tx_path_count; i++)
      |         ^~~
/home/adves/rtw88/rtw8723x.c:167:18: error: redefinition of ‘i’
  167 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:156:18: note: previous definition of ‘i’ with type ‘int’
  156 |         for (int i = 0; i < tx_path_count; i++)
      |                  ^
/home/adves/rtw88/rtw8723x.c:167:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  167 |         for (int i = 0; i < tx_path_count; i++)
      |         ^~~
make[2]: *** [scripts/Makefile.build:295: /home/adves/rtw88/rtw8723x.o] Error 1
make[1]: *** [Makefile:1912: /home/adves/rtw88] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.136-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source'
make: *** [Makefile:166: all] Error 2

Here is my output for $lsmod:

[... 129 lines cut ...]
x_tables               49152  1 ip_tables
ipv6                  503808  115 bridge
pwm_fan                20480  0
pwm_tegra              20480  1
tegra_bpmp_thermal     16384  0
tegra_xudc             45056  0
ucsi_ccg               28672  0
typec_ucsi             36864  1 ucsi_ccg
typec                  61440  1 typec_ucsi
nvme                   49152  1
nvme_core             106496  2 nvme
phy_tegra194_p2u       16384  9
pcie_tegra194          40960  0

I had previously gained the following insights:

Usage as a Normal WiFi Connection:
- Single Connection Works: Initially connecting to a network works
- Network Switching Issue: Switching to another network fails.
- Reboot Fix: After a reboot, the new connection works.
- DHCP Issue: When the connection fails, the device gets an IP from the DHCP server, but even a ping to the router's address fails with "Destination Host Unreachable".
- Intermittent Boot Issue: Sometimes this error also appears on boot, but often it works fine after a reboot.

Usage as a Hotspot:
- Data Connection Drops: Both with nmcli and hostapd, I can establish a connection, but it drops after a few seconds.

ADV-LUT, 2024-07-17:
I got the driver running again. To do this, I added "EXTRA_CFLAGS += -std=gnu99" to the Makefile.

Here is the current $lsmod:

> Module                  Size  Used by
joydev                 32768  0
nvidia_modeset       1302528  5
lzo_rle                16384  64
lzo_compress           16384  1 lzo_rle
zram                   32768  16
zsmalloc               36864  1 zram
ramoops                28672  0
reed_solomon           20480  1 ramoops
algif_hash             20480  1
algif_skcipher         20480  1
af_alg                 32768  6 algif_hash,algif_skcipher
snd_soc_tegra186_asrc    40960  1
snd_soc_tegra210_mixer    49152  1
snd_soc_tegra210_admaif   131072  1
snd_soc_tegra_pcm      16384  1 snd_soc_tegra210_admaif
snd_soc_tegra186_arad    28672  2 snd_soc_tegra186_asrc
snd_soc_tegra210_afc    20480  6
snd_soc_tegra210_ope    36864  1
snd_soc_tegra210_mvc    20480  2
snd_soc_tegra186_dspk    20480  2
snd_soc_tegra210_dmic    20480  4
snd_soc_tegra210_sfc    61440  4
snd_soc_tegra210_adx    32768  4
snd_soc_tegra210_amx    36864  4
snd_soc_tegra210_i2s    28672  6
r8168                 503808  0
tegra210_adma          24576  0
snd_soc_tegra210_ahub  1273856  3 snd_soc_tegra210_ope,snd_soc_tegra210_sfc
nvvrs_pseq_rtc         16384  0
spidev                 28672  0
smsc                   16384  1
option                 57344  1
usb_wwan               24576  1 option
qmi_wwan               36864  0
cdc_wdm                28672  2 qmi_wwan
usbserial              45056  4 usb_wwan,option
rtw_8822cu             16384  0
rtw_8822c             479232  1 rtw_8822cu
rtw_usb                24576  1 rtw_8822cu
smsc95xx               36864  0
rtk_btusb              77824  0
btusb                  49152  0
rtw_core              225280  2 rtw_usb,rtw_8822c
usbnet                 45056  2 smsc95xx,qmi_wwan
btrtl                  24576  1 btusb
mac80211              864256  2 rtw_usb,rtw_core
btintel                45056  1 btusb
libarc4                16384  1 mac80211
btbcm                  28672  1 btusb
bluetooth             454656  22 btrtl,btintel,btbcm,btusb,rtk_btusb
cfg80211              851968  2 rtw_core,mac80211
ecdh_generic           16384  2 bluetooth
ecc                    36864  1 ecdh_generic
cdc_acm                36864  0
snd_soc_tegra_machine_driver    16384  0
snd_soc_tegra_utils    32768  2 snd_soc_tegra210_admaif,snd_soc_tegra_machine_driver
snd_soc_simple_card_utils    28672  1 snd_soc_tegra_utils
crct10dif_ce           20480  1
tegra234_oc_event      16384  0
nvpmodel_clk_cap       16384  0
tegra_cactmon_mc_all    16384  0
mttcan                 69632  0
thermal_trip_event     16384  0
nvpps                  32768  1 mttcan
can_dev                40960  1 mttcan
[... 89 lines cut ...]
- However, if I disconnect the WLAN and then reconnect it, it no longer works, a ping shows the following:

> ping -I wlan0 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.99.59 wlan0: 56(84) bytes of data.
From 192.168.99.59 icmp_seq=1 Destination Host Unreachable
From 192.168.99.59 icmp_seq=2 Destination Host Unreachable
From 192.168.99.59 icmp_seq=3 Destination Host Unreachable
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 0 received, +3 errors, 100% packet loss, time 4079ms

- AP mode sets up a WLAN network and you can connect to it and you get an IP address, but it always breaks off after a few seconds and a functioning, permanent network connection could never be established.

lshw shows the following:

> *-usb:3
             description: Bluetooth wireless interface
             product: 802.11ac NIC
             vendor: Realtek
             physical id: 4
             bus info: usb@1:2.4
             version: 0.00
             serial: 123456
             capabilities: usb-2.00 bluetooth
             configuration: driver=rtw_8822cu maxpower=500mA speed=480Mbit/s

dubhater, 2024-07-22:
@ADV-LUT The compilation problem should be fixed now. I enabled RX aggregation yesterday. Could you retest station mode, just to make sure it still works?

@morrownr Could you retest with RTL8812BU, for the same reason?

dubhater, 2024-07-22:
Nevermind for now. There is nothing to test because it's not actually working.

morrownr, 2024-07-22:
I had just started testing and could not even get an interface. Was just starting to try to figure out why. I'll standby.

dubhater, 2024-07-22:
That's not supposed to happen. It should still work like before, just with no Rx aggregation.

morrownr, 2024-07-22:
> That's not supposed to happen.

Famous last words.

Sorry, had to recognize the humor in the situation.

dubhater, 2024-07-22:
Indeed. I don't know what happened. You may have to bisect it.

The RX aggregation is ready for testing now.

morrownr, 2024-07-22:
@dubhater 

> Could you retest with RTL8812BU, for the same reason?

Just finished test rtw_8822bu with the latest:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.135 port 35168 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  65.1 MBytes   546 Mbits/sec    0    977 KBytes       
[  5]   1.00-2.00   sec  68.8 MBytes   577 Mbits/sec    0    977 KBytes       
[  5]   2.00-3.00   sec  67.5 MBytes   566 Mbits/sec    0    977 KBytes       
[  5]   3.00-4.00   sec  67.5 MBytes   566 Mbits/sec    0    977 KBytes       
[  5]   4.00-5.00   sec  66.2 MBytes   556 Mbits/sec    0    977 KBytes       
[  5]   5.00-6.00   sec  68.8 MBytes   577 Mbits/sec    0    977 KBytes       
[  5]   6.00-7.00   sec  66.2 MBytes   556 Mbits/sec    0    977 KBytes       
[  5]   7.00-8.00   sec  66.2 MBytes   556 Mbits/sec    0    977 KBytes       
[  5]   8.00-9.00   sec  66.2 MBytes   556 Mbits/sec    0    977 KBytes       
[  5]   9.00-10.00  sec  66.2 MBytes   556 Mbits/sec    0    977 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   669 MBytes   561 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   666 MBytes   558 Mbits/sec                  receiver

iperf Done.

```

Your efforts to increase managed mode performance have worked. That is outstanding for an AC1200 class wifi adapter.

I did extra testing on usb:

```
Class=Vendor Specific Class, Driver=rtw_8822bu, 5000M
```

I tested in usb2 ports and usb3 ports. Not seeing any problems.

I'll look around the log and check back in.

Are we still pending Realtek help to fix something that causes problems with AP mode?

morrownr, 2024-07-22:
@dubhater 

This finishes up my current testing on rtw_8822 in managed mode:

```
$ sudo dmesg | grep 8822
   
[    1.188226] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    5.485617] rtw_8822bu 6-4.3:1.0: Firmware version 27.2.0, H2C version 13
[    5.730839] usbcore: registered new interface driver rtw_8822bu
[    5.735006] rtw_8822bu 6-4.3:1.0 wlx00c0caadde9e: renamed from wlan0

```

Clean log. Speed is excellent. Super stable. I can support you upstreaming the needed patches.

Tests accomplished on Debian 12 with kernel 6.9.

morrownr, 2024-07-22:
@dubhater 

I've continued testing rtw_8822bu. I ran into the following which is a little concerning:

```
$ iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.135 port 44658 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  29.0 MBytes   244 Mbits/sec                  
[  5]   1.00-2.00   sec  12.2 MBytes   103 Mbits/sec                  
[  5]   2.00-3.00   sec  5.20 MBytes  43.6 Mbits/sec                  
[  5]   3.00-4.00   sec  8.42 MBytes  70.6 Mbits/sec                  
[  5]   4.00-5.00   sec  8.05 MBytes  67.6 Mbits/sec                  
[  5]   5.00-6.00   sec  7.90 MBytes  66.3 Mbits/sec                  
[  5]   6.00-7.00   sec  7.86 MBytes  65.9 Mbits/sec                  
[  5]   7.00-8.00   sec  7.90 MBytes  66.3 Mbits/sec                  
[  5]   8.00-9.00   sec  7.75 MBytes  65.0 Mbits/sec                  
[  5]   9.00-10.00  sec  7.86 MBytes  65.9 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   103 MBytes  86.3 Mbits/sec   29             sender
[  5]   0.00-10.00  sec   102 MBytes  85.7 Mbits/sec                  receiver

iperf Done.

```

morrownr, 2024-07-23:
@dubhater 

As I continue testing my rtl8812bu based adapter with rtw8822bu, I ran into some noise in the log after leaving the system idle while being away for around an hour. When I came back, the connection was down and the following was in the log:

```
$ sudo dmesg | grep 8822
       
[    1.188226] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    5.485617] rtw_8822bu 6-4.3:1.0: Firmware version 27.2.0, H2C version 13
[    5.730839] usbcore: registered new interface driver rtw_8822bu
[    5.735006] rtw_8822bu 6-4.3:1.0 wlx00c0caadde9e: renamed from wlan0
[ 5954.450691] rtw_8822bu 6-4.3:1.0: read register 0x23c failed with -71
[ 5954.454205] rtw_8822bu 6-4.3:1.0: read register 0x23e failed with -71
[ 5954.482702] rtw_8822bu 6-4.3:1.0: read register 0x23c failed with -71
[ 5954.486172] rtw_8822bu 6-4.3:1.0: read register 0x23e failed with -71
[ 5955.602636] rtw_8822bu 6-4.3:1.0: write register 0x674 failed with -71
[ 5955.606147] rtw_8822bu 6-4.3:1.0: write register 0x670 failed with -71
[ 5955.613108] rtw_8822bu 6-4.3:1.0: write register 0x1fc failed with -71
[ 5955.616650] rtw_8822bu 6-4.3:1.0: write register 0x1dc failed with -71
[ 5956.023648] rtw_8822bu 6-4.3:1.0: failed to do USB write, ret=-2
[ 5956.023658] rtw_8822bu 6-4.3:1.0: failed to send h2c packet
[ 5964.458768] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5964.665247] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5964.672248] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5964.770214] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5964.777212] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.293165] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.303664] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.314163] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.321162] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.328161] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.335161] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.342161] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.349159] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.356159] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.370158] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.377157] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.384156] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.391157] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.398155] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.405155] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.412154] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.419156] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.426153] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.433152] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.440151] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.447153] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.454150] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.461149] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.468150] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.475150] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.482147] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.489148] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.496146] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.503146] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.510145] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.517144] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.524145] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.531143] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.538143] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.545142] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5965.552141] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5966.141087] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5966.151586] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5966.162085] rtw_8822bu 6-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 5966.995781] rtw_8822bu 5-4.3:1.0: Firmware version 27.2.0, H2C version 13
[ 5967.220194] rtw_8822bu 5-4.3:1.0 wlx00c0caadde9e: renamed from wlan0
[ 8986.851781] rtw_8822bu 5-4.3:1.0: Firmware version 27.2.0, H2C version 13
[ 8987.092667] rtw_8822bu 5-4.3:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
```

dubhater, 2024-07-23:
That's interesting. Can I see the unfiltered dmesg, please?

ADV-LUT, 2024-07-23:
@dubhater 
The driver compilation now works fine.
I have now tested the driver further and have gathered the following information:
1. If I connect to a WiFi during operation, I get the following error in dmesg:
```
$ sudo dmesg | grep 8822
[   10.480004] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000c lmp_ver=0a lmp_subver=8822
[   10.480995] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_fw.bin
[   10.481838] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_config.bin
[   10.520177] rtw_8822cu 1-2.4:1.2: WOW Firmware version 9.9.4, H2C version 15
[   10.520246] rtw_8822cu 1-2.4:1.2: Firmware version 9.9.15, H2C version 15
[   13.127154] usbcore: registered new interface driver rtw_8822cu
[  177.348146] rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware
```

2. If I automatically connect to the network during a reboot, station mode works. Dmesg if it works:
```
$ sudo dmesg | grep 8822
[   10.414692] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000c lmp_ver=0a lmp_subver=8822
[   10.415577] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_fw.bin
[   10.416467] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_config.bin
[   10.466831] rtw_8822cu 1-2.4:1.2: WOW Firmware version 9.9.4, H2C version 15
[   10.466942] rtw_8822cu 1-2.4:1.2: Firmware version 9.9.15, H2C version 15
[   13.054744] usbcore: registered new interface driver rtw_8822cu
```

When I reconnect, the following error message appears again in the dmesg and nothing works:
```
[  346.034133] rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware
```

dubhater, 2024-07-23:
@ADV-LUT Did it ever work better? Can you try an older commit, like 1086d09?

ADV-LUT, 2024-07-23:
@dubhater I tested 1086d09 and compared it with the current version and can hardly see any differences. It seems as if I can reconnect faster with the current commit. However, I get the same error messages in both versions.

ADV-LUT, 2024-07-23:
That sounds great, we are in Germany. We don't have an adapter ourselves either, because the chip is soldered onto the circuit board.

However, we can offer to order this adapter and send it to you. According to the data sheet, this adapter contains the rtl8822cu:
https://de.farnell.com/en-DE/lm-technologies/lm842-8425/usb-adapter-ble-5-0-1-3mbps-5v/dp/4416535

What is a safe way to exchange contact details?

dubhater, 2024-07-23:
Does email work? Mine is rtl8821cerfe2 at gmail.com.

henkv1, 2024-07-25:
AP mode improved a lot with the latest commit https://github.com/lwfinger/rtw88/commit/4a1ee64475672a9186e10d87ac5e251448f86914. But I still observe these errors:
```
[Wed Jul 24 18:13:00 2024] rtw_8822bu 1-1.3:1.0: error beacon valid
[Wed Jul 24 18:13:00 2024] rtw_8822bu 1-1.3:1.0: failed to download drv rsvd page
[Wed Jul 24 18:13:00 2024] rtw_8822bu 1-1.3:1.0: failed to download beacon
[Wed Jul 24 18:16:08 2024] rtw_8822bu 1-1.3:1.0: error beacon valid
[Wed Jul 24 18:16:08 2024] rtw_8822bu 1-1.3:1.0: failed to download drv rsvd page
[Wed Jul 24 18:20:25 2024] rtw_8822bu 1-1.3:1.0: error beacon valid
[Wed Jul 24 18:20:25 2024] rtw_8822bu 1-1.3:1.0: failed to download drv rsvd page
[Wed Jul 24 18:29:25 2024] rtw_8822bu 1-1.3:1.0: error beacon valid
[Wed Jul 24 18:29:25 2024] rtw_8822bu 1-1.3:1.0: failed to download drv rsvd page
```

WallyZambotti, 2024-07-26:
See my response regarding slow bit rate in mainline and downstream driver :
https://github.com/lwfinger/rtw88/issues/224

dubhater, 2024-08-05:
@ADV-LUT I got the LM842 today, thank you!

Bad news, though: it's working perfectly so far with my Arch Linux system. Tomorrow I will try with Ubuntu 22.04.

dubhater, 2024-08-06:
Ubuntu 22.04 with kernel 5.15 also works. I can switch between my 2.4 GHz and 5 GHz networks multiple times.

WallyZambotti, 2024-08-07:
@dubhater : Can you confirm (when you say working) that there is no bit rate degradation when under network load.

I use this simple continuous monitor command to watch the rx/tx bit rates:

while true; do iw <wandev> link; sleep 1; done

(Substitute <wandev> for your wan device.  I have have been using speed test sites for testing. )

I have noticed under load both receiving and transmitting the RX bit rate drops very low and the TX bit rate remains fairly stable.

So only the RX bit rate drops when both receiving or transmitting.

dubhater, 2024-08-07:
@WallyZambotti  Yes, the RX speed is stable here with RTL8822CU.

WallyZambotti, 2024-08-09:
@dubhater Can you confirm I am using the correct repo :

```
[remote "origin"]
	url = https://github.com/lwfinger/rtw88
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master

```

dubhater, 2024-08-09:
@WallyZambotti That looks correct.

WallyZambotti, 2024-08-12:
The latest update of the driver seems to be unstable and the network disappears with "Device not ready".  It will work for a limited time after a reboot but then fails.  lsusb and iw dev show the device is still there.

dubhater, 2024-08-13:
@WallyZambotti Does an older version of the driver work better?

How long does it work before it fails?

Can you show the output of ``journalctl -b`` after it fails?

WallyZambotti, 2024-08-13:
Not sure how long it is lasting.  Seems to be dying over night.   I might check the power settings to see if this system is going to sleep.

If you need the journalctl output after a failure then we will have to wait until that happens again, until then here is segment of the messages being displayed :

```
Aug 13 08:32:30 WZ-MS-7B51 kernel: rtw_8822bu 2-6:1.0: failed to get tx report from firmware
Aug 13 08:33:00 WZ-MS-7B51 kernel: rtw_8822bu 2-6:1.0: failed to get tx report from firmware
Aug 13 08:33:24 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:33:30 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:34:27 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:35:52 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:36:02 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:40:29 WZ-MS-7B51 kernel: rtw_8822bu 2-6:1.0: failed to get tx report from firmware
Aug 13 08:40:34 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:41:14 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:41:21 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:43:22 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:43:36 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:45:40 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:45:42 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:45:58 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:16 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:20 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:22 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:23 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:27 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:30 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:32 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:42 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:46:56 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:48:01 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:48:01 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:48:04 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:48:32 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:48:44 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:49:04 WZ-MS-7B51 kernel: workqueue: output_poll_execute [drm_kms_helper] hogged CPU for >10000us 256 times, consider switching to WQ_UNBOUND
Aug 13 08:49:05 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:49:15 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:49:25 WZ-MS-7B51 kernel: workqueue: i915_hpd_poll_init_work [i915] hogged CPU for >10000us 256 times, consider switching to WQ_UNBOUND
Aug 13 08:49:40 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:49:41 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:52:34 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:54:27 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:54:39 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:54:57 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:55:30 WZ-MS-7B51 kernel: rtw_8822bu 2-6:1.0: failed to get tx report from firmware
Aug 13 08:56:06 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:56:21 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:56:22 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:56:27 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:56:40 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:57:37 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 08:57:50 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
[... cut ...]
Aug 13 09:11:23 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 09:15:33 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 09:16:02 WZ-MS-7B51 wpa_supplicant[903]: wlxbcec4322ff20: WPA: Group rekeying completed with 74:83:c2:b5:56:ca [GTK=CCMP]
Aug 13 09:16:08 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26
Aug 13 09:16:32 WZ-MS-7B51 wpa_supplicant[903]: TDLS: Invalid frame - payloadtype=1 category=240 action=26

```

Do you want the full output, regardless?

morrownr, 2024-08-13:
@WallyZambotti 

I am running a 8812bu adapter with rtw_8822bu right now:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.135 port 54978 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  70.6 MBytes   592 Mbits/sec    0    976 KBytes       
[  5]   1.00-2.00   sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
[  5]   2.00-3.00   sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
[  5]   3.00-4.00   sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
[  5]   4.00-5.00   sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
[  5]   5.00-6.00   sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
[  5]   6.00-7.00   sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
[  5]   7.00-8.00   sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
[  5]   8.00-9.00   sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
[  5]   9.00-10.00  sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   672 MBytes   564 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   670 MBytes   561 Mbits/sec                  receiver

```

Can you give me some details on what you are doing when you see the performance issue? I'll try to duplicate.

I git pulled earlier today.

morrownr, 2024-08-13:
Ouch:

```
$ iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.135 port 47964 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  15.0 MBytes   126 Mbits/sec                  
[  5]   1.00-2.00   sec  7.52 MBytes  63.1 Mbits/sec                  
[  5]   2.00-3.00   sec  7.60 MBytes  63.8 Mbits/sec                  
[  5]   3.00-4.00   sec  7.61 MBytes  63.8 Mbits/sec                  
[  5]   4.00-5.00   sec  7.45 MBytes  62.5 Mbits/sec                  
[  5]   5.00-6.00   sec  7.44 MBytes  62.4 Mbits/sec                  
[  5]   6.00-7.00   sec  7.51 MBytes  63.0 Mbits/sec                  
[  5]   7.00-8.00   sec  7.65 MBytes  64.2 Mbits/sec                  
[  5]   8.00-9.00   sec  7.54 MBytes  63.2 Mbits/sec                  
[  5]   9.00-10.00  sec  7.63 MBytes  64.0 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  84.2 MBytes  70.7 Mbits/sec   34             sender
[  5]   0.00-10.00  sec  82.9 MBytes  69.6 Mbits/sec                  receiver

```

WallyZambotti, 2024-08-13:
> @WallyZambotti
>
> I am running a 8812bu adapter with rtw_8822bu right now:
>
> ```
> $ iperf3 -c 192.168.1.1
> Connecting to host 192.168.1.1, port 5201
> [  5] local 192.168.1.135 port 54978 connected to [...]

Basically occurs with any net traffic.  I usually test with a speed test site.

morrownr, 2024-08-13:
@WallyZambotti 

> Basically occurs with any net traffic. I usually test with a speed test site.

I used:

https://testmy.net/

Results: (download test)

rtw_8822bu  -  40.5 Mbps
mt7921u  - 524.3 Mbps

Geez... I see what you mean. Yes, the mt7921u is a WiFi 6 class driver but my internet service is limited to 500 Mbps and really, the number should not be that different. Test conducted on the same desktop system with both wifi devices on the same system running Debian 12 x86_64.

@dubhater 

I have a package nearly ready to send. Do you want a rtl8812bu adapter?

morrownr, 2024-08-15:
@WallyZambotti 

> Are you able to test both those adapters on the same hardware using Windows to confirm they do get roughly the same performance?

Well, I've never been a Windows user but I will uninstall rtw_8822bu and replace it with the out-of-kernel driver here in this repo to see what we get a little later.

morrownr, 2024-08-15:
Testing site. Same as before:

https://testmy.net/

New results: (download test)

88x2bu - 523.7 Mbps (this is the vendor driver from this repo)
mt7921u - 528.0 Mbps

The vendor driver is smoking rtw_8822bu.

WallyZambotti, 2024-08-15:
@morrownr : Explain how you enabled the "vendor" driver.  If it that good why isn't it the default?

morrownr, 2024-08-16:
@WallyZambotti 

> Explain how you enabled the "vendor" driver.

I first uninstalled the rtw88 driver that we are testing, then I installed the driver in this repo.

> If it is that good why isn't it the default?

Good question. Realtek's usb team, in their infinite wisdom, has not moved over to fully standards compliance with their usb  drivers so the driver in this repo will not be allowed into the Linux kernel.

The rtw_8822bu driver can be as fast as 88x2bu. We just need to figure out what the issue is. It would be good if @dubhater had a rtl8812bu adapter so he can directly test but he knows he can ask me to test and provide results. He has some adapter that I do not and vice versa. In fact @dubhater is pretty sharp. I'll bet the numbers that I posted give him an idea of what is not correct.

WallyZambotti, 2024-08-16:
Maybe I'm using the wrong driver because I can't find any definitions for : 

Device 0xb812

```
57: USB 00.0: 0282 WLAN controller
  [Created at usb.122]
  Unique ID: R_O4.cPKg_gaSCLF
  Parent ID: pBe4.xYNhIwdOaa6
  SysFS ID: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0
  SysFS BusID: 2-6:1.0
  Hardware Class: network
  Model: "Realtek USB3.0 802.11ac 1200M Adapter"
  Hotplug: USB
  Vendor: usb 0x0bda "Realtek Semiconductor Corp."
  Device: usb 0xb812 "USB3.0 802.11ac 1200M Adapter"
  Revision: "3.00"
  Serial ID: "123456"
  Driver: "rtw_8822bu"
  Driver Modules: "rtw_8822bu"
  Device File: wlxbcec4322ff20
  Features: WLAN
  HW Address: bc:ec:43:22:ff:20
  Permanent HW Address: bc:ec:43:22:ff:20
  Link detected: yes
  WLAN channels: 1 2 3 4 5 6 7 8 9 10 11 12 13 36 40 44 48 52 56 60 64 100 104 108 112 116 132 136 140 144 149 153
  WLAN frequencies: 2.412 2.417 2.422 2.427 2.432 2.437 2.442 2.447 2.452 2.457 2.462 2.467 2.472 5.18 5.2 5.22 5.24 5.26 5.28 5.3 5.32 5.5 5.52 5.54 5.56 5.58 5.66 5.68 5.7 5.72 5.745 5.765
  WLAN encryption modes: WEP40 WEP104 TKIP CCMP
  WLAN authentication modes: open sharedkey wpa-psk wpa-eap
  Module Alias: "usb:v0BDApB812d0300dc00dsc00dp00icFFiscFFipFFin00"
  Driver Info #0:
    Driver Status: rtw88_8822bu is not active
    Driver Activation Cmd: "modprobe rtw88_8822bu"
  Driver Info #1:
    Driver Status: rtw_8822bu is active
    Driver Activation Cmd: "modprobe rtw_8822bu"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #61 (Hub)

```

dubhater, 2024-08-16:
>  I'll bet the numbers that I posted give him an idea of what is not correct.

The numbers from iw tell me what to investigate. The AP is transmitting at a low rate, most likely because the AP thinks RTL8812BU doesn't hear a lot of what the AP is transmitting.

>  It would be good if @dubhater had a rtl8812bu adapter so he can directly test but he knows he can ask me to test and provide results.

Having the adapter makes it a lot easier. Debugging by proxy is not fun.

morrownr, 2024-08-17:
@dubhater 

My wife and I are dealing with health issues currently and it is slowing my activity here at my github site greatly. I may not be available much at all over the next week.

While I have an extra adapter with a rtl8812bu chip, the shipping from my location is rather expensive so if it was a situation where I was shipping several adapters, I could probably justify the expense. I'll check in with you via email to see what we can work out.

mikhailnov, 2024-10-10:
@morrownr I have compared list of devices supported by the in-kernel driver and this one and see that 2 devices are not supported by the upstream driver:

```bash
$ diff -u --color <(modinfo /lib/modules/6.6.47-generic-1rosa2021.1-x86_64/kernel/net/wireless/88x2bu.ko.zst | grep ^alias | sort -u) <(modinfo /lib/modules/6.11.2-generic-1rosa2021.1-x86_64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822bu.ko.zst | grep ^alias | sort -u)
```

```diff
--- /dev/fd/63  2024-10-10 14:12:41.720801602 +0300
+++ /dev/fd/62  2024-10-10 14:12:41.721801620 +0300
@@ -1,5 +1,4 @@
 alias:          usb:v04CAp8602d*dc*dsc*dp*icFFiscFFipFFin*
-alias:          usb:v056Ep4011d*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v0846p9055d*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v0B05p1841d*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v0B05p184Cd*dc*dsc*dp*icFFiscFFipFFin*
@@ -16,7 +15,6 @@
 alias:          usb:v2001p331Cd*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v2001p331Ed*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v2001p331Fd*dc*dsc*dp*icFFiscFFipFFin*
-alias:          usb:v20F4p805Ad*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v20F4p808Ad*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v2357p0115d*dc*dsc*dp*icFFiscFFipFFin*
 alias:          usb:v2357p0116d*dc*dsc*dp*icFFiscFFipFFin*
```

morrownr, 2024-10-10:
Hi @mikhailnov 

> I have compared list of devices supported by the in-kernel driver and this one and see that 2 devices are not supported by the upstream driver...

Let me see if I am understanding correctly. You are saying that the out-of-kernel driver in this repo has two more vid-pids than does the in-kernel rtw88_8822bu ? If that is correct, I am aware of this. It is on a list of things to do that I keep. Would you be interested in helping me put together a patch to submit to linux-wireless?

mikhailnov, 2024-10-10:
Yes, correct, thanks for reply! I can try to help, which help is needed? Unfortunately I do not have such hardware to test.

mikhailnov, 2024-10-10:
Is just adding vid-pids into the in-kernel driver not enough?

morrownr, 2024-10-10:
> I can try to help, which help is needed?

I can put together the patch which we will send in to linux-wireless to be merged into the kernel, however, I need help verifying that what we are doing is correct.

> Is just adding vid-pids into the in-kernel driver not enough?

It is as long as things are correct. I accidently made a mistake several months ago and added a 8822cu vid-pid to the 8821cu driver as part of a large patch. I have since sent in a patch to correct the error but we need to be very careful so if you can check my work, that would be great.

mikhailnov, 2024-10-10:
Of course I can :-) Feel free to write here or at m.novosyolov@rosalinux.ru

WallyZambotti, 2024-10-12:
I can report after the last update from github (https:/github.com/lwfinger/rtw88) that the bit rate dropping under load issue seems to have been fixed.  My RX rate is now the same or better than the TX rate and latest speed test to 2.4 GHZ router (114.5 Mbps download, 109.1 Mbps upload).  Since the connection is capped at 100 Mbps that looks perfect.

[8 more replies]