morrownr/88x2bu-20210702 #225: Request: Add support for Mercusys MA30H

nicolascolla, 2024-07-11:
Hello, @morrownr and thank you very much for maintaining this fork.

I have a Mercusys MA30H, and the official drivers page (https://www.mercusys.com/en/download/ma30h/v2) offers the sources of the RTL88x2BU driver (that fail to build on newer kernels), so I'm pretty sure this is a RTL88x2BU device. However, after building the driver from this repo, the Wi-Fi dongle is still not recognized.

dmesg output:
```
[  642.056301] usb 1-1.2: new high-speed USB device number 4 using ehci-pci
[  642.166212] usb 1-1.2: New USB device found, idVendor=2c4e, idProduct=0107, bcdDevice= 2.10
[  642.166230] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  642.166237] usb 1-1.2: Product: 802.11ac NIC
[  642.166241] usb 1-1.2: Manufacturer: Realtek
[  642.166245] usb 1-1.2: SerialNumber: 123456
```

lsusb output:
```
Bus 001 Device 004: ID 2c4e:0107 Mercucys INC 802.11ac NIC

```

Taking a look at the /os_dep/linux/usb_intf.c of the source code provided by the manufacturer in the previous link, I can see that there are additional adapters included there. However, I have no idea of where to start to make it work in this codebase.

Could support be added for this USB dongle?

Thank you very much!

morrownr, 2024-07-11:
Hi.

Support can be added. I am working on it. Hopefully I have posted later today.

morrownr, 2024-07-12:
@nicolascolla 

Requested support has been added but I need you to test and confirm that the driver does now work on your adapter... or I will need to remove the support. Here is the easiest way to recompile:

$ git pull
$ sudo sh install-driver.sh

Once you reboot ,,, let me know.

@morrownr

nicolascolla, 2024-07-12:
@morrownr, I can confirm the dongle works wonderfully after the last commit.

Thank you very much!

closed 2024-07-12 by nicolascolla