morrownr/88x2bu-20210702 #228: Issues on wake up from suspend

kandrelczyk, 2024-08-07:
Hi,

My system freezes on wake up from time to time and what I see in syslog is:

```
2024-08-06T18:49:01.234844+02:00 c295 kernel: CPU: 1 PID: 1257 Comm: wpa_supplicant Tainted: G           OE      6.9.12-amd64 #1  Debian 6.9.12-1
2024-08-06T18:49:01.234844+02:00 c295 kernel: Hardware name: Micro-Star International Co., Ltd. MS-7C96/B550M-A PRO (MS-7C96), BIOS 2.50 07/01/2021
2024-08-06T18:49:01.234844+02:00 c295 kernel: RIP: 0010:usb_submit_urb+0x616/0x630 [usbcore]
2024-08-06T18:49:01.234845+02:00 c295 kernel: Code: be 5b c0 e8 7c 47 15 d9 eb 96 b8 fe ff ff ff e9 45 fd ff ff 48 89 fe 48 c7 c7 60 e9 5d c0 c6 05 93 b1 01 00 01 e8 ca d3 b1 d8 <0f> 0b e9 4d ff ff ff b8 f8 ff ff ff e9 1e fd ff ff 66 0f 1f 84 00
2024-08-06T18:49:01.234845+02:00 c295 kernel: RSP: 0018:ffffa5fc0174b330 EFLAGS: 00010282
2024-08-06T18:49:01.234845+02:00 c295 kernel: RAX: 0000000000000000 RBX: ffff88b6aac15000 RCX: 0000000000000027
2024-08-06T18:49:01.234846+02:00 c295 kernel: RDX: ffff88bc91aa1708 RSI: 0000000000000001 RDI: ffff88bc91aa1700
2024-08-06T18:49:01.234846+02:00 c295 kernel: RBP: ffffa5fc0082d000 R08: 0000000000000000 R09: 0000000000000003 
2024-08-06T18:49:01.234846+02:00 c295 kernel: R10: ffffa5fc0174b1c0 R11: ffffffff9acca3a8 R12: ffff88b6b3a1c000
2024-08-06T18:49:01.234847+02:00 c295 kernel: R13: ffff88b705367c80 R14: ffff88b68d093800 R15: ffff88b6baa70048
2024-08-06T18:49:01.234847+02:00 c295 kernel: FS:  00007fb67e9c7840(0000) GS:ffff88bc91a80000(0000) knlGS:0000000000000000
2024-08-06T18:49:01.234847+02:00 c295 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
2024-08-06T18:49:01.234847+02:00 c295 kernel: CR2: 00007fc84c0060b8 CR3: 000000013aa68000 CR4: 0000000000750ef0
2024-08-06T18:49:01.234848+02:00 c295 kernel: PKRU: 55555554
2024-08-06T18:49:01.234848+02:00 c295 kernel: Call Trace:
2024-08-06T18:49:01.234848+02:00 c295 kernel:  
2024-08-06T18:49:01.234849+02:00 c295 kernel:  ? __warn+0x80/0x120  
2024-08-06T18:49:01.234849+02:00 c295 kernel:  ? usb_submit_urb+0x616/0x630 [usbcore] 
2024-08-06T18:49:01.234849+02:00 c295 kernel:  ? report_bug+0x164/0x190
2024-08-06T18:49:01.234849+02:00 c295 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
2024-08-06T18:49:01.234850+02:00 c295 kernel:  ? handle_bug+0x3c/0x80
2024-08-06T18:49:01.234850+02:00 c295 kernel:  ? exc_invalid_op+0x17/0x70
2024-08-06T18:49:01.234850+02:00 c295 kernel:  ? asm_exc_invalid_op+0x1a/0x20
2024-08-06T18:49:01.234851+02:00 c295 kernel:  ? usb_submit_urb+0x616/0x630 [usbcore]
2024-08-06T18:49:01.234851+02:00 c295 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
2024-08-06T18:49:01.234851+02:00 c295 kernel:  usb_read_port+0x128/0x1e0 [88x2bu]
2024-08-06T18:49:01.234851+02:00 c295 kernel:  ? __pfx_usb_read_port+0x10/0x10 [88x2bu]
2024-08-06T18:49:01.234852+02:00 c295 kernel:  rtl8822bu_inirp_init+0x67/0xb0 [88x2bu]
2024-08-06T18:49:01.234852+02:00 c295 kernel:  rtw_hal_inirp_init+0x15/0x20 [88x2bu]
2024-08-06T18:49:01.234852+02:00 c295 kernel:  usb_intf_start+0x16/0x30 [88x2bu]
2024-08-06T18:49:01.234853+02:00 c295 kernel:  rtw_intf_start+0x1e/0x40 [88x2bu]
2024-08-06T18:49:01.234853+02:00 c295 kernel:  _halmac_init_hal+0xb1/0x150 [88x2bu]
2024-08-06T18:49:01.234853+02:00 c295 kernel:  rtw_halmac_init_hal_fw+0xe/0x20 [88x2bu] 
2024-08-06T18:49:01.234853+02:00 c295 kernel:  rtl8822b_hal_init+0x33/0x70 [88x2bu]
2024-08-06T18:49:01.234854+02:00 c295 kernel:  rtl8822b_init+0x1a/0x70 [88x2bu]
2024-08-06T18:49:01.234854+02:00 c295 kernel:  rtl8822bu_init+0x1b/0x70 [88x2bu]
2024-08-06T18:49:01.234854+02:00 c295 kernel:  rtw_hal_init+0x39/0x110 [88x2bu] 
2024-08-06T18:49:01.234854+02:00 c295 kernel:  ips_netdrv_open+0x31/0x70 [88x2bu]
2024-08-06T18:49:01.234855+02:00 c295 kernel:  rtw_ips_pwr_up+0x23/0x40 [88x2bu]
2024-08-06T18:49:01.234855+02:00 c295 kernel:  _ips_leave+0x4b/0x80 [88x2bu]
2024-08-06T18:49:01.234855+02:00 c295 kernel:  ips_leave+0x2b/0x70 [88x2bu]
2024-08-06T18:49:01.234855+02:00 c295 kernel:  _rtw_pwr_wakeup+0x1e2/0x220 [88x2bu]
2024-08-06T18:49:01.234856+02:00 c295 kernel:  cfg80211_rtw_scan+0x41b/0x660 [88x2bu]
2024-08-06T18:49:01.234856+02:00 c295 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
2024-08-06T18:49:01.234856+02:00 c295 kernel:  ? schedule_hrtimeout_range_clock+0xd2/0x140
2024-08-06T18:49:01.234856+02:00 c295 kernel:  rdev_scan+0x48/0xf0 [cfg80211]
2024-08-06T18:49:01.234857+02:00 c295 kernel:  nl80211_trigger_scan+0x48e/0x750 [cfg80211]
```

I'm on `Linux c295 6.9.12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.9.12-1 (2024-07-27) x86_64 GNU/Linux`

I'm using this adapter: https://www.brostrend.com/products/ac3l
With drivers installed from here: https://linux.brostrend.com/

Can someone decode this log? I'm not even sure if this is what actually causes my wake up issues but that's my best guess at the moment.

morrownr, 2024-08-07:
Hi @kandrelczyk 

The first thing I would look at is the suspend/sleep settings in the bios of your system. These settings vary wildly from computer to computer but basically, if your bios offers settings like S4, S3 or S1, try a different setting and see if that helps. Work your way to lower numbers... if the setting is S4, try S3 and if the setting is S4, try S1.

There are a lot of bioses with bugs in power saving. This is not a new thing but is less common these days.

kandrelczyk, 2024-08-07:
Hi @morrownr ,

Thanks for replying. I've check and my bios has only S3 and S0ix support. I will test some other setup but can you confirm that the logs I pasted are related to this? Am I right it's the WiFi adapter that fails to wake up correctly?

morrownr, 2024-08-07:
> can you confirm that the logs I pasted are related to this?

Probably.

> Am I right it's the WiFi adapter that fails to wake up correctly?

Does the wifi adapter work after you come out of power saving? If it does not, it very well could be the wifi adapter that is having a problem... but is could be something in the usb stack interacting with your hardware. This can be hard to diagnose.

Try the S0 setting, that should give us an idea.

I can probably work up a little script to work around this if it is needed. I could also put you to testing a newer version of a driver that is going upstream to help the driver in the kernel.

Would you run the following, what do you see:

sudo lsmod | grep 88

kandrelczyk, 2024-08-07:
Ok, if it can be anything on the USB stack I will remove the wifi adapter for now and test for some time. If I won't have any wake up issues anymore I will know at least it's the adapter. Doing more testing is complicated as it's my main workstation and the issue doesn't occur on every suspend so I would have to spend some time doing nothing but suspending it and waking it up. I will give it a try when I have some more time.

morrownr, 2024-08-07:
Just let me know when you need something.