morrownr/88x2bu-20210702 #232: Bad exit status: 2 after linux 6.1.0 kernel update

kanav662, 2024-09-20:
Dkms build error: 10

I just updated the kernel from 5.1.5 to 6.1.0

I forgot to remove the drive before and now the driver is not functioning.

I have tried uninstalling and installing again but i get this error. 

Please help

morrownr, 2024-09-20:
Hi @kanav662 

It is a really good idea to update the driver before updating the kernel.

Find a way to get internet connectivity.

Run the following in the driver directory:

$ git pull
$ sudo sh install-driver.sh

It is a good idea to keep a low cost adapter around that is plug and play:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

kanav662, 2024-09-20:
[image]
Still facing this issue.
I had been using this driver on my previous kernel and it was working flawlessly. Now its giving me this error

morrownr, 2024-09-20:
Try a manual installation:

$ make clean
$ make
$ sudo make install

Post the results of make if you see any problems.

danergo, 2025-03-11:
Hi @morrownr,

I'm also experiencing this problem.

Please note, I upgraded from 5.10.160 to 6.1.99 kernel. I have other network from eth0, so it's not the problem.

After kernel update, I removed the driver with `./remove-driver.sh`.

I tried to install it onto the new kernel, but it keeps failing. During manual compilation, here is the exact error:

```
/home/pi/work/wifi/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/pi/work/wifi/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:471:3: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
  471 |   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/work/wifi/88x2bu-20210702/include/osdep_service_linux.h:98,
                 from /home/pi/work/wifi/88x2bu-20210702/include/osdep_service.h:54,
                 from /home/pi/work/wifi/88x2bu-20210702/include/drv_types.h:27,
                 from /home/pi/work/wifi/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8522:6: note: declared here
 8522 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/work/wifi/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:490:2: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  490 |  cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/work/wifi/88x2bu-20210702/include/osdep_service_linux.h:98,
                 from /home/pi/work/wifi/88x2bu-20210702/include/osdep_service.h:54,
                 from /home/pi/work/wifi/88x2bu-20210702/include/drv_types.h:27,
                 from /home/pi/work/wifi/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8505:6: note: declared here
 8505 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/work/wifi/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c: At top level:
cc1: note: unrecognized command-line option ‘-Wno-stringop-overread’ may have been intended to silence earlier diagnostics
make[2]: *** [scripts/Makefile.build:250: /home/pi/work/wifi/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2053: /home/pi/work/wifi/88x2bu-20210702] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.99'
make: *** [Makefile:2533: modules] Error 2
```

morrownr, 2025-03-11:
Hi @danergo 

I can see why your installation is failing. Let's talk about how Linux differs from Windows:

Linux has adopted a continuous update development philosophy. Windows will lock large portions of its kernel for long periods. I think mostly has to do with Linux' need to take advantage of the resources that are available, when they are available. As a result, every two months or so, a kernel kernel is released and it may have new capability and it may have changed or updated API's that require changes in drivers. One of the benefits is that Linux often has new features before Windows. Often it is long before Windows. On the other hand, you need to understand that you need to buy products that are supported the way they should be supported. For USB WiFi adapters, Mediatek does a really good job, Realtek, not so much.

To mediate problems with the continuous development methods used in Linux, the normal method of driver development is for companies to have devs working with the kernel team. For example: linux-wireless is the mailing list where all kernel wireless devs communicate. You have employees from Intel, Qualcomm, Mediatek, Realtek and a few more that work together and continuously keep their drivers up to date. Drivers released with 6.14 will have those drivers updated for kernel 6.14 so users usually never know that their driver has been updated. It is interesting to watch all of the wireless devs work together and I do follow linux-wireless.

However, Realtek's USB wireless team does not participate with linux-wireless and they only release out of kernel drivers. This is not good and causes folks like me to have to do their job for them. This is why I adamentaly recommend that Linux users buy adapters that use in-kernel drivers. More information available at:

https://github.com/morrownr/USB-WiFi

Scroll down and read menu items 1 and 2 as they are very informative. I try to help USB WiFi users with information so that they buy adapters that are as reliable as possible. I also put warnings up when I see products that Linux users should avoid:

https://github.com/morrownr/USB-WiFi/issues/449

Back to the subject:

Go you your driver directory and run the following:

$ git pull
$ sudo sh install-driver.sh

After reboot, let me know if things are working.

danergo, 2025-03-12:
Thank you. Yes, I know all about this. Just I don't want to throw out my USB Wifi due to kernel update :)

Back to the subject:

```
$ git pull
Already up to date.

$ sudo sh install-driver.sh
: ---------------------------
: install-driver.sh v20241003
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 3998584 (total system memory)
: 6.1.99 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Debian 10.2.1-6) 10.2.1 20210110
: dkms:2.8
: mokutil not installed (Secure Boot status unknown)
: ---------------------------

Checking for previously installed drivers...
Removing a driver that was added to dkms.

------------------------------
Deleting module version: 5.13.1
completely from the DKMS tree.
------------------------------
Done.
Removing 88x2bu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl88x2bu-5.13.1
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1

Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source ->
                 /usr/src/rtl88x2bu-5.13.1

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
kernelver=6.1.99 ./dkms-make.sh....................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.99 (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

$ cat /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log 
DKMS make.log for rtl88x2bu-5.13.1 for kernel 6.1.99 (aarch64)
Wed Mar 12 05:15:56 CET 2025
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.99/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.99'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gcc (ctng-1.25.0-119g-FA) 11.3.0
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o
[... 71 lines cut ...]
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8522:6: note: declared here
 8522 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:490:2: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  490 |  cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8505:6: note: declared here
 8505 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: At top level:
cc1: note: unrecognized command-line option ‘-Wno-stringop-overread’ may have been intended to silence earlier diagnostics
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2053: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.99'
make: *** [Makefile:2533: modules] Error 2
```

It's the same error as I posted above.

danergo, 2025-03-12:
Switched to https://github.com/lwfinger/rtw88. Works as intended.

morrownr, 2025-03-13:
> Switched to https://github.com/lwfinger/rtw88. Works as intended.

That repo has been used over the last year to improve the in-kernel rtw88 and add additional driver such as the rtl8812au, rtl8821/11au and rtl8814au. With your rtl8812bu adapter, once you get to kernel 6.12, you won't need to install a driver. You adapter should be plug and play.