morrownr/88x2bu-20210702 #233: array access out of boundary

spcharc, 2024-09-29:
In function `phydm_dfs_histogram_radar_distinguish`

```
	for (i = 0; i < 6; i++) {
		dfs->pw_hold[dfs->hist_idx][i] = (u8)g_pw[i];
		dfs->pri_hold[dfs->hist_idx][i] = (u8)g_pri[i];
		/*@collect whole histogram report may take some time
		 *so we add the counter of 2 time slots in FCC and ETSI
		 */
		dfs->pw_hold_sum[i] = dfs->pw_hold_sum[i] +
			dfs->pw_hold[(dfs->hist_idx + 1) % 3][i] +
			dfs->pw_hold[(dfs->hist_idx + 2) % 3][i];
		dfs->pri_hold_sum[i] = dfs->pri_hold_sum[i] +
			dfs->pri_hold[(dfs->hist_idx + 1) % 3][i] +
			dfs->pri_hold[(dfs->hist_idx + 2) % 3][i];
	}
	/*@For long radar type*/
	for (j = 1; j < 4; j++) {
		dfs->pw_long_hold_sum[i] = dfs->pw_long_hold_sum[i] +
			dfs->pw_hold[(dfs->hist_long_idx + j) % 4][i];
		dfs->pri_long_hold_sum[i] = dfs->pri_long_hold_sum[i] +
			dfs->pri_hold[(dfs->hist_long_idx + j) % 4][i];
	}

```

After the 1st loop ends, the 2nd loop starts with `i=6`, which causes out of boundary access of `pw_long_hold_sum` and `pw_hold`

morrownr, 2024-09-29:
Wonderful! Sarcasm...

Have you considered or tested any fixes?

I do have a later very of the source so I could take a look at that but I am leaving on a trip right now so it could be a few days.

spcharc, 2024-09-30:
I just removed the 2nd for loop (after `/*@For long radar type*/`). Works without issue. It is definitely not the correct way to fix, so I created a ticket not a pull request.