morrownr/88x2bu-20210702 #234: Kali LINUX HELP...DRIVER :(

freitas666, 2024-10-16:
Hello friends...

Please someone can help me?

Actually i did all to try to install this mot*** drive to work on my kali linux...but i did not have success...

When install not find any network available...when not install not work lol....

Some comands:

6.10.11-amd64
ii linux-headers-6.10.11-amd64 6.10.11-1kali1 amd64 Header files for Linux 6.10.11-amd64
ii linux-headers-6.10.11-common 6.10.11-1kali1 all Common header files for Linux 6.10.11
ii linux-headers-6.10.9-amd64 6.10.9-1kali1 amd64 Header files for Linux 6.10.9-amd64
ii linux-headers-6.10.9-common 6.10.9-1kali1 all Common header files for Linux 6.10.9

wconfig wlan0
wlan0 unassociated Nickname:"[WIFI@REALTEK](mailto:WIFI@REALTEK)"
Mode:Managed Frequency=2.412 GHz Access Point: Not-Associated
Sensitivity:0/0
Retry:off RTS thr:off Fragment thr:off
Encryption key:off
Power Management:off
Link Quality:0 Signal level:0 Noise level:0
Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
Tx excessive retries:0 Invalid misc:0 Missed beacon:0

airmon-ng start wlan0
PHY Interface Driver Chipset
phy1 wlan0 rtl88x2bu Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
(monitor mode enabled)

dkms status
rtl88x2bu/5.8.7.1, 6.10.11-amd64, x86_64: installed

I am using Vmware workstation...

Thanks... :)

morrownr, 2024-10-16:
Hi @freitas666 

> Please someone can help me?

It is likely we can help but it is not as simple as do this one step. Let me explain.

Based on what you posted, your problem is not a driver problem. It is a problem with the VM.

I am not familiar with Vmware. I do have a guide for using an adapter in VirtualBox:

https://github.com/morrownr/USB-WiFi/blob/main/home/Using_USB_Adapter_in_VirtualBox.md

If you are inclined to give Virtualbox a try, then the guide can probably get you going. 

Next subject:

You may not be pleased with an adapter that uses a Realtek chip. I have a list that shows usb wifi adapters that are really good at what you are trying to do:

https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_Adapters_for_Kali_Linux.md

FYI: The Main Menu for this site is located at:

https://github.com/morrownr/USB-WiFi

Let us know if you have additional questions.

freitas666, 2024-10-16:
Hey thanks for reply me... Yes i understand your completely, I will try to download Virtual Box to test... but strange part is because somedays ago worked in vmware!
In anyway, i will back to give some feedback

closed 2024-10-20 by freitas666

morrownr, 2024-10-21:
@freitas666 

Thanks for the report. Not as uncommon as we would like.