morrownr/88x2bu-20210702 #235: connected but no internet on Ubuntu 22.04.4 LTS

bookreader, 2024-10-29:
Hello,

I just bought a usb wifi receiver, CF-812AC and installed your driver without any installation errors. But after reboot, the connection works but no internet.

Here is my system info:
/etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.4 LTS"

and  from uname -a:
6.8.0-47-generic #47~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Oct  2 16:16:55 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

Any hint of help will be greatly appreciated.

Thanks very much

Ling

dubhater, 2024-11-08:
If you are using it with rtw_switch_usb_mode=1 try 0 instead.

bookreader, 2024-11-09:
> If you are using it with rtw_switch_usb_mode=1 try 0 instead.

I checked the /etc/modprob.d/88x2bu.conf
it has been that rtw_switch_usb_mode=0.

I tried setting it to 1 and then reboot and then setting it back to 0 and then reboot.
I was able to use the Comfast CF-812AC usb but the speed is terribly slow. I returned the usb.

Thanks so much