morrownr/88x2bu-20210702 #236: Compilation error on AlmaLinux 9.5

chofchof, 2024-12-02:
I had a compilation error on **AlmaLinux 9.5** with the kernel 5.14.0-503.14.1.el9_5.x86_64.
I could solve this problem with the following patch. Hope to see @misha4gps's reply soon.

```
diff --git a/Makefile b/Makefile
index 396817a..ee7478b 100644
--- a/Makefile
+++ b/Makefile
@@ -78,6 +78,9 @@ endif
 ifeq ($(shell test $(RHEL_SVER) -ge 427; echo $$?),0)
 EXTRA_CFLAGS += -DRHEL94
 endif
+ifeq ($(shell test $(RHEL_SVER) -ge 503; echo $$?),0)
+EXTRA_CFLAGS += -DRHEL95
+endif
 endif
 endif
 
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index e5be490..2328011 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -465,7 +465,7 @@ u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset,
 		 */
 
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0) || defined(RHEL88))
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 9, 0))) || defined(RHEL89)
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 9, 0))) || (defined(RHEL89) && !defined(RHEL95))
 		cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false, 0);
 #else
 		cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false);
@@ -484,7 +484,7 @@ u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset,
 		goto exit;
 
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 2) || defined(RHEL88))
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 9, 0))) || defined(RHEL89)
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 9, 0))) || (defined(RHEL89) && !defined(RHEL95))
 	cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0, 0);
 #else
 	cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
diff --git a/os_dep/linux/usb_intf.c b/os_dep/linux/usb_intf.c
index f3946f3..abe178b 100644
--- a/os_dep/linux/usb_intf.c
+++ b/os_dep/linux/usb_intf.c
@@ -361,7 +361,7 @@ struct rtw_usb_drv usb_drv = {
 	.usbdrv.reset_resume   = rtw_resume,
 #endif
 
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 19)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 8, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 19)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 8, 0)) && !defined(RHEL95)
 	.usbdrv.drvwrap.driver.shutdown = rtw_dev_shutdown,
 #else
 	.usbdrv.driver.shutdown = rtw_dev_shutdown,
```

misha4gps, 2024-12-05:
Hi @chofchof ,
I have just created PR#237
Thanks a lot for the patch.

morrownr, 2024-12-05:
Patch merged.

chofchof, 2024-12-06:
@misha4gps Good to see you again.
@morrownr Thanks for maintaining such a good site.

morrownr, 2024-12-06:
@chofchof @misha4gps 

I was just pondering something. The in-kernel rtw88 series of drivers has really been improving and is or soon will be far better than these out-of-kernel drivers. In fact, we just upstreamed new drivers for the rtl8812au and rtl8821/11au. They went into 6.13.

I know that is not immediate help for RHEL kernel users but maybe you guys might consider modifying the repo that we use for development:

https://github.com/lwfinger/rtw88

If it was modified, then you would have standards compliant drivers. The only WiFi 5 Realtek driver remaining is the one for the rtl8814au and it is in development.

misha4gps, 2024-12-06:
@morrownr,
I test the [rtw88](https://github.com/lwfinger/rtw88) from time to time and am thinking of porting the driver to rhel9+ in Q1 next year.