morrownr/88x2bu-20210702 #238: Error loadijng on Debian

gerroon, 2024-12-13:
Hi

I am trying to install this driver on a remote machine. It seems like the driver isn't loaded after the installation using the install script so I try manual loading

```
modprobe 88x2bu
modprobe: ERROR: could not insert '88x2bu': Key was rejected by service
```

```
cat /etc/debian_version
12.8

 uname -ra
Linux debian 6.1.0-28-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22) x86_64 GNU/Linux

 lsusb

Bus 003 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

```

I didn't get any errors during the installation.

morrownr, 2024-12-13:
Hi @gerroon 

> I didn't get any errors during the installation.

There probably were not any errors.

> modprobe: ERROR: could not insert '88x2bu': Key was rejected by service

So the system is running secure boot. There is not as much standardization among Linux distros as far as secure boot is concerned. At least not yet. This drivers installation as it currently is will handle secure boot on Ubuntu and the derivatives very well but beyond that it may take an investigation.

If you can look around the Debian web site for information about secure boot and how to do it, and send it to me, I'll see what we can do.

gerroon, 2024-12-13:
Thanks for the reply. I just checked dmesg and I see. It is a remote PC so not sure how much I can deal with the secure boot thing, but thanks for the info.

```
[Cum Ara 13 21:50:07 2024] brcmsmac bcma0:1: wl0: brcms_c_d11hdrs_mac80211:  txop exceeded phylen 159/256 dur 1778/1504
[Cum Ara 13 21:50:07 2024] brcmsmac bcma0:1: wl0: brcms_c_d11hdrs_mac80211:  txop exceeded phylen 137/256 dur 1602/1504
[Cum Ara 13 21:50:07 2024] brcmsmac bcma0:1: wl0: brcms_c_d11hdrs_mac80211:  txop exceeded phylen 372/256 dur 4110/1504
[Cum Ara 13 21:54:45 2024] usb 3-2: USB disconnect, device number 3
[Cum Ara 13 21:54:51 2024] usb 3-2: new high-speed USB device number 4 using xhci_hcd
[Cum Ara 13 21:54:51 2024] usb 3-2: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[Cum Ara 13 21:54:51 2024] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Cum Ara 13 21:54:51 2024] usb 3-2: Product: USB3.0 802.11ac 1200M Adapter
[Cum Ara 13 21:54:51 2024] usb 3-2: Manufacturer: Realtek
[Cum Ara 13 21:54:51 2024] usb 3-2: SerialNumber: 123456
[Cum Ara 13 21:54:51 2024] Loading of module with unavailable key is rejected

```

morrownr, 2024-12-13:
> It is a remote PC so not sure how much I can deal with the secure boot thing, but thanks for the info.

It is possible to key the driver so that it works. It just depends on the distro that is being used. FYI: By the time Debian 13 comes out, this won't be an issue as the adapter will be plug and play.

morrownr, 2024-12-13:
Another thought: Here is the Main Menu for this site:

https://github.com/morrownr/USB-WiFi

If you read menu items 1 and 2, you will see that there are many plug and play adapters available for Linux. You can go as high as WiFi 6e adapter with the kernel and Debian 12 that you are using. No more hunting for and compiling drivers.

gerroon, 2024-12-13:
Thanks for the useful information. Most of these cards aren't available in the host country's Amazon stores. I will keep looking at them, to see if I can find the same chipsets in other brands. I am guessing the chipset is what matters.

Do you think that upgrading to "testing" would resolve the issue for me? If not I am guessing that I can't make a signed compiled driver myself.

morrownr, 2024-12-14:
@gerroon 

It is. for the most part, the chipset that matters. You can see the Plug and Play List and there is Chipset menu item on the Main Menu as well.

Here is message showing another adapter that I am considering adding to the Plug and Play List. Maybe it is available to you. Be careful about model numbers as TP-Link has many model numbers that are nearly the same.

mqu, 2025-12-03:
seems to work out of the box with ubuntu 24.04 : 
```
$ dmesg -T
...
[Wed Dec  3 11:53:54 2025] perf: interrupt took too long (7128 > 6165), lowering kernel.perf_event_max_sample_rate to 28000
[Wed Dec  3 17:45:27 2025] usb 3-1.2: new high-speed USB device number 4 using ehci-pci
[Wed Dec  3 17:45:27 2025] usb 3-1.2: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[Wed Dec  3 17:45:27 2025] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Dec  3 17:45:27 2025] usb 3-1.2: Product: 802.11ac NIC
[Wed Dec  3 17:45:27 2025] usb 3-1.2: Manufacturer: Realtek
[Wed Dec  3 17:45:27 2025] usb 3-1.2: SerialNumber: 123456
[Wed Dec  3 17:45:27 2025] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[Wed Dec  3 17:45:27 2025] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[Wed Dec  3 17:45:27 2025] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[Wed Dec  3 17:45:27 2025] rtw_8822bu 3-1.2:1.0: Firmware version 27.2.0, H2C version 13
[Wed Dec  3 17:45:28 2025] usbcore: registered new interface driver rtw_8822bu
[Wed Dec  3 17:45:28 2025] rtw_8822bu 3-1.2:1.0 wlx40a5ef574f03: renamed from wlan0
...
$ lsmod  | grep rtw88
rtw88_8822bu           12288  0
rtw88_usb              24576  1 rtw88_8822bu
rtw88_8822b           229376  1 rtw88_8822bu
rtw88_core            356352  2 rtw88_usb,rtw88_8822b
mac80211             1720320  2 rtw88_core,rtw88_usb
cfg80211             1323008  2 rtw88_core,mac80211
...
# iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
...
```