morrownr/88x2bu-20210702 #239: Driver installed but no wlan interface on Debian 12

niponic, 2024-12-20:
Hello, first of all thank you for your work.
I installed the driver perfectly, without errors. When connecting the adapter I also get the message that it was detected correctly, but the WLAN interface was not created. What am I doing wrong?

[image]

morrownr, 2024-12-20:
Hi @niponic 

>  mine is 0BDA:C812. Could this be the problem?

Yes, it appears that your adapter has then rtl8812cu chip. Let me recommend that you totally uninstall this driver:

$ sudo sh remove-driver.sh

And then go over to this repo:

https://github.com/lwfinger/rtw88

Follow the README as the instructions are somewhat different than here.

I also work at that repo. That repo is for work that goes into the kernel. You won't need to install a driver at some point... probably once you get to kernel 6.12 or later.

niponic, 2024-12-20:
It worked, thanks

closed 2024-12-20 by niponic