morrownr/88x2bu-20210702 #243: Linux

DaemonMCR, 2025-01-28:
Why have you not submitted this code to be merged into mainline upstream?

morrownr, 2025-01-28:
Because it would not be accepted as it is not standards compliant. Realtek's USB support is poor. You can learn more at my Main Menu:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 could be useful.

A standards compliant driver rtl88x2bu is in modern kernels. If you are using kernel 6.11 or later, it should work for you.

DaemonMCR, 2025-01-28:
Yeah I mean the Linux driver sort of works, it’s just every once in a while the driver dies and requires either a restart or to replug in the device, also without ur driver thing it does not flash the blue light in the hardware. What confuses me is how this wouldn’t get accepted you know, since the current code at least in my experience is buggy in upstream

morrownr, 2025-01-30:
> Yeah I mean the Linux driver sort of works...

Give the in-kernel rtw88 driver another go when you get to kernel 6.14 or later. A lot of work has taken place over the last year and the driver is in much better shape starting with kernel 6.14.