morrownr/88x2bu-20210702 #244: P2P SSID visible in wifi but not in wifi direct list

ptmrankzy, 2025-02-01:
I have a raspberry pi zero w with brostrend ac1200(showing up as 0dba:b812).
When trying to set it in P2P_GO mode, it broadcasts a random SSID: DIRECT-XX, but when i go to search for it in wifi direct settings on my phone and laptop, it isn't listed.

Raspbian OS 6.6.62

```
Bus 001 Device 005: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 001 Device 004: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 003: ID 0461:4e8e Primax Electronics, Ltd HP USB Keyboard
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: phy1: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl88x2bu/5.13.1, 6.6.62+rpt-rpi-v6, armv6l: installed
phy#1
        Interface wlan1
                ifindex 4
                wdev 0x100000001
                addr 98:03:cf:03:56:e3
                ssid DIRECT-ZQ
                type P2P-GO
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 14.00 dBm
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr b8:27:eb:78:c4:ef
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
global
country BE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

```

morrownr, 2025-02-01:
Hi @ptmrankzy 

I just merged a patch that may help:

activate RTW_P2P_GROUP_INTERFACE

Run the following in the driver directory:

$ git pull

$ sudo sh install-driver.sh

Let me know.

ptmrankzy, 2025-02-02:
Thanks for the fast response!

I have updated the driver and when i try to start it with wpa_supplicant i get 
`nl80211: kernel reports: Registration to specific type not supported`

morrownr, 2025-02-02:
@ptmrankzy 

I do not have good visibility on what you are doing so let me point you to:

docs/Concurrent_Mode.md

Keep in mind that these Realtek out-of-kernel drivers do not do many things the way it is supposed to be done.

ptmrankzy, 2025-02-02:
I tried concurrent mode and when now it won't even get into p2p mode.

So to replicate my orignal issue,

wpa_supplicant.conf:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
device_name=RaspberryPi
device_type=1-0050F204-1
p2p_go_intent=15
```

commands:
```
sudo pkill wpa_supplicant
sudo wpa_supplicant -B -i wlan1 -c /etc/wpa_supplicant/wpa_supplicant.conf
sudo wpa_cli -i wlan1 p2p_group_add
sudo wpa_cli -i wlan1 wps_pin any 12345678
sudo ifconfig wlan1 192.168.66.1
```

On the raspberry pi, everything looks good(before that patch).
Only problem is that devices don't recognise it as p2p but as a normal wifi.

morrownr, 2025-02-05:
Should I remove the patch so that we can start over?

ptmrankzy, 2025-02-10:
Yeah that's probably a good idea. If have tried AP mode and that works on the commit before that patch.

morrownr, 2025-02-10:
I reverted the patch. You can update with:

$ git pull

Sorry I am not much help as what you are doing is something that I do not have experience with.

ptmrankzy, 2025-02-20:
I found out that in rtw_p2p there is a lot of todo comments, so it isn't fully implemented.

morrownr, 2025-02-20:
@ptmrankzy 

Let me make a suggestion. Use the following driver instead:

https://github.com/lwfinger/rtw88

That is the repo that we are using to make new drivers for the kernel and fix existing drivers with Realtek chips. It supports your adapter. If you find problems with P2P, then you can report them right in that repo and the devs will see it, and the fix will go into the Linux mainline kernel.

ptmrankzy, 2025-02-22:
Thanks! I had tried compiling linux 6.14 because i heard that it got a big update for those drivers, unfortunatly it didn't even show p2p as supported then. But when it tried compiling the driver repo ant that one works!

closed 2025-02-22 by ptmrankzy