morrownr/88x2bu-20210702 #246: Kernel 6.14.0-rc1 Issues with compiling

Patronos, 2025-02-10:
I had to remove the .get_tx_power line to make it compile. Maybe they changed some last minute structures. Please if you could have a review. Thanks.

`  CC [M]  core/crypto/sha256-internal.o
  CC [M]  core/crypto/sha256.o
  CC [M]  core/crypto/sha256-prf.o
  CC [M]  core/crypto/rtw_crypto_wrap.o
  CC [M]  core/rtw_swcrypto.o
  CC [M]  os_dep/osdep_service.o
  CC [M]  os_dep/linux/os_intfs.o
  CC [M]  os_dep/linux/usb_intf.o
  CC [M]  os_dep/linux/usb_ops_linux.o
  CC [M]  os_dep/linux/ioctl_linux.o
os_dep/linux/ioctl_linux.c:12276:63: warning: initializer-string for array of ‘char’ is too long [-Wunterminated-string-initialization]
12276 |                 IW_PRIV_TYPE_INT | IW_PRIV_SIZE_FIXED | 1, 0, "wps_assoc_req_ie"
      |                                                               ^~~~~~~~~~~~~~~~~~
  CC [M]  os_dep/linux/xmit_linux.o
  CC [M]  os_dep/linux/mlme_linux.o
  CC [M]  os_dep/linux/recv_linux.o
  CC [M]  os_dep/linux/ioctl_cfg80211.o
os_dep/linux/ioctl_cfg80211.c:10468:25: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  int *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, int *)’ [-Wincompatible-pointer-types]
10468 |         .get_tx_power = cfg80211_rtw_get_txpower,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:10468:25: note: (near initialization for ‘rtw_cfg80211_ops.get_tx_power’)
os_dep/linux/ioctl_cfg80211.c:4647:12: note: ‘cfg80211_rtw_get_txpower’ declared here
 4647 | static int cfg80211_rtw_get_txpower(struct wiphy *wiphy,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/kernels/6.14.0-0.rc1.15.fc42.x86_64/scripts/Makefile.build:207: os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** [/usr/src/kernels/6.14.0-0.rc1.15.fc42.x86_64/Makefile:2040: .] Error 2
make[2]: *** [/usr/src/kernels/6.14.0-0.rc1.15.fc42.x86_64/Makefile:263: __sub-make] Error 2
make[2]: Leaving directory '/home/aakcaagac/Downloads/88x2bu-20210702'
make[1]: *** [Makefile:263: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.14.0-0.rc1.15.fc42.x86_64'
make: *** [Makefile:2533: modules] Error 2
strip 88x2bu.ko --strip-unneeded
strip: '88x2bu.ko': No such file
make: *** [Makefile:2536: strip] Error 1
install -p -m 644 88x2bu.ko  /lib/modules/6.14.0-0.rc1.15.fc42.x86_64/kernel/drivers/net/wireless/
install: cannot stat '88x2bu.ko': No such file or directory
make: *** [Makefile:2539: install] Error 1
/home/aakcaagac/Downloads
At main.c:302:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: /usr/lib/modules/6.14.0-0.rc1.15.fc42.x86_64/kernel/drivers/net/wireless/88x2bu.ko
/usr/lib/modules/6.14.0-0.rc1.15.fc42.x86_64/kernel/drivers/net/wireless/88x2bu.ko (1/1)
xz: /usr/lib/modules/6.14.0-0.rc1.15.fc42.x86_64/kernel/drivers/net/wireless/88x2bu.ko: No such file or directory
`

morrownr, 2025-02-11:
Hi @Patronos 

I just checked and I see what you mean.

My plan was to discontinue API maintenance on the majority of WiFi 5 drivers here as kernel 6.14 has a better driver in it than this one. It is one of the rtw88 series of drivers... rtw88_8822bu. We have been working hard to get rtw88 in-kernel up to speed over the last year. Have you tried the in-kernel driver?

You will need to uninstall this driver to uninstall the blacklist for rtw88_8822bu:

$ sudo sh remove-driver.sh

Patronos, 2025-02-12:
Hi there and thanks for the resonse.

Yes I am aware of that Kernel driver and had very bad experiences with it. The reason why I stuck with the driver found here and somehow rely on a working driver.

The reasons are quite easy to explain:
- The Kernel intern driver permanently drops connection of the internet (That I already detected when I first tried the Kernel internal driver around 1-2 years ago (something like that)).
- The other issues is that it interferes with my wireless USB BT Mouse (Logitech). The Mouse keeps locking during movement over the screen.

Somehow I have the feeling that both devices are somehow "fighting" each other. As soon as the Network drops, the Mouse operates normally and as soon as the Mouse is left where it is, the Connection operates perfectly.

After your response I just re-enabled the internal 6.14 driver and ensured that the firnware blob is being loaded as well. It comes up correctly but still suffers from the same issues that I experienced 1-2 years ago.

This does NOT happen with the github driver found here. No issues with mouse, no issues with internet drops, no connection issues, works perfectly.

I think the Kernel driver still has a longer way to go to work "normally" out of the box.

morrownr, 2025-02-14:
@Patronos 

I understand what you are saying. We need to find a peaceful resolution. I really do not want to do the kernel 6.14 work on any of these WiFi 5 drivers. Where does it stop? I have maintained these drivers for several years and am doing my best to help wireless on Linux but I must shed some of the work as it is too much.

Over the last year, I worked on the small team that has greatly improved the in-kernel rtw88. Many users and I have tested and tested and tested. The rtw88 driver for this chip is much better than this driver. I understand that you see a problem. I believe that you see the problem. What I want is to see if we can fix it, if it is a driver issue.

To help us fix it, go to the following repo:

https://github.com/lwfinger/rtw88

Uninstall this driver by running:

$ sudo sh remove-driver.sh

Now install rtw88 from the repo above. It is the repo where we work on the drivers that are upstreamed to the kernel. If there is a problem in the rtw88 drivers, we need it reported at that repo. Please help.

Thanks

morrownr, 2025-02-22:
@Patronos 

I went ahead and merged the changes for kernel 6.14. Give it a test and let me know if it works.

Patronos, 2025-03-09:
> [@Patronos](https://github.com/Patronos)
> 
> I went ahead and merged the changes for kernel 6.14. Give it a test and let me know if it works.

Hi there. Sorry for not responding earlier. The patch you created compiles perfectly with the current kernel version and worked up to RC4. With RC5 it still compiles but the module can't be loaded anymore.

Patronos, 2025-03-09:
Update:

For 6.14.0-RC5 they enabled module compression inside the kernel config files. Even if you happen to have .xz files before or manually xz compressed this module, you may end up that you receive the following dmesg message.

decompression failed with status 6

After searching inside the Fedora spec file for the kernel I detected that they are using a different lzma dict than I was using before. I usually wasy simply running "xz -f9v 88x2bu.ko" before.

So in case anyone if someone walks over this, please compress this module with:

xz --compress --force --check="crc32" -9 --lzma2="dict=1MiB" --verbose 88x2bu.ko

Signature of the module works as usual. No issues detected. secure boot works.

morrownr, 2025-03-10:
@Patronos 

You have got to be kidding me. What in the hell is fedora thinking? 

We already have 3 main compression methods in use with various distros and now this. What we need is an increased level of cooperation and standardization instead of crap like this. This just pisses off people like me that are trying to help users.

Patronos, 2025-07-25:
Please note, that I haven't added or messed with kernel specific ifdefs here. Module compiles for me but I have to test it.

Patronos, 2025-08-03:
```
diff -ruN 88x2bu-20210702.orig/os_dep/linux/ioctl_cfg80211.c 88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c
--- 88x2bu-20210702.orig/os_dep/linux/ioctl_cfg80211.c	2025-08-03 00:00:00.000000000 +0200
+++ 88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c	2025-08-03 00:00:00.000000000 +0200
@@ -3601,7 +3601,7 @@
 }
 #endif
 
-static int cfg80211_rtw_set_wiphy_params(struct wiphy *wiphy, u32 changed)
+static int cfg80211_rtw_set_wiphy_params(struct wiphy *wiphy, int radio_idx, u32 changed)
 {
 #if 0
 	struct iwm_priv *iwm = wiphy_to_iwm(wiphy);
@@ -4584,7 +4584,7 @@
 
 static int cfg80211_rtw_set_txpower(struct wiphy *wiphy,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
-	struct wireless_dev *wdev,
+	struct wireless_dev *wdev, int radio_idx,
 #endif
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 36)) || defined(COMPAT_KERNEL_RELEASE)
 	enum nl80211_tx_power_setting type, int mbm)
@@ -4646,7 +4646,7 @@
 
 static int cfg80211_rtw_get_txpower(struct wiphy *wiphy,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
-	struct wireless_dev *wdev,
+	struct wireless_dev *wdev, int radio_idx,
 #endif
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 14, 0))
 	unsigned int link_id,

```

Patronos, 2025-12-05:
Fix for kernel-6.18.0-65.fc44.x86_64

morrownr, 2025-12-05:
@Patronos 

Thanks for the 6.18 patch. I have new 2025 source for this driver and it has a lot of updates. It may be better to create a new repo for the new driver and add your 6.18 patch than try to work this older driver. Would you be interested in helping?

FYI: Could you test this driver to see if it is working with USB3? I have had reports that it is not working with USB3 but this new code does work with USB3.