morrownr/88x2bu-20210702 #247: Driver disconnects when channel changes

ubarbosa, 2025-03-17:
I have two realtek adapters (8811au and 8812bu).
The 5 GHz connection starts at channel 149. At some interval, the router changes to channel 56. These adapters don't do well with this, and disconnect. The exact messages are (log level 4).

Mar 17 05:53:03 ufer2004 kernel: RTW: rtw_ies_get_chbw ch inconsistent, DSSS:56, HT primary:149
Mar 17 05:53:03 ufer2004 kernel: RTW: rtw_check_bcn_info(wlxc006c3914137) new beacon occur!!
Mar 17 05:53:03 ufer2004 kernel: RTW: rtw_check_bcn_info(wlxc006c3914137) cur beacon key:
Mar 17 05:53:03 ufer2004 kernel: RTW: ssid = Bibi AM FM 5G (len = 13)
Mar 17 05:53:03 ufer2004 kernel: RTW: ch = 149,2,1
Mar 17 05:53:03 ufer2004 kernel: RTW: proto_cap = 0x0e
Mar 17 05:53:03 ufer2004 kernel: RTW: rate_set = 8C 12 98 24  B0 48 60 6C  
Mar 17 05:53:03 ufer2004 kernel: RTW: sec = 3, group = 0x8, pair = 0x18, akm = 0x00000002
Mar 17 05:53:03 ufer2004 kernel: RTW: rtw_check_bcn_info(wlxc006c3914137) new beacon key:
Mar 17 05:53:03 ufer2004 kernel: RTW: ssid = Bibi AM FM 5G (len = 13)
Mar 17 05:53:03 ufer2004 kernel: RTW: ch = 56,2,1
Mar 17 05:53:03 ufer2004 kernel: RTW: proto_cap = 0x0e
Mar 17 05:53:03 ufer2004 kernel: RTW: rate_set = 8C 12 98 24  B0 48 60 6C  
Mar 17 05:53:03 ufer2004 kernel: RTW: sec = 3, group = 0x8, pair = 0x18, akm = 0x00000002
Mar 17 05:53:03 ufer2004 kernel: RTW: OnBeacon(wlxc006c3914137) ap has changed, disconnect now
Mar 17 05:53:03 ufer2004 kernel: RTW: receive_disconnect

And I got a disconnect/reconnect cycle, again, on channel 149, until next change.
An old adapter, RTL8723bs (non USB) stays on.

morrownr, 2025-03-17:
Hi @ubarbosa 

> At some interval, the router changes to channel 56. These adapters don't do well with this, and disconnect.

These Realtek out-of-kernel drivers are not the best. Let me offer you some options:

1. A new in-kernel driver for the rtl8811au just went into kernel 6.14. It is part of the rtw88 series of drivers. rtw88 has had a driver for the rtl8812bu for some time but I recommend kernel 6.12 or later.

If you are not on at least kernel 6.12 to test the rtl8812bu, you can go to the following repo and install the rtw88 code into older kernels after unistallation this driver:

$ sudo sh remove-driver.sh

Go to:

https://github.com/lwfinger/rtw88

Follow the installation instructions.

2. We can dig into the details to see what might be causing the problem. I prefer not to go this direction as I am busy and the new driver options are better.

Let me know.

ubarbosa, 2025-03-18:
Actually, I just bypassed the receive_disconnect()  in rtw_mlme_ext.c if only the channel is different. The connection doesn't recycle, staying on the old channel. It is not ideal, I know, but fixed the issue.

ubarbosa, 2025-03-24:
Seems the router fakes the change. Originally on channel 149, it is sending beacons on channells 56, 60, 112, 136 and 165 at each 12 minutes. But it still works on 149. It is a burst of beacons lasting 1-2 seconds. Preventing the driver to change and remain on channel 149, avoided the disconnect/reconnect.

morrownr, 2025-03-26:
What might work better for you is to fully uninstall this driver:

$ sudo sh remove-driver.sh

...and then install the following driver:

https://github.com/lwfinger/rtw88

The above driver is used to work on code that is upstreaming into mainline. 

I'm am trying to let this driver die a slow death as we are approaching the time when it won't be needed.