morrownr/88x2bu-20210702 #248: install issues with kernel headers on debian bookworm

NicolasDu49, 2025-03-22:
Hi,

I'm trying to install the realtek driver on my Debian bookworm machine (this is a server version), but I get stuck during the install-driver.sh execution.
I get an error about missing kernel headers while I checked that it was well installed. Here is the log message I get:
```
: install-driver.sh v20241003
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 12/12 (in-use/total processing units)
: 16270108 (total system memory)
: 6.1.0-32-amd64 (kernel version)
: gcc-12 (version of gcc used to compile the kernel)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers...
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.
: ---------------------------

Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Error! Your kernel headers for kernel 6.1.0-32-amd64 cannot be found at /lib/modules/6.1.0-32-amd64/build or /lib/modules/6.1.0-32-amd64/source.
Please install the linux-headers-6.1.0-32-amd64 package or use the --kernelsourcedir option to tell DKMS where it's located.
An error occurred. dkms build error:  1
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

I can see that it suggest using the --kernelsourcedir option but as I don't know where my linux-headers are actually located, I didn't try it. I also noticed that some files were installed in the /usr/src/linux-headers-6.1.0-32-amd64/ directory, I don't know if this is something that can help.

Thanks a lot for any help you could provide!

morrownr, 2025-03-23:
How did you install the header files?

NicolasDu49, 2025-03-27:
I tried to run the command you put in the tutorial, which is `sudo apt update && sudo apt install -y linux-headers-$(uname -r)` (along with the other commands installation).
I also tried to run the command with the exact version instead of $(uname -r), but same issue.