morrownr/88x2bu-20210702 #253: Failed to create interface in p2p_group_add

rbsouzajr, 2025-05-29:
Hi,

I am using the rtl8822bu chipset. It reports an error after the command wpa_cli -i wlan1 p2p_group_add. This is because it is not possible to create the p2p-dev-wlan interface. The repository informs that this driver is compatible with MIRACAST. How to use this feature on the raspberry bullseye 32 bits?

> lsusb
Bus 001 Device 004: ID 0bda:b82c Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

> iw dev
phy#0
        Interface wlan1
                ifindex 4
                wdev 0x2
                addr 92:de:80:92:62:39
                type managed
                channel 11 (2462 MHz), width: 40 MHz, center1: 2452 MHz
                txpower -100.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 90:de:80:92:62:39
                type managed
                channel 11 (2462 MHz), width: 40 MHz, center1: 2452 MHz
                txpower -100.00 dBm

morrownr, 2025-05-29:
@rbsouzajr 

Whether you use the rtw88 repo, https://github.com/lwfinger/rtw88 , or the same driver in kernels 6.12+, you can repo bugs at the rtw88 repo. Let me know if you have any problems.

rbsouzajr, 2025-05-29:
Hello,

Thanks for your help. I installed the driver on the Raspberry Pi 2 with Bullseye OS and the p2p-wlan0-0 interface worked normally.
Driver instalado: [https://github.com/lwfinger/rtw88](url)

But how to enable CONCURRENT mode, with two interfaces? Is it possible?
Is there no documentation to check the available parameters?

> iw dev
phy#0
        Interface p2p-wlan0-0
                ifindex 4
                wdev 0x2
                addr 92:de:80:92:62:39
                ssid DIRECT-hN
                type P2P-GO
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 30.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcolt                                                                                                                                                             x-bytes tx-packets
                        0       0       0       0       0       0       0      0                                                                                                                                                             0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 90:de:80:92:62:39
                type managed
                txpower 30.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcolt                                                                                                                                                             x-bytes tx-packets
                        0       0       0       0       0       0       0      0

morrownr, 2025-05-29:
> But how to enable CONCURRENT mode, with two interfaces? Is it possible?

CONCURRENT mode is Realtek specific. The driver you are using now is Linux Wireless Standards (mac80211) compliant so it will work with standard Linux wireless docs. What you want is valid interface combinations:

$ iw list

Then search in the output for lines similar to the below line and that will tell you what the driver is capable of:

```
valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2

```

If you need me to explain it, ask.

> Is there no documentation to check the available parameters?

You can ask the driver for the allowable parameters but I have warn you that this is very different from Realtek drivers. The Realtek drivers are like a standard transmission whereas the standards compliant drivers are like an automatic transmission. Most things are automatic. Example: No more parameter for USB3. The driver will detect and set USB3 mode if it is available, if not, it leaves USB2 mode.

You can run the following to see your drivers:

$ lsmod

The driver that your adapter is using will be:

rtw88_8822bu

You can check for parameters with:

$ modinfo rtw88_8822bu

If you have further questions, you can ask over at the new rtw88 repo. I hang out there as well.