morrownr/88x2bu-20210702 #255: failed to get tx report

craigerl, 2025-07-24:
I realize this project is closed, but I see this on kernel 6.12 (raspberry Pi OS updated).  After
two days of use, we get to this point,

[193556.322942] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193557.132928] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193559.052925] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193561.052922] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193563.052915] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193565.213909] wlan1: authenticate with 94:10:3e:0c:62:09 (local address=bc:07:1d:72:1f:37)
[193565.243250] wlan1: send auth to 94:10:3e:0c:62:09 (try 1/3)
[193565.752910] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193566.622952] wlan1: send auth to 94:10:3e:0c:62:09 (try 2/3)
[193567.132906] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193567.665446] wlan1: send auth to 94:10:3e:0c:62:09 (try 3/3)
[193568.172904] rtw_8822bu 1-1:1.0: failed to get tx report from firmware 
[193568.622955] wlan1: authentication with 94:10:3e:0c:62:09 timed out
[193574.888999] wlan1: authenticate with c0:56:27:68:4f:b9 (local address=bc:07:1d:72:1f:37)
[193574.943162] wlan1: send auth to c0:56:27:68:4f:b9 (try 1/3) 
[193575.452886] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193575.562926] wlan1: send auth to c0:56:27:68:4f:b9 (try 2/3)
[193576.072889] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193576.622945] wlan1: send auth to c0:56:27:68:4f:b9 (try 3/3)
[193577.132885] rtw_8822bu 1-1:1.0: failed to get tx report from firmware
[193577.662988] wlan1: authentication with c0:56:27:68:4f:b9 timed out

(manually reconnect usb cable)

[194316.341399] usb 1-1: USB disconnect, device number 2
[194323.481259] usb 1-1: new high-speed USB device number 3 using dwc2
[194323.701988] usb 1-1: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
[194323.702024] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[194323.702039] usb 1-1: Product: 802.11ac NIC
[194323.702050] usb 1-1: Manufacturer: Realtek
[194323.702061] usb 1-1: SerialNumber: 123456
[194323.735060] rtw_8822bu 1-1:1.0: Firmware version 27.2.0, H2C version 13
[194324.022007] rtw_8822bu 1-1:1.0: write register 0xc4 failed with -71
[194324.022109] rtw_8822bu 1-1:1.0: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[194324.023097] usb 1-1: USB disconnect, device number 3
[194324.631256] usb 1-1: new high-speed USB device number 4 using dwc2
[194324.841936] usb 1-1: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
[194324.841972] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[194324.841986] usb 1-1: Product: 802.11ac NIC
[194324.841998] usb 1-1: Manufacturer: Realtek
[194324.842008] usb 1-1: SerialNumber: 123456
[194324.844244] rtw_8822bu 1-1:1.0: Firmware version 27.2.0, H2C version 13
[194332.846749] wlan1: authenticate with c0:56:27:68:4f:b9 (local address=bc:07:1d:72:1f:37)
[194332.901619] wlan1: send auth to c0:56:27:68:4f:b9 (try 1/3)
[194332.908276] wlan1: authenticated
[194332.911349] wlan1: associate with c0:56:27:68:4f:b9 (try 1/3)
[194332.933055] wlan1: RX AssocResp from c0:56:27:68:4f:b9 (capab=0x411 status=0 aid=6)
[194332.941902] wlan1: associated

craigerl, 2025-07-24:
Linux cool 6.12.25+rpt-rpi-v7

```
$ lsmod | grep rtw
rtw88_8822bu           12288  0
rtw88_8822b           221184  1 rtw88_8822bu
rtw88_usb              16384  1 rtw88_8822bu
rtw88_core            159744  2 rtw88_8822b,rtw88_usb
mac80211              958464  2 rtw88_core,rtw88_usb
cfg80211              888832  3 rtw88_core,mac80211,brcmfmac
```

morrownr, 2025-07-25:
Yes, you have the correct rtw88 repo. That one was and is used to upstream work to the Linux kernel.

morrownr, 2025-07-26:
@craigerl 

That should work as long as your distro has turned on compiling on the new in-kernel driver and if they are now installed the proper firmware file. Most of the mainsteam distros should have handled that. If you have a problem, let me know.